MichaIng/DietPi #3508: Issue

mamton, 2020-05-05:
#### Details:
- Date           | Tue May  5 14:10:55 +04 2020
- DietPi version | v6.29.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux MediaHQ-DLAgent-RPi3 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4)
- Command        | umount /dev/mmcblk0p1
- Exit code      | 32
- Software title | DietPi-Drive_Manager
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
umount: /boot: target is busy
        (In some cases useful info about processes that
         use the device is found by lsof(8) or fuser(1).)
```
---------------------------------------------------------------------

Joulinar, 2020-05-05:
Hi,

many thanks for your report. Pls can you describe your error message a little bit more. What exactly you are trying to do? Where do you get this issue? I guess you are trying to move your bootFS to an USB device? if yes, pls ensure on CLI you are not located in `/boot` or a sub directory. Best would be to stay on `root level /`.

MichaIng, 2020-05-18:
Marking as closed tue to missing reply. Usual reason is what Joulinar already assumed: Current directory of any shell is inside /boot.

I guess the task was a fsck, as we do not support moving the boot file system to an external drive currently.

closed 2020-05-18 by MichaIng