MichaIng/DietPi #3509: dietpi-ramlog: No such file or directory

Vec7or, 2020-05-05:
#### Details:
- Date           | Tue May  5 14:59:38 BST 2020
- DietPi version | v6.29.2 (Fourdee/master)
- Image creator    | DietPi Core Team
- Pre-image          | FriendlyARM
- Hardware           | NanoPi NEO (armv7l) (ID=60)
- Kernel version   | Linux DietPi 4.14.0 #1 SMP Mon Dec 18 11:43:22 CST 2017 armv7l GNU/Linux
- Distro                 | stretch (ID=4)
- Command          | /boot/dietpi/func/dietpi-ramlog 1
- Exit code            | 127
- Software title      | DietPi-Software
#### Steps to reproduce:

1. Download DietPi_v6.14_NanoPiNEO-ARMv7-Stretch.img
2. Write the image to SD-Card
3. Adjust both the software password and user passwords to the same password
4. Don't keep p7zip-full
5. Opt Out of DietPi-survey
6. Log in after reboot
7. Don't select any software
8. Select install
#### Expected behaviour:
- DietPi should be installed
#### Actual behaviour:
- The error mentioned in `Additional logs` is happening
#### Extra details:
- ...
#### Additional logs:
```
/boot/dietpi/func/dietpi-globals: line 939: /boot/dietpi/func/dietpi-ramlog: No such file or directory
```

Joulinar, 2020-05-05:
Hi,

many thanks for your report. As workaround you could try to create the file and reboot your system afterwards.

`touch /boot/dietpi/func/dietpi-ramlog`

Vec7or, 2020-05-07:
> Hi,
> 
> many thanks for your report. As workaround you could try to create the file and reboot your system afterwards.
> 
> `touch /boot/dietpi/func/dietpi-ramlog`

I can confirm that the workaround seems to be functioning, thanks. The file `/boot/dietpi/dietpi-login` also seems to be missing however this does not seemingly impair the installation.

```
Mode: Installation completed

[ SUB1 ] DietPi-Services > restart 
[  OK  ] DietPi-Services | restart : cron
/boot/dietpi/dietpi-login: line 231: /boot/dietpi/func/dietpi-banner: No such file or directory
```

Joulinar, 2020-05-07:
well it's more `/boot/dietpi/func/dietpi-banner` missing. Usually it should be gone if you login as user `root`.

Vec7or, 2020-05-07:
> well it's more `/boot/dietpi/func/dietpi-banner` missing. Usually it should be gone if you login as user `root`.

I am logged on as user root and the message still remains (even after reboot). As previously stated it does not impair functionality I just wanted you to know in case something needs to be done about it.

Vec7or, 2020-05-07:
> well it's more `/boot/dietpi/func/dietpi-banner` missing. Usually it should be gone if you login as user `root`.

```
Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
/boot/dietpi/dietpi-login: line 231: /boot/dietpi/func/dietpi-banner: No such file or directory
root@DietPi:~# 
```
I am logged on as user root and the message still remains (even after reboot). As previously stated it does not impair functionality I just wanted you to know in case something needs to be done about it.

Joulinar, 2020-05-07:
can you open `dietpi-banner` to have it running once. Usually it should create the file

Vec7or, 2020-05-07:
Like this?
```
root@DietPi:~# dietpi-banner
-bash: /boot/dietpi/func/dietpi-banner: No such file or directory
root@DietPi:~#
```

Joulinar, 2020-05-07:
uhh looks like it's missing completely. Can you have a look to

`ls -la /boot/dietpi/func/`

Vec7or, 2020-05-07:
As the error states the file is indeed not present:
```
root@DietPi:~# ls -la /boot/dietpi/func/
total 224
drwxr-xr-x 2 root root  2048 May  7 10:43 .
drwxr-xr-x 4 root root  4096 May  7 10:44 ..
-rwxr-xr-x 1 root root  1638 May  7 10:38 change_hostname
-rwxr-xr-x 1 root root  1909 May  7 10:38 create_mysql_db
-rwxr-xr-x 1 root root 12741 May  7 10:38 dietpi-benchmark
-rwxr-xr-x 1 root root 71702 May  7 10:38 dietpi-globals
-rwxr-xr-x 1 root root  1944 May  7 10:38 dietpi-optimal_mtu
-rwxr-xr-x 1 root root     0 May  7 10:43 dietpi-ramlog
-rwxr-xr-x 1 root root  8224 May  7 10:38 dietpi-set_cpu
-rwxr-xr-x 1 root root 62295 May  7 10:38 dietpi-set_hardware
-rwxr-xr-x 1 root root 19591 May  7 10:38 dietpi-set_software
-rwxr-xr-x 1 root root  4590 May  7 10:38 dietpi-set_swapfile
-rwxr-xr-x 1 root root  5572 May  7 10:38 dietpi-set_userdata
-rwxr-xr-x 1 root root  9165 May  7 10:38 dietpi-wifidb
-rwxr-xr-x 1 root root  4374 May  7 10:38 obtain_network_details
-rwxr-xr-x 1 root root  4177 May  7 10:38 run_ntpd
```

Joulinar, 2020-05-07:
ok you are missing quite some files. Seems something did goes wrong during update. I would recommend to clone the directory again

Vec7or, 2020-05-07:
Well this is a clean installation so i can't really do something about it or can I? Or do I have to build the image myself?

Joulinar, 2020-05-07:
What I meant was, that something during the update to DietPi v6.29.2 did not succeeded correctly as you are missing some data. So there are basically 2 ways. First one would be to clone the entire directory `/boot/dietpi/func/` from git to get the missing files or to start with a new fresh install (just in case you don't have installed to much atm)

Vec7or, 2020-05-07:
Okay thank you for the clarification. I have cloned the folder, since a fresh installation would not work due to the image update process being broken when using v6.14 image (this was a fresh install from today). 

With the cloned folder there are no more errors. Looks good to me as a workaround.

Joulinar, 2020-05-07:
Ok would it be possible to post your solution, how you cloned the directory. It might help others if they will hit similar

Vec7or, 2020-05-07:
I did this (be aware i have not extensively tested if everything works but as far as I have tested it does): 
```
apt install git
git clone https://github.com/MichaIng/DietPi.git Dietpi
rm -r /boot/dietpi/func/*
cp -r DietPi/dietpi/func/* /boot/dietpi/func/
rm -r Dietpi
apt remove git
init 6
```

Vec7or, 2020-05-07:
Just installed a new image. As soon as the `/boot/dietpi/func/dietpi-ramlog: No such file or directory` error occurs return to the command line, do the steps mentioned above and restart.

After the restart the image works as expected.

Joulinar, 2020-05-07:
Thx for sharing the workaround.

Regarding the initial issue, I guess @MichaIng would need to have a look.

MichaIng, 2020-05-07:
@Vec7or @Joulinar 
Many thanks for report and workaround.

Very strange, especially since the update does not remove files from /boot/dietpi but only copies the ones from /DietPi/dietpi to /boot/dietpi. So even if this copy process would fail, the files would still be present. Even more, since the update is internally restarted, files are re-downloaded to /boot/dietpi actively. So this must have been failed at two stages... I'll run a test boot to see if I can replicate within VM. Own RPi device/server is started tomorrow, the ISP technician took 10 days longer to finally tomorrow enable our broad band internet....

Joulinar, 2020-05-07:
@MichaIng 
that happen if you move from a big city to a small country side laughing:

btw have seen similar on our forum https://dietpi.com/phpbb/viewtopic.php?f=11&t=7607 as well as on the duplicate GitHub #3513 

All seems to run Strech

MichaIng, 2020-05-08:
Analysing the case on fresh NanoPi NEO (Stretch) image within systemd-nspawn container:
```
2020-05-08 20:48:04 root@micha:/mnt/sda# l m/boot/dietpi/func/
total 230K
-rwxr-xr-x 1 root root 1.6K May  8 20:39 change_hostname
-rwxr-xr-x 1 root root 1.9K May  8 20:39 create_mysql_db
-rwxr-xr-x 1 root root  13K May  8 20:39 dietpi-benchmark
-rwxr-xr-x 1 root root  71K May  8 20:39 dietpi-globals
-rwxr-xr-x 1 root root 4.5K May  8 20:38 dietpi-led_control
-rwxr-xr-x 1 root root 1.9K May  8 20:39 dietpi-optimal_mtu
-rwxr-xr-x 1 root root 8.1K May  8 20:39 dietpi-set_cpu
-rwxr-xr-x 1 root root 4.6K May  8 20:38 dietpi-set_dphys-swapfile
-rwxr-xr-x 1 root root  61K May  8 20:39 dietpi-set_hardware
-rwxr-xr-x 1 root root  20K May  8 20:39 dietpi-set_software
-rwxr-xr-x 1 root root 4.5K May  8 20:39 dietpi-set_swapfile
-rwxr-xr-x 1 root root 5.5K May  8 20:39 dietpi-set_userdata
-rwxr-xr-x 1 root root 9.0K May  8 20:39 dietpi-wifidb
-rwxr-xr-x 1 root root 4.3K May  8 20:39 obtain_network_details
-rwxr-xr-x 1 root root 4.1K May  8 20:39 run_ntpd
```
- Indeed those scripts are missing.
- Those match the ones that were present in /DietPi/dietpi/ in the past instead of /DietPi/dietpi/func/, but e.g. `dietpi-obtain_hw_model`, which was always in func/, is also missing.
- Checking scripts and other places show that the following code was definitely run: https://github.com/MichaIng/DietPi/blob/master/dietpi/patch_file#L73-L94
- `cp -Rf /DietPi/* /boot/` should have copied all code, that was installed before... did it fail (no error message) or an async issue or were scripts removed afterwards somehow? And why is this issue present on some Stretch SBCs only?

I'll create a debug branch to pause the update at certain stages and have a subshell for investigation...

Joulinar, 2020-05-08:
I see I need to get more and different SBC to be able test more in detail

MichaIng, 2020-05-08:
@Joulinar 
Yes same for me. It's just out of scope for me currently to get an SBC battery, I have simply no space here to place them ready like a testing farm 🤔. Will change hopefully until next year, and Fourdee (Dan/Founder) already offered to send over his space SBCs (he had ALL that DietPi officially supported).

However back to topic:
- 1. Update debug break: https://github.com/MichaIng/DietPi/blob/debug/dietpi/patch_file#L19
  - All files are present.
- 2. ...

Joulinar, 2020-05-09:
@MichaIng 
ok let's continue discussion on testing and SBC farm somewhere else to don't spam people to much

closed 2020-05-10 by MichaIng