MichaIng/DietPi #3510: RPi 3 b+ Wifi Hardware not found

elitearmedforce, 2020-05-05:
### Creating a bug report/issue

#### Required Information
- DietPi version | v6.29.2
- Distro version | Buster
- Kernel version | Linux DietPi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
- SBC model | RPi 3 Model B+ (armv7l)
- Power supply used | 5V
- SDcard used | Samsung EVO 32 GB

#### Additional Information (if applicable)
- Software title | NONE
- Was the software title installed freshly or updated/migrated? NONE
- Can this issue be replicated on a fresh installation of DietPi? YES, tried on another board, same issue

- Bug report ID | 96edd9cc-fb67-4662-ada7-59515d679138

#### Steps to reproduce

1. This is a clean version of DietPi installed from the website.
2. After flashing, switched on the rpi3 b+ with ethernet cable connected. OpenSSH
3. I am not sure about this step. Either first, it updated to the latest DietPi 6.29.2 then prompted global password change or first prompt global password change then updated to latest version.
4. Ran Dietpi-update to confirm it is updated or not.
5. Ran dietpi-config to configure the wifi. 
6. Issues found(Check the screenshot attached below)

#### Expected behaviour

Duh!! Once enabled, I can scan Wifi in my area, connect to my wifi by adding a password.

#### Actual behaviour

No supported WiFi hardware was found. Then it prompts me to "Would you like to disable Wifi?"

#### Extra details

- The previous version of DietPi never had such issues. It would be preferred that the users get the option to upgrade or not.
- To confirm its a hardware problem or not, I checked with another RPi3 b+ board and faced the same issue. Please help!!! 
- I mostly use this for Media Server : Emby.

wifi => Not Enabled(before installing)
[image]
after installing wifi thingy => No hardware found (Even after rebooting same issue)
[image]

elitearmedforce, 2020-05-06:
Workaround, somewhat. 
This one is for people who are facing same issue, and need immediate working. This method is a kind of headache. Hoping for an immediate fix from the devs.

Requirements:
Monitor/screen(coz u need to run dietpi with ssh)

Steps:
1) First of all, you need a clean format and flash of Dietpi from the website. (v6.28)
2) Before booting, disconnect the hardware from the internet. (Very important, as dietpi always try update/upgrade to latest version)(Remember do not attach Ethernet, yet)
3) Enter the default login and password
4) It might prompt to connect to internet, now type "dietpi-config"
5) Go to the network section, enable Wifi modules.  After, reboot the system.
6) Once reboot, connect to your wifi with scan in dietpi-config section.
7) Tada!!! ur connected with Wifi.
8) Now it will automatically update/upgrade to v6.29.2. Don't worry, no issues will be faced after this. 

This procedure worked for me successfully. I hope it works for others. Trial Error Method

Joulinar, 2020-05-06:
Hi,

many thanks for your report and sharing a ways around it. I can confirm the behavior on my RPi3B+ as well. I guess there is an issue in correct hardware detection. Unfortunately we would need to wait for the developer to have a look.

elitearmedforce, 2020-05-06:
My pleasure, I hope the same.

MichaIng, 2020-05-07:
Many thanks for reporting and workaround guys, I found the bug: https://github.com/MichaIng/DietPi/commit/2f032f1be24dba297762bf50d8d9ff1db65e9c39
Could you try:
```
wget https://raw.githubusercontent.com/MichaIng/DietPi/dev/dietpi/func/dietpi-obtain_hw_model -O /boot/dietpi/func/dietpi-obtain_hw_model
/boot/dietpi/func/dietpi-obtain_hw_model
dietpi-config
```

The network options menu should now contain an additional entry: `Onboard WiFi`
These are disabled on first boot if WiFi has not been enabled via `dietpi.txt`. It must then be enabled + reboot. However the menu entry was missing since RPi onboard WiFi was broken.

Joulinar, 2020-05-07:
@MichaIng 
I can confirm the solution. Option for Onboard WiFi is back again.

elitearmedforce, 2020-05-08:
@MichaIng & @Joulinar 
It works, perfect. Thank you for the solution. Keep up the good work.

closed 2020-05-10 by elitearmedforce