MichaIng/DietPi #3511: RootFS is currently Read Only (R/O) mounted. Aborting...

Adsouza98, 2020-05-06:
#### Required Information
- DietPi version | v6.29.2
- Distro version | buster
- Kernel version | Linux DietPi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
- SBC model | RPi 4 Model B (armv7l)
- Power supply used | 5V 1A RAVpower
- SDcard used | 32 GB SanDisk ultra

#### Additional Information (if applicable)
- The software title installed freshly
- Can this issue be replicated on a fresh installation of DietPi? **YES**.

#### Steps to reproduce

1. Installed Fresh version of DietPi from website
2. Flashed SD Card with image
3. Instlled Sonarr, Nzbget, Deluge
4. Set a static IP address
6. Installed Open VPN client with `sudo apt-get install -y openvpn`
7. mounted my NAS via NFS

#### Expected behaviour
- Running as usual 
- Softwares working

#### Actual behaviour
-  After rebooting RootFS becomes Read only
- All Software stops working
- Unable to use dietpi-_______ commands except for dietpi-explorer and dietpi-drive_manager
- When I try and run dietpi-Software for example I get:

> RootFS is currently Read Only (R/O) mounted. Aborting
>DietPi requires RootFS to be Read/Write (R/W) mounted. Please run "dietpi-drive_manager" to re-enable.

[image]

When I run dietpi-drive_manager it notices all mounted drives but at the end it says:
> cp: cannot create regular file '/etc/fstab': Read-only file system

[image]

in the dietpi-drive_manager it says
> Please select a drive to see available options.
>      - User data location: RootFS (/mnt/dietpi_userdata)

[image]

[image]

When I choose the `/boot` option and select `Transfer RootFS` it says:

> It says [WARNING] Partition must be unmounted, before format and RootFS transfer can begin.
Please unmount the partition, then try again.

[image]

I tried doing a fresh install 3 times and still have the same problem.

Any help to fix this will be appreciated.

zegxro, 2020-05-06:
Same here.. at 1st it happen when ever i install plex media server.. but after a while.. it seem to happen just after i freshly installed dietpi..

I also notice it will change RootFS to read-only after the swap failed.. when you try to mount using dietpi-drive manager.. sometimes it will mount.. but most of the time it wont mount no matter how many time you try using dietpi-drive manager..

The only way to mount it is by using [ mount -o remount,rw / ] but.. after a reboot.. RootFS will change back to read-only stat..

Been struggling the whole day since yesterday but failed to find permanent solution.. change 3 new SD Card with different size but same result.. tested on 2 raspberry pi version - raspberry pi v2 n v3 both have the same error.

sealteam666, 2020-05-06:
same happened with I changed the hostname from dietpi to pi then i rebooted then it  changed to rootfs all the software stopped working also i updated it yesterday from 6.28 to 6.29.2..

please help

elitearmedforce, 2020-05-06:
I faced exactly same issue too, This happens only when update/upgrade from v6.28 to v6.29.2.
I solved this issue with clean installation. 
(Warning) The problem with a clean installation, that if u use wifi instead of ethernet. It will mention no WIFI hardware found. I have already posted an issue for that, hoping for some fixes. By meantime I found a workaround for the wifi issue, u can check in Issues section

Joulinar, 2020-05-06:
Hi Guys,

looks like it's happen to everybody and it's doesn't seems to have a pattern. Just for information, pls can you have a look to `/etc/fstab` how the rootFS mount entry looks like?

elitearmedforce, 2020-05-06:
> Hi Guys,
> 
> looks like it's happen to everybody and it's doesn't seems to have a pattern. Just for information, pls can you have a look to `/etc/fstab` how the rootFS mount entry looks like?

Could u pls mention what, exactly u need from fstab file?

using rpi3 b+

Joulinar, 2020-05-06:
Hi,

in theory it should contain these 2 lines at the end of the file for rootFS as well as bootFS

```
#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
PARTUUID=6c586e13-02 / ext4 noatime,lazytime,rw 0 1
PARTUUID=6c586e13-01 /boot vfat noatime,lazytime,rw 0 2

```

elitearmedforce, 2020-05-06:
> Hi,
>
> in theory it should contain these 2 lines at the end of the file for rootFS as well as bootFS
>
> ```
> #----------------------------------------------------------------
> # PHYSICAL DRIVES [...]

PARTUUID=6c586e13-02 / ext4 noatime,lazytime,rw 0 1
PARTUUID=6c586e13-01 /boot vfat noatime,lazytime,rw 0 2
/var/swap none swap sw 0 0
tmpfs /var/log tmpfs size=50M,noatime,lazytime,nodev,nosuid,mode=1777
/dev/sda2 /mnt/ssd ntfs-3g defaults 0 0
/dev/sdb2 /mnt/ssd ntfs-3g defaults 0 0
/dev/sdc2 /mnt/ssd ntfs-3g defaults 0 0

Last 3 lines are added by me, for mounting drives after boot

Joulinar, 2020-05-06:
that's fine but if I understood right, your installation is working now. So we need a failed setup to compare.

elitearmedforce, 2020-05-06:
> that's fine but if I understood right, your installation is working now. So we need a failed setup to compare.

Yes, presently. I am w8ing too for comparison

Joulinar, 2020-05-06:
OK guys, I was able to produce the error by using `dietpi-drive_manager`. On my test case drive manager was removing the line responsible for the rootFS

`PARTUUID=6c586e13-02 / ext4 noatime,lazytime,rw 0 1`

It was simply not there anymore. After reboot the rootFS was ro and all stop working. I got it back as follow

1. login as `root`
2. `mount -o remount,rw /`
3. `nano /etc/fstab`
4. add the line for rootFS > `PARTUUID=<YOUR-ID>-02 / ext4 noatime,lazytime,rw 0 `
5. `PARTUUID` needs to be adjusted, should be same like `/boot` if you use SD card. Be careful and use the right partition `-02`
6. save and exit
7. `reboot`

Daniel-Myers, 2020-05-06:
My setup currently isn't working. My etc/fstab contains
`PARTUUID=6c586e13-01 /boot vfat noatime,lazytime,rw 0 2
#UUID=2ab3f8e1-7dc6-43f5-b0db-dd5759d51d4e /mnt/2ab3f8e1-7dc6-43f5-b0db0dd5759d51d4e ext4 [noatime,lazytime,rw,nofail,noauto,x-systemd.automount`
[image]

zegxro, 2020-05-06:
> OK guys, I was able to produce the error by using `dietpi-drive_manager`. On my test case, drive manager was removing the line responsible for the rootFS
> 
> `PARTUUID=6c586e13-02 / ext4 noatime,lazytime,rw 0 1`
> 
> It was simply not there anymore. After reboot the rootFS was ro and all stop working. I got it back as follow
> 
> 1. login as `root`
> 2. `mount -o remount,rw /`
> 3. `nano /etc/fstab`
> 4. add the line for rootFS > `PARTUUID=<YOUR-ID>-02 / ext4 noatime,lazytime,rw 0 1`
> 5. `PARTUUID` needs to be adjusted, should be same like `/boot` if you use SD card. Be careful and use the right partition `-02`
> 6. save and exit
> 7. `reboot`

Testing it.. so far so good.. but i need to test it with plex media server + NAS drive mounted.. will post back ASAP..

Joulinar, 2020-05-06:
@Daniel-Myers 
yes pls try the fix I posted above https://github.com/MichaIng/DietPi/issues/3511#issuecomment-624525544

zegxro, 2020-05-06:
> @Daniel-Myers
> yes pls try the workaround I posted above [#3511 (comment)](https://github.com/MichaIng/DietPi/issues/3511#issuecomment-624525544)
> 
> @zegxro
> be careful while suing drive manager. It could lead to same again

got it.. but if it happen again.. i just need to add [ PARTUUID=<YOUR-ID>-02 / ext4 noatime,lazytime,rw 0 1 ] again ?..

Joulinar, 2020-05-06:
yes, that would be the workaround for the time being until we have a permanent fix

Daniel-Myers, 2020-05-06:
> 
> 
> @Daniel-Myers
> yes pls try the workaround I posted above [#3511 (comment)](https://github.com/MichaIng/DietPi/issues/3511#issuecomment-624525544)

Thanks, that's fixed it for now. Do have any idea why that line get's removed?

Joulinar, 2020-05-06:
probably a bug inside the new release. Would need to be verified by the developer.

mamton, 2020-05-06:
Just to help or not. I'm just rebooting after your fix.

I have had this happen now on 3 separate RPi3s. I have some spares to thought it was a problem with the first one. Then tried multiple cards as well. 8gb/16gb/32gb from different manufacturers. Forgot I tried it on a RPi4 as well. yesterday.

So far your fix has worked. Just rebooted. I'm going to open fstab and copy/paste the same like with a comment so its easier to put in next time.

UlrikWKoren, 2020-05-06:
Hi, i get this error after installing nextcloud with external hdd mounted on rpi4.
im wondering where do i find my rootFS id to put in the fstab file?

ludovickk, 2020-05-06:
The problem occurs right after using DietPi-Drive_Manager.

Joulinar, 2020-05-06:
@ludovickk 
Yes known. A workaround is provided above

LeVraiRoiDHyrule, 2020-05-06:
Hi, I've got the same issue but I can't figure out how to apply the solution. I'm new and I don't know how to manage the fstab file. Here is mine : 

https://pastebin.com/Nhx8Zr4T

I can't find what I should write instead of <YOUR-ID> .
I don't know how to find this ID. Can someone help me ? Sorry if I'm a noob. Any answer is welcome and have a good day

UlrikWKoren, 2020-05-06:
> Hi, I've got the same issue but I can't figure out how to apply the solution. I'm new and I don't know how to manage the fstab file. Here is mine :
>
>
>
> https://pastebin.com/Nhx8Zr4T
>
>
>
> I can't find [...]

Im wondering the same, i Can’t find the ID.

Daniel-Myers, 2020-05-06:
> Hi, I've got the same issue but I can't figure out how to apply the solution. I'm new and I don't know how to manage the fstab file. Here is mine :
>
> https://pastebin.com/Nhx8Zr4T
>
> I can't find what [...]

The ID is the same as the previous PARTUUID, so it is 6c586e13, that's what I did anyway and it worked. This may be different for different people.

UlrikWKoren, 2020-05-06:
> The ID is the same as the previous PARTUUID, so it is 6c586e13, that's what I did anyway and it worked. This may be different for different people.

Thanks, i will try when i get home😊

LeVraiRoiDHyrule, 2020-05-06:
> The ID is the same as the previous PARTUUID, so it is 6c586e13, that's what I did anyway and it worked. This may be different for different people.

I would like to try that but I've got another issue : I can't edit my fstab file, it's protected and read-only

[image]
How can I unlock it ?

Joulinar, 2020-05-06:
Guys it depends on your setup. If you use SD card only, than the ID is same as used for `/boot`. In my case it's like this. Both ID are same except the `-02` at the end

```
PARTUUID=6c586e13-02 / ext4 noatime,lazytime,rw 0 1
PARTUUID=6c586e13-01 /boot vfat noatime,lazytime,rw 0 2
```

However if you transfered your rootFS to an external device like USB HDD, the ID will be different than `/boot`

You can use lsblk to display our USB devices and to get the ID just use `blkid`

UlrikWKoren, 2020-05-06:
> I would like to try that but I've got another issue : I can't edit my fstab file, it's protected and read-only
>
>
>
> [image]
>
> How can I unlock it ?

Write this first in command line, then try to Edit FSTab $ mount -o remount,rw /

Joulinar, 2020-05-06:
@LeVraiRoiDHyrule 
did you have done `mount -o remount,rw /` before as described on the workaround?

LeVraiRoiDHyrule, 2020-05-06:
No I didn't, sorry I'm just dumb.......

Joulinar, 2020-05-06:
nothing to excuse, we are there to help 😉

LeVraiRoiDHyrule, 2020-05-06:
@Joulinar It's fixed, thanks !!!! As I didn't modified anything about the rootfs partition, it was the same ID as yours. Thank you so much for your patience and your help. Have a good day !!!

Chublo, 2020-05-06:
Ok, so it appears I made an error when putting the PARTUUID into FSTAB. Now I can't run mount -o remount,rw / so that I can edit the file again. Any ideas?

root@DietPi:~# mount -o remount,rw /
mount: /: can't find PARTUUID=cb7b86f7-01-02.

Joulinar, 2020-05-06:
@Chublo 
do you have stored your rootFS on an external device or still using SD card? Pls can you provide output of `lsblk`

Chublo, 2020-05-06:
Still on SD

root@DietPi:~# lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0  7.4G  0 disk 
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0  7.1G  0 part /
root@DietPi:~# blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="F661-303B" TYPE="vfat" PARTUUID="cb7b86f7-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="8d008fde-f12a-47f7-8519-197ea707d3d4" TYPE="ext4" PARTUUID="cb7b86f7-02"
/dev/mmcblk0: PTUUID="cb7b86f7" PTTYPE="dos"

Joulinar, 2020-05-06:
ok issue is that you have -01-02 at the end of your ID but it should be -02 only

just try following
```
mkdir /mnt/temp
mount /dev/mmcblk0p2 /mnt/temp
nano /mnt/temp/etc/fstab
```

Chublo, 2020-05-06:
Read-only file system :(

`root@DietPi:~# mkdir /mnt/temp
mkdir: cannot create directory ‘/mnt/temp’: Read-only file system
root@DietPi:~# 
`

Joulinar, 2020-05-06:
ok can you create something on `/boot` like

`mkdir /boot/temp`

Chublo, 2020-05-06:
That works, but can't mount to it - already mounted.
`root@DietPi:~# mkdir /boot/temp
root@DietPi:~# mount /dev/mmcblk0p2 /boot/temp
mount: /boot/temp: /dev/mmcblk0p2 already mounted on /.
root@DietPi:~# `

reeslo, 2020-05-06:
Today I've updated 2 Pi `RPi B+` and `RPi 3 B`

I have the issue on RPi 3 B only, just after a power outage (due to improper fan pin placement :/).
Just after reboot I see RootFS was ro :'(

I thought of a problem following the update, I rebooted the RPi B, and I did not notice the issue.

For me this line was not present into /etc/fstab, and above solution, resolve it :)
```
PARTUUID=6c586e13-02 / ext4 noatime,lazytime,rw 0 1
```
Thanks !

Joulinar, 2020-05-06:
@Chublo 
do you have another Linux box like a different DietPi system? Or a system that can read ext4 file systems?

Chublo, 2020-05-06:
> That works, but can't mount to it - already mounted.
> 
> ```
> root@DietPi:~# mkdir /boot/temp
> root@DietPi:~# mount /dev/mmcblk0p2 /boot/temp
> mount: /boot/temp: /dev/mmcblk0p2 already mounted on /.
> root@DietPi:~# 
> ```
This solved the issue for me and enabled me to edit the FSTAB again to correct it.  Thanks @Joulinar for your help.
'root@DietPi:/boot/temp# sudo mount -o remount,rw /dev/mmcblk0p2 /'

Adsouza98, 2020-05-06:
Fix Worked! Thank you for the help @Joulinar. Mad respects to moderators and Github forms!

Please update this thread when their is a proper solution in a new update.

Thanks again.

loris-av, 2020-05-07:
New reboot today, issue repeated :'(

Joulinar, 2020-05-07:
It could happen again if you use DietPi Drive Manager as long as there is no fix on the DietPi script's

UlrikWKoren, 2020-05-07:
Won’t the problem occour if you mount your drives manually in fstab? Instead of using dietpi-drive_manager.

Joulinar, 2020-05-07:
nope, this should not happen. During my investigation I just found that Drive Manager is deleting the line. If you edit `/etc/fstab` directly it should be ok. I mean this is the way how we implement the workaround as well 😉

mamton, 2020-05-07:
Joulinar it is a continuous problem. Every time you reboot the same problem comes up. Anytime the RPi is rebooted you have to mount/remount and adjust the fstab file. I wouldn't really call this a workaround as the problem never goes away really, every reboot the problem starts again.

Chublo, 2020-05-07:
I’ve rebooted and the issue is still resolved.

dunxd, 2020-05-07:
I'm getting the same issue with a Raspberry Pi 2 after running dietpi-update.  Was using this for Pihole, which doesn't work until the root partition is set to writeable. The workaround is working for me.

ElTav, 2020-05-08:
Also running into this same issue as well. @MichaIng any idea what the issue might be?

Joulinar, 2020-05-08:
@ElTav 
Did you used Drive Manager?

reeslo, 2020-05-08:
I confirm, issue appear every time after use `dietpi-drive_manager`.
And sometime there is problems to mount new disks.

For me issue source is `dietpi-drive_manager`

campaquino, 2020-05-08:
Hey guys, I took a look into the dietpi-drive_manager script. It seems it is using the global variable G_ROOTFS_DEV due to a recent change. But i cannot find any definition for it, neither in the script itself, nor in the called dietpi-globals script.

reeslo, 2020-05-08:
@Chublo try to mount SD card to another computer and open `/etc/fstab` file as root (from `boot` drive I think).

MichaIng, 2020-05-08:
Found and fixed the bug... nasty thing that on RPi `df` prints /dev/root for the rootfs mount source, which does not exist and hence no info on that mount was estimated: https://github.com/MichaIng/DietPi/commit/1570e47acac88414ecc8586845f24e70fb38461b

Due to circumstances I was not able to test the latest changes on native RPi. Was the very last change before DietPi v6.29 release, hence slipped through beta. Midterm we really need some clear testing schedule or even automated CI/testing. The problem just is that the very important stuff is based on the physical machine (like exactly this issue on RPi only), which cannot be covered by CI providers.

However, I'll go through the last issue with missing DietPi script files and push a hotfix release as fast as possible.

Joulinar, 2020-05-08:
@MichaIng 
Was already thinking the same to create a test plan/scenario or to create some kind of checklist for most common features. But yeah something to develop.

AstroTaco84, 2020-05-09:
Thank you all. New to rpi4 and ran into this few days ago. Work around fix provided worked.

Callumro, 2020-05-09:
Hey all, I wanted to note the workaround seems like a temporary fix. My FS keeps reverting to read-only

WolfShadowheart, 2020-05-09:
> OK guys, I was able to produce the error by using `dietpi-drive_manager`. On my test case, drive manager was removing the line responsible for the rootFS
>
> `PARTUUID=6c586e13-02 / ext4 [...]

Hello, new to the OS and so my familiarity with certain things is lacking. I've been having a heck of a time with this update because I was already having issues permanently mounting USB drives. 

I have just done another fresh flash of DietPi because I put the OS into emergency mode trying to properly configure this workaround. I have a 128GB SD card so my line shows vfat instead of ext4, I'm not sure if that has something to do with it. 

So I am very unsure as to where I'm supposed to be adding all of these lines. Upon a fresh install of dietpi, mounting drives via drive manager so that I can permanently mount my 2 hard drives, I can navigate to 

nano /etc/fstab  

and can see that I have one line

PARTUUID=6C586E13-02 /boot vfat noatime,lazytime,rw 0 1

and it seems that adding another instance above or below the line doesn't do much (like so): 

PARTUUID=6C586E13-01 /boot vfat noatime,lazytime,rw 0 1

If possible, I'd very much appreciate some guidance for how I can permanently mount my 2 USB hard drives, I'm going crazy and becoming a tad frustrated :( 

This is what I've got in my fstab 

__________________________
PHYSICAL DRIVES

PARTUUID=6C586E13-01 /boot vfat noatime,lazytime,rw 0 1
PARTUUID=6C586E13-02 /boot vfat noatime,lazytime,rw 0 1
UUID=F4A0F077A0F041A2 /mnt/F4A0F077A0F041A2 ntfs noatime,lazytime,rw,permissions,nofail,noauto,x-systemd.auto$

UUID=41428830-e1b0-8a2d-ca7bacca /mnt/41428830-e1b0-8a2d-ca7bacca ext4 noatime,lazytime,rw,permissions,nofail,noauto,x-systemd.auto$

#UUID=2ab3f8e1-7dc6-43f5-b0db-dd5759d51d4e /mnt/2ab3f8e1-7dc6-43f5-b0db-dd5759d51d4e ext4 noatime,lazytime,rw,permissions,nofail,noauto,x-systemd.auto$

UUID=F4A0F077A0F041A2 / auto defaults,noatime,rw 0 1
UUID=41428830-e1b0-44af-8a2d-ca7bacca9992 / auto defaults,noatime,rw 0 1

________________________

Each reboot I am told that /etc/fstab is unwritable, so I imagine I'm doing something wrong. 

I'm at a loss of what to do, I see that #3511 is the answer however I am not understanding how to use it to resolve my problem and keep these hard drives permanently mounted. 

Thanks a bunch for any help offered.

[18 more replies]

closed 2020-05-10 by MichaIng