MichaIng/DietPi #3513: Update to 6.29.2 failed

littis, 2020-05-07:
### Creating a bug report/issue

#### Required Information
- Date           | Wed  6 May 21:08:41 EDT 2020
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1253 SMP Thu Aug 15 11:49:46 BST 2019
- Distro         | stretch (index=4)
- Command        | /DietPi/dietpi/func/dietpi-ramdisk 1
- Exit code      | 127
- Power supply used | 5V 2.5A
- SDcard used | Samsung 32GB EVO

#### Additional Information (if applicable)
- Software title | DietPi-Update

#### Steps to reproduce

1. ran sudo dietpi-update

#### Expected behaviour

Successful installed

#### Actual behaviour

Failed install

#### Extra details:
[FAILED] DietPi-Update | /DietPi/dietpi/func/dietpi-ramdisk 1
[FAILED] DietPi-Update | /DietPi/dietpi/func/dietpi-ramdisk 1
shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
chdir: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
[FAILED] DietPi-Update | /DietPi/dietpi/func/dietpi-ramdisk 1

#### Additional logs:
```
Log file contents:
/DietPi/dietpi/func/dietpi-globals: line 1316: /DietPi/dietpi/func/dietpi-ramdisk: No such file or directory
```

--------------------------------------------------------------------

[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

Joulinar, 2020-05-07:
Hi,

many thanks for your report. I guess it's similar to #3509

Joulinar, 2020-05-07:
If possible, pls wait on the update because there some issues on the new version.

closed 2020-05-10 by MichaIng