MichaIng/DietPi #3514: Buster image serial problem

Vec7or, 2020-05-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | CORE=6, SUB=29, RC=2, GITBRANCH=master, GITOWNER=MichaIng
- Distro version | buster
- Kernel version | Linux DietPi 5.4.14-sunxi #rc1 SMP Sat Jan 25 15:19:01 CET 2020 armv7l GNU/Linux
- SBC model | NanoPi Neo
- Power supply used | 5V 2A
- SDcard used | SanDisk Micro SDHC I 16GB

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi? --> Yes
- On the stretch image with the exact same hardware the error is not present.

#### Steps to reproduce
1. Download DietPi_NanoPiNEO-ARMv7-Buster.img
2. Write the image to SD-Card
3. Opt Out of DietPi-survey
4. Adjust both the software password and user passwords to the same password
5. Disable serial console
6. Don't select any software
7. Select install
8. Have a UART-device connected to ttyS2
9. cat /dev/ttyS2

#### Expected behaviour
- cat should be able to connect to the UART-device

#### Actual behaviour
```
cat /dev/ttyS2
cat: /dev/ttyS2: Input/output error
```
In fact this error shows on all serial interfaces except ttyS0. On the stretch image ttyS0 - ttyS3 are available.

#### Extra details
- Worked like a charm on stretch image with same hardware.

MichaIng, 2020-05-07:
@Vec7or 
Many thanks for your report.

Let me have a look into the image. Major difference is that the new image is based on Armbian while the old one was based on FriendlyARM ones, which are not available anymore. Since Armbian uses a different (mainline) kernel, it is possible that some feature either are not available or need to be explicitly enabled int eh device tree.

Vec7or, 2020-05-07:
I have done a quick test and on first glance it looks good but i will be able to test this more thoroughly tomorrow.

Vec7or, 2020-05-08:
I have now tested the functionality of the UART-Device. I can send and also receive data over UART. Therefore the port seems to be working correctly.

Unfortunately with this new image based on Armbian WiringNP (which is used to control the pins of the NanoPi) is no longer working:
```
piBoardRev: Unable to determine board revision from /proc/cpuinfo
 -> Is not NanoPi based board. 
 ->  You may want to check:
 ->  http://www.lemaker.org/
open /sys/class/sunxi_info/sys_info failed.
```
I have extracted both mentioned files but am at a loss as to how i can inject the sys_info file into the /sys/class folder in order to test if the library works as expected afterwards.

@MichaIng I realize this problem might not be within the scope of Dietpi depending on how you look at it, nevertheless do you have an idea regarding this problem?

If you have no idea I think this issue can be closed since the root problem has been fixed.

MichaIng, 2020-05-08:
@Vec7or 
Looks like this WiringNP only supports the official FriendlyARM NanoPi images. Armbian kernel comes with different /proc and /sys content when it's about board info strings and such. Could you send a link to the exact source you got WiringNP from?

Vec7or, 2020-05-08:
I got it from: https://github.com/friendlyarm/WiringNP
If no easy solution is available i will have to take a look at the SCB detection code and will then hopefully be able to adjust it.

MichaIng, 2020-05-08:
@Vec7or 
I'll also have a look into it, let's see if we find something.

Vec7or, 2020-05-11:
I have created a modified Version of WiringNP which only supports the `NanoPi Neo` but works fine with the buster image [WiringNP](https://github.com/PlantWire/WiringNP). So in my opinion this issue may be closed.

MichaIng, 2020-05-11:
@Vec7or 
Great work. Can you give me a hint which essential part was required to change? I am pretty sure it is something about the board definitions [here](https://github.com/PlantWire/WiringNP/commit/d20ebf8e527aed06dd7e8af90d363d0c67630289#diff-cd23037256f145f040458d31596b720f) but my understanding about it is limited 😅. Probably we can extend support for other NanoPi boards with Armbian image and include it into DietPi-Software, like we support WiringPi for RPi.

Vec7or, 2020-05-12:
@MichaIng
Well what needs to change is [this](https://github.com/friendlyarm/WiringNP/blob/master/wiringPi/boardtype_friendlyelec.c) file. All functions within this file are dedicated to finding out what board the library is running on. Unfortunately the logic relies on files that are not present in the Armbian images. 

If there would be a simple file with the boardTypeId(7002 for NanoPi Neo) or something similar in the image it should be quite simple to adjust the code. The library would then work with all NanoPi boards which are both supported by the original library and supported by DietPi.

MichaIng, 2020-05-12:
@Vec7or 
Not sure if there is some meaningful info in `/proc/cpuinfo` or somewhere else in `/proc` or `/sys`. There is the `/etc/armbian-release` which contains board name at least. But since we aim to remove the Armbian package which provides this file for future releases, I would then better go with our own hardware identifier: `/etc/.dietpi_hw_model_identifier` => "60" = NanoPi NEO
But best would be to use info served by the kernel itself.

Vec7or, 2020-05-12:
@MichaIng 
I think the easiest solution would be to use `/etc/.dietpi_hw_model_identifier`. Do you have a list of the IDs? 
Of course kernel data would be more reliable but I have no Idea where in the Kernel I could find data identifying the board.

Maybe a new issue should be created? Since this discussion does not really correspond with the original issue.

MichaIng, 2020-05-12:
@Vec7or 
https://github.com/MichaIng/DietPi/blob/dev/dietpi/func/dietpi-obtain_hw_model

Only issue is that some SBCs with same SoC and manufacturer share the same ID which are listed as different devices in the [original file](https://github.com/friendlyarm/WiringNP/blob/master/wiringPi/boardtype_friendlyelec.c), e.g. NanoPi M3+T3+Fire3, M4+T4+NEO4 and such.

> Maybe a new issue should be created? Since this discussion does not really correspond with the original issue.

Ah yes that's true, I'll open one tomorrow.

MichaIng, 2021-05-18:
Referencing discussion in the other issue: https://github.com/MichaIng/DietPi/issues/4149#issuecomment-830688008

The problem is that the WiringNP code contains a lot of bugs (build warnings), is nearly unmaintained and works reliably only with FriendlyARM kernel, which is outdated. To add good support, we need to fork the project and add an own detection, as found above already. I don't think it makes much sense to start with this before we migrated all NanoPi images from Armbian to Debian kernel, so that we don't need to retest and in case rework WiringPi/NP with the new kernel.

MichaIng, 2021-09-24:
We decided to cleanup open issues a bit and close those which will obviously not be addressed soon. If anyone wants to start working on implementing this software, please feel free to to a start, and we'll help where we can.

Here you find instructions about how to implement a new software title to `dietpi-software`: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

The problem here really is that this WiringNP repo has a very bad quality code base and only works reliably on the official FriendlyARM images, while we use the much newer Armbian kernel as bases.

closed 2021-09-24 by MichaIng