MichaIng/DietPi #3515: Can't change locales. Locales reset after reboot

SergioRius, 2020-05-08:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.29.2
- Distro version | Buster
- Kernel version | Linux media 4.19.0-8-amd64 #1 SMP Debian 4.19.98-1+deb10u1 (2020-04-27) x86_64 GNU/Linux
- SBC model | Virtual Machine (x86_64)
- Power supply used | /none/
- SDcard used | /virtual/

#### Additional Information (if applicable)
Newly fresh virtual machine created in Esxi as per Dietpi Wiki instructions, with version 6.28
During first boot, it auto updates to 6.29.2
This same issue is present in my other 6.28 VMs

#### Steps to reproduce
Start the VM for the first time.
Set the language locales to es_ES
Mark software to install, or continue minimal and install them manually after reboot.

#### Actual behaviour
Installing software, like docker or samba shows:
```
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = "en_GB.UTF-8",
        LANG = (unset)
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
```
```
# locale
LANG=
LANGUAGE=
LC_CTYPE="es_ES.UTF-8"
LC_NUMERIC="es_ES.UTF-8"
LC_TIME="es_ES.UTF-8"
LC_COLLATE="es_ES.UTF-8"
LC_MONETARY="es_ES.UTF-8"
LC_MESSAGES="es_ES.UTF-8"
LC_PAPER="es_ES.UTF-8"
LC_NAME="es_ES.UTF-8"
LC_ADDRESS="es_ES.UTF-8"
LC_TELEPHONE="es_ES.UTF-8"
LC_MEASUREMENT="es_ES.UTF-8"
LC_IDENTIFICATION="es_ES.UTF-8"
LC_ALL=es_ES.UTF-8
```
```
# cat /etc/locale.gen
es_ES.UTF-8 UTF-8
en_GB.UTF-8 UTF-8
```
```
# cat /etc/default/locale | grep -m1 '^LANG=' | sed 's/.*=//'
<empty>
```
Issuing the following commands does not have any effect
```
export LANGUAGE=es_ES.UTF-8
export LANG=es_ES.UTF-8
export LC_ALL=es_ES.UTF-8
locale-gen es_ES.UTF-8
dpkg-reconfigure locales
```

Creating a /etc/environment file with the following contents, does not have any effect:
```
# cat /etc/environment 
LC_ALL=es_ES.UTF-8
LANG=es_ES.UTF-8
```

#### Expected behaviour
locale Should show the locale choice correctly setup
Installing software should not complain about missing locales.

MichaIng, 2020-05-08:
@SergioRius 
Many thanks for your report.

`/etc/locale.gen` looks like it should, containing your chosen locale and en_GB.UTF-8 which was used by DietPi scripts until v6.29. Now C.UTF-8 is used, when re-choosing locales via dietpi-config, en_GB.UTF-8 is not added anymore, as long as not chosen.

To know if compiling `es_ES.UTF-8` has succeeded, can you paste the output of: `locale -a`

`LANG` and `LANGUAGE` are indeed not set by DietPi, since those are fallbacks only and `LC_ALL` overrides everything anyway.
Instead check `cat /etc/default/locale` which should contain `LC_ALL=es_ES.UTF-8` only.

`/etc/environment` basically is not used on Debian anymore, especially not for setting locales. This must be done in `/etc/default/locale` which requires them to be compiled successfully which seems to have been failed in your case.

MichaIng, 2020-06-01:
Marking as closed due to missing reply. Feel free to reopen if the issue is still present.

closed 2020-06-01 by MichaIng

bamyasi, 2021-05-13:
Same problem here, after updating to DietPi v7.1.2. I am also getting the following error when trying to change locale via dietpi-config:

`/boot/dietpi/func/dietpi-set_software: line 79: debconf-set-selection: command not found`

However, the locale change seems to work but perl is still complaining:

```
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
```

MichaIng, 2021-05-13:
The first error was a typo in the script indeed: https://github.com/MichaIng/DietPi/commit/396434d00925329be1e8e3ec8927bc19517cbbc8

And it leads to the case the the new locale is generated but not set as new system default. You can fix it like that:
```
update-locale 'LANG=en_US.UTF-8'
```
if en_US was your chosen language.

To fix it in the code:
```
sed -i 's/debconf-set-selection /debconf-set-selections /' /boot/dietpi/func/dietpi-set_software
```

MichaIng, 2021-05-13:
Changelog: https://github.com/MichaIng/DietPi/commit/7b7c9e8a84e0995b4313b79d9405ee99363ef40b
Thanks for reporting 👍.

bamyasi, 2021-05-13:
Cool, this has fixed the errors for me. Thank you, @MichaIng !