MichaIng/DietPi #3516: Device Trees

Zingisa1, 2020-05-08:
Good day,

Thanks for this amazing project, it's really marvelous.

Please include dts files from CoreElec project into your source code and be automatically compiled on running PREP_SYSTEM_FOR_DIETPI.sh script so to have dtb files stored on /boot/dietpi-dtb directory.

Kind regards,

MichaIng, 2020-05-08:
@Zingisa1 
Many thanks for your suggestion.

What is the advantage of using the CoreELEC device tree files instead of the regular ones provided by manufacturer or 3rd party (Armbian) repositories?

Zingisa1, 2020-05-08:
@MichaIng 

Thank you for your prompt reply,

I'm using Tanix TX92 TV Box, only CoreElec's "gxm_q201_2g_1gbit.dtb" is able to identify my machine's wifi module. 

I believe armbian's lack of wifi support is everybody's nightmare and mine too.

Kind regards,

MichaIng, 2020-05-08:
@Zingisa1 
Probably we can suggest Armbian to fix/enhance their device trees by comparing with CoreELEC ones: https://github.com/CoreELEC/linux-amlogic/blob/4e04b70860643864496cc76ca2a39d914ba01fda/arch/arm64/boot/dts/amlogic/gxm_q201_2g.dts#L157

Zingisa1, 2020-05-08:
@MichaIng Your help in such regard will be highly appreciated.

MichaIng, 2020-05-10:
@Zingisa1 
Unfortunately Armbian devs do not like DietPi too much, hence it wouldn't be a push but a bumper if I do this required. It makes sense to do a direct comparison of official Armbian image on a specific device within and without switching the device tree to CoreELEC ones. Then report this to [Armbian forums](https://forum.armbian.com/) without mentioning DietPi. This should lead to best chance to those guys having a look into.

Just as a note from our side:
- I'll definitely not add a bunch of additional device trees to any of our images so that all possible missing or broken board features can be enabled/fixed with one of them. But it is good to know that experienced users can fix or enable features probably by manually enabling them. However the manual download is then the smallest effort anyway.
- Especially we cannot be sure if the CoreELEC device tree does not lead to other regressions on other end, so that is always a very individual and device-specific step that might fix the ones system and might break the other ones.
- Generally, I'll always suggest to have bugs and limitations reported to the maintainers of the repositories we use, instead of hacking own fixes/changes inside, so all users can benefit and it is done with full review and support and of course the required knowledge to rule out regressions.

MichaIng, 2021-06-02:
I mark this as closed. Custom device trees is out of our scope. We use Debian or other 3rd party kernel and device tree packages and it will remain like this as long as there does not show up a long-term contributor that is able and motivated to take care of it. But when there are requests regarding the device tree, we are happy to receive them and will forward them to upstream developers, either as request for the main device tree, or as additional overlay.

closed 2021-06-02 by MichaIng