MichaIng/DietPi #3518: Add support for Orchard Audio PecanPi DAC

orchardaudio, 2020-05-09:
#### Is your feature request related to a problem? Please describe:
Asus recently added support for the PecanPi DAC starting in Stretch V2.1.11-20200310.
More info on the PecanPi can be found here:
www.orchardaudio.com/pecanpi-dac

#### Describe the solution you'd like:
Please add support for this DAC in the tinker board builds. I have included the dtbo file in this request. To enable the DAC the following needs to be added to the hw_intf.conf file:
`intf:dtoverlay=rpi-dac`

#### Describe alternatives you've considered:
N/A

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/+160

[rpi-dac.zip](https://github.com/MichaIng/DietPi/files/4603368/rpi-dac.zip)

MichaIng, 2020-05-10:
@orchardaudio 
Many thanks for your request and providing the dtoverlay.

Note that our new image is based on Armbian, however the overlay can be added as simple by giving the file some prefix and add the overlay to `/boot/armbianEnv.txt`.

orchardaudio, 2020-05-10:
> @orchardaudio
> Many thanks for your request and providing the dtoverlay.
> 
> Note that our new image is based on Armbian, however the overlay can be added as simple by giving the file some prefix and add the overlay to `/boot/armbianEnv.txt`.

I thought it was not just a matter of adding the device in the txt file but the kernel also needs to support the overlay, maybe I am wrong.

I have also submitted a request for this in the Armbian forum here:
https://forum.armbian.com/topic/13998-need-support-for-the-pecanpi-dac/?tab=comments#comment-101248

orchardaudio, 2020-05-10:
I am not a software or Linux person and don't have a good understanding of how all this stuff is compliled. I will try to provide as much help as I can.

MichaIng, 2020-05-10:
@orchardaudio 
Basically the overlay needs to match the full device tree of course. But as long as this is the case and the kernel supports device trees at all, AFAIK it should work. In this case the only thing I am not sure about is if Armbian kernel/bootloader uses `hw_intf.conf` or not. And as said you need to add a certain prefix to the .dtbo file. See `/boot/boot.cmd` about which and how those overlays are added.

orchardaudio, 2020-05-10:
I am sorry I don't know what you are asking me to do. Is there somebody here who can take care of it?

MichaIng, 2020-05-10:
@orchardaudio 
As you said
> I thought it was not just a matter of adding the device in the txt file but the kernel also needs to support the overlay, maybe I am wrong.

I though it did not work this way:
> To enable the DAC the following needs to be added to the hw_intf.conf file:
intf:dtoverlay=rpi-dac

?

I'll have a deeper look into it during v6.31 development phase.

orchardaudio, 2020-05-13:
Ok. Thanks.

orchardaudio, 2020-05-25:
I found this, it may be useful:
https://github.com/TinkerBoard/debian_kernel/commit/b6cf89fc400eea83cab24f88f8d5e969a5e417f7

MichaIng, 2020-05-25:
Indeed, however the "bad" news is that the overlay itself is not be sufficient but a kernel driver component needs to be compiled as well. So we need to check whether Armbian did or aims to port this as well or not, or otherwise if the same is true for upstream mainline Linux.

orchardaudio, 2020-08-27:
Any info on when this can be added to dietpi for tinkerboard?

MichaIng, 2020-08-27:
Since we do no kernel development/compiling and it requires a kernel patch, we need to wait for Armbian for that. Did you get any response from `le51` in this regards? Probably the TinkerOS commit on their forums will as well help to reanimate it. Other benefit is that Armbian might be able to integrate it into their base mainline kernel so that it will be available for more boards. I'm sorry that I cannot help much with that topic.