MichaIng/DietPi #3519: GMediaRender stopped working after update 6.29.2

fnsnyc, 2020-05-09:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.29.2
- Kernel version | Linux DietPi 4.19.
- SBC model |  RPi4

#### Steps to reproduce

Need to correct the file /etc/systemd/system/gmrender.service 

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2020-05-09:
Hi,

many thanks for your report. Looks like on of the issue is the missing symlink from `/DietPi > /boot` 

`mawk: cannot open /DietPi/dietpi/.network (No such file or directory)`

could be fixed like this

`ln -s /DietPi /boot`

Next to that `/run/dietpi/.network` is located on new location on new release 6.29.2. (as expected)

@MichaIng 
I guess `gmrender.service` would need to be adjusted

MichaIng, 2020-05-10:
@fnsnyc @Joulinar 
Many thanks for reporting this issue.

Interesting that the `gmrender.service` contains the correct location already (it is indeed adjusted during update) but the error messages shows it checking the old location. I guess there is simply some `systemctl daemon-reload` missing. Actually I thought that this is done at the end of every DietPi update, let me see...

@fnsnyc 
Just to verify, please run:
```
systemctl daemon-reload
systemctl restart gmrender
systemctl status gmrender
```

fnsnyc, 2020-05-10:
@MichaIng 

Same error.

MichaIng, 2020-05-10:
Edit, this is actually indeed done:
- Service file adjustment: https://github.com/MichaIng/DietPi/blob/master/dietpi/patch_file#L2377
- systemd reload: https://github.com/MichaIng/DietPi/blob/master/dietpi/patch_file#L2425

@fnsync
If the above steps do indeed not solve it, I see that you have edited the service file before, as there is a `/etc/systemd/system/gmrender.service.ddietpi-services_edit.conf`. Could you please paste the following to show the finally effective service code:
```
systemctl cat gmrender
```

fnsnyc, 2020-05-10:
```
dietpi@DietPi:~$ sudo systemctl cat gmrender
# /etc/systemd/system/gmrender.service
[Unit]
Description=GMediaRender (DietPi)
Wants=network-online.target
After=network-online.target dietpi-boot.service

[Service]
User=gmrender
ExecStartPre=+/bin/dash -c 'systemctl set-environment ACTIVE_IP=$(mawk 'NR==4' /run/dietpi/.network)'
ExecStart=/usr/local/bin/gmediarender -u '941892a6-f8c7-42e8-b8eb-b407257a8bf0' -f 'DietPi' --gstout-audiosink=alsasink --gstout-audiodevice=sysdefault --gstout-initial-volume-db=0 --log
file=stdout -I "$ACTIVE_IP"
#ExecStart=/usr/local/bin/gmediarender -u '941892a6-f8c7-42e8-b8eb-b407257a8bf0' -f 'DietPi' --gstout-audiosink=alsasink --gstout-audiodevice=sysdefault --gstout-initial-volume-db=0 --lo
gfile=stdout -I "192.168.4.107"

[Install]
WantedBy=multi-user.target

# /etc/systemd/system/gmrender.service.d/dietpi-process_tool.conf
# WARNING: Do not manually edit this file, use "dietpi-services" to adjust values!
[Service]
Nice=-10
IOSchedulingPriority=0

# /etc/systemd/system/gmrender.service.d/dietpi-services_edit.conf
[Unit]
#Description=GMediaRender (DietPi)
#Wants=network-online.target
#After=network-online.target dietpi-boot.service

[Service]
#User=gmrender
#ExecStartPre=+/bin/dash -c 'systemctl set-environment ACTIVE_IP=$(mawk 'NR==4' /DietPi/dietpi/.network)'
#ExecStart=/usr/local/bin/gmediarender -u '941892a6-f8c7-42e8-b8eb-b407257a8bf0' -f 'DietPi' --gstout-audiosink=alsasink --gstout-audiodevice=sysdefault --gstout-initial-volume-db=0 --lo
gfile=stdout -I "$ACTIVE_IP"

ExecStartPre=
ExecStartPre=/DietPi/dietpi/func/obtain_network_details
ExecStartPre=+/bin/dash -c 'systemctl set-environment ACTIVE_IP=$(mawk 'NR==4' /DietPi/dietpi/.network)'

[Install]
#WantedBy=multi-user.target
dietpi@DietPi:~$ 
``

MichaIng, 2020-05-10:
@fnsnyc 
Ah there it is. I remember this being a fix for another issue if network interface bring-up takes very long. Let's keep the fix in place, so do:
```
sed -Ei 's@/(DietPi|boot)/dietpi/\.network@/run/dietpi/.network@g' /etc/systemd/system/gmrender.service.d/dietpi-services_edit.conf
systemctl daemon-reload
systemctl restart gmrender
systemctl status gmrender
```

fnsnyc, 2020-05-10:
voilà! now it's working!! Thanks so much!

Joulinar, 2020-05-11:
@MichaIng 
I guess this would need to be implemented by default. I did a test installation on 6.30 and it fails as network interface bring-up takes longer. See attached picture.

```
root@DietPi3:~# systemctl cat gmrender.service
# /etc/systemd/system/gmrender.service
[Unit]
Description=GMediaRender (DietPi)
Documentation=https://github.com/hzeller/gmrender-resurrect/blob/master/INSTALL.md#commandline-options
Wants=network-online.target
After=network-online.target sound.target dietpi-boot.service

[Service]
User=gmrender
ExecStartPre=+/bin/dash -c 'systemctl set-environment ACTIVE_IP=$(mawk 'NR==4' /run/dietpi/.network)'
ExecStart=/usr/local/bin/gmediarender -u 'be44fcef-43f6-4b60-85f1-9479df37c41a' -f 'DietPi3' --gstout-audiosink=alsasink --gstout-audiodevice=default --logfile=stdout -I "$ACTIVE_IP"

[Install]
WantedBy=multi-user.target
root@DietPi3:~#
```
LAN IP is not detected right after reboot, therefore GMediaRender  is going to fail.
[image]

MichaIng, 2020-05-11:
@Joulinar 
Agree. I did another step to delay network data estimation until network time sync has proven internet connectivity (in most cases): https://github.com/MichaIng/DietPi/commit/7fb8b09e0a3161289368b4a09142c97691304e08
Will also add the info refresh to all service starts which read the info file, so also on service restarts, when users e.g. connect to a different WiFi, or DHCP changed the IP, info will be also up-to-date.

MichaIng, 2020-05-11:
Done, as well for WireGuard: https://github.com/MichaIng/DietPi/commit/e5fc3bb195ffa99d0116ab741222803b7b9df327
Changelog: https://github.com/MichaIng/DietPi/commit/9633e7127caae400e9255cb04d73f0138d30b8d9

closed 2020-06-30 by MichaIng