MichaIng/DietPi #3520: Welcome screen on NanoPi NEO2 (aarch64)

fnsnyc, 2020-05-09:
### Creating a bug report/issue

#### Required Information
- DietPi version | `6.29.2`
- Distro version | `stretch`
- Kernel version | `Linux DietPi-HMA 4.14.0 #82 SMP Fri Dec 8 14:33:14 CST 2017 aarch64 GNU/Linux`
- SBC model | `NanoPi NEO2 (aarch64)`

#### Steps to reproduce

When I log on ... below is the welcome screen.
It's not capturing the IP address. Weird.
IPv6 is off either turning on.. the same behavior.

```
 ─────────────────────────────────────────────────────
 DietPi v6.29.2 : 10:04 - sáb 09-05-2020
 ─────────────────────────────────────────────────────
 - Device model : NanoPi NEO2 (aarch64)
 - CPU temp : 25'C : 77'F (Who put me in the freezer!)
 - LAN IP : Use dietpi-config to setup a connection (eth0)
/boot/dietpi/func/dietpi-banner: linha 222: /run/dietpi/.dietpi_motd: Permissão negada
 ─────────────────────────────────────────────────────

```

fnsnyc, 2020-05-09:
BTW, it's in portuguese. Linha = Line and Permissão negada is Permission denied :-)

Joulinar, 2020-05-09:
Hi,

many thaks for your report. Yes that's a know issue,. Pls can you login as user `root`. This should fix it.

fnsnyc, 2020-05-09:
Great. Fixed. Thanks.

closed 2020-05-09 by fnsnyc