MichaIng/DietPi #3521: contents of hash.txt should be available for the .7z file

flavio-fernandes, 2020-05-09:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
It feels a little too late to be checking SHA256 on the image file

#### Describe the solution you'd like:
Co-located with the download link, it would be awesome to have the sha256 for the .7z file, so we are certain that it has not been tampered with even before we uncompress it.

#### Describe alternatives you've considered:
A good reference on what I'm proposing can be seen on the download links for [Raspbian](https://www.raspberrypi.org/downloads/raspbian/) or [Ubuntu](https://ubuntu.com/download/raspberry-pi/thank-you?version=20.04&architecture=armhf+raspi), under verify your download.

#### Additional context
By having the checksum co-located with the link, we can make it harder for ill intended people to tamper with the images.

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2020-05-10:
@flavio-fernandes 
Many thanks for your request.

Hmm it is very common to include hashes/checksums into the archive, to the final image, which must be original and not corrupted by failure download AND faulty unpacking. However of course if the archive itself is corrupt already, one could skip unpacking in the first place. Not sure how to do best, place a hash file for a specific algorithm into our download directory, file meaningful file ending, to be easily and automated compared, or put the same style hash file with MD5+SHA1+SHA256 contained, like the one in the archive, into the download dir, with archive file name + ".hash" file ending?

flavio-fernandes, 2020-05-10:
> @flavio-fernandes
> Many thanks for your request.
>
> Hmm it is very common to include hashes/checksums into the archive, to the final image, which must be original and not corrupted by failure download [...]

@MichaIng Thanks for following up!

I think that the checksum as it is now is fine for detecting issues with corruption during download.

The category I'm talking about here is more geared towards some ill-intended person deliberately messing with the compressed file and carefully rebuilding the image. So, maybe it is best to separate these two cases.

So, to make the hacker's life harder, I would think that having a checksum published right next to the compressed file is a good thing. Different sites do it differently, but at the end of the day -- assuming the website is not compromised -- having a checksum to compare against the downloaded file would be awesome.

-- flaviof

yumiris, 2020-05-11:
Love those kinds of threads - hope it's fine to join in!

Just a thought: hashes are designed for proving integrity, but aren't inherently reliable for proving **authenticity**. If a website is compromised to the point of altering the download, then nothing should stop an attacker from altering the hash file itself.

Signatures (a la GPG) would be more suitable for this sort of scenario, since a GPG signature aims to prove authenticity (and implicitly, integrity). Mind you, authenticity flies out the window if the GPG key is stolen, though the practical risk of that depends on your adversary's capabilities.

---

As an example: the Hyper-V images are published with a GPG signature for their respective archives. Additionally, the commits representing the releases are also signed with GPG:

[image]

If GitHub were to be compromised, the attacker would also need to compromise my own key to impersonate and release tampered releases on my behalf.

MichaIng, 2020-05-11:
@yumiris 
> hope it's fine to join in!

More than welcome!

> hashes are designed for proving integrity, but aren't inherently reliable for proving authenticity

Indeed, these hashes placed beside or inside the download are ONLY meant to check for corruptions through download or unpacking, before flashing it.

> The category I'm talking about here is more geared towards some ill-intended person deliberately messing with the compressed file and carefully rebuilding the image. So, maybe it is best to separate these two cases.

This is indeed another request (though we can fulfil both together), since placing hashes/checksums onto the same server where the download is placed, cannot guarantee authenticity. An attacker can modify both, the file and the hashes, without much more effort. Note that due to HTTPS, in theory it is impossible that you get a file from a wrong source, e.g. an attacker/MITM cheats you out a faulity download, or presents you a fake website, when using the official domain.

So we basically talk about us creating a key to sign all downloads and enable users to verify all downloads against our public key. Luckily this is on the roadmap anyway since I plan to setup a DietPi APT repository, which then should be fully signed in the first place. The same key pair can then be used for all other downloads outside of the repo as well. Should be easy then to not only flash the signatures besides the downloads but also add them to the website next/below the download link.

muxnulljack, 2020-08-07:
Just wanted to let you know that I was interested in deploying your project on my Pi as an alternative to Raspbian because it's much better suited for my use-case, but I will not be using it because you do not offer signed images or shas.

> So we basically talk about us creating a key to sign all downloads and enable users to verify all downloads against our public key. Luckily this is on the roadmap anyway since I plan to setup a [...]

+1, you should do all of this.

Joulinar, 2020-08-07:
Well feel free to download and install Raspberry OS lite and use DietPi PREP script to transform it into DietPi. Than you don't need to use DietPi images and will get same result.

muxnulljack, 2020-08-07:
@Joulinar or I can go use another distribution with signed shasums and not bother deal with a workaround, which is exactly what I'm going to do? Is there any reason **not** to do what's described above?

Joulinar, 2020-08-07:
Sorry I don't won't force you to use DietPi. Feel free to use whatever you want. I just offered a way to reach a goal on short notice and not wait on some feature update. DietPi is developed and maintained by a single person. Therefore not everything could be done right away. And others might take some more time to be implemented. But as I said, you are free to use whatever you want.

MichaIng, 2020-08-07:
@muxnulljack 
The download archives contain MD5 checksum, SHA1 and SHA256 hashes for the contained image, which should in combination with strict HTTPS serve for both, integrity and authenticity.

The topic here is only to additionally place hashes onto the download pages, which is more a convenience topic instead of adding any security... Yeah okay you can additionally check the archive before extracting it instead of the contained image only.

As this is hence minor, we simply didn't find the time and mood to add all hashes to the website, especially since this means that those need to be updated on every new image upload, which practically requires some automation. But as the website goes through some rework anyway, I think we'll do that soon.

CyberPunkInterface, 2021-05-15:
@Joulinar 
>Well feel free to download and install Raspberry OS lite and use DietPi PREP script to transform it into DietPi. Than you don't >need to use DietPi images and will get same result.

Wheres The "DietPi PREP script" ?

ravenclaw900, 2021-05-15:
You can download it at https://raw.githubusercontent.com/MichaIng/DietPi/master/PREP_SYSTEM_FOR_DIETPI.sh.

CyberPunkInterface, 2021-05-15:
@ravenclaw900 Thanks Appreciate it.

Joulinar, 2021-05-16:
Have a look to our online docs for a guide how to use the script https://dietpi.com/docs/hardware/#make-your-own-distribution