MichaIng/DietPi #3522: [FAILED] DietPi-Update

carlosspre, 2020-05-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2020-05-10:
Hi,

Many thanks on your report. Pls wait with an update to DietPi v6.29.2 independent on your issue.

carlosspre, 2020-05-10:
I can't install any programs on my dietpi either ..... but in a new installation of dietpi it's all normal

Joulinar, 2020-05-10:
Let's close this issue as you opened another with same content

Joulinar, 2020-05-10:
Continuing in #3523

closed 2020-05-10 by Joulinar

carlosspre, 2020-11-22:
root@NetPlexPi4:~# apt upgrade

E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
root@NetPlexPi4:~#

Joulinar, 2020-11-22:
Did you tried running `dpkg --configure -a` as suggested by the error message?