MichaIng/DietPi #3523: dietpi v6.28.0 samba erro install and any other program does not install because

carlosspre, 2020-05-10:
#### Details:
- Date           | Sun 10 May 15:44:14 BST 2020
- Bug report     | d7a220ef-87b4-40aa-85fb-dba9d86db73f
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux NetPlexPi4 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
(Reading database ... 44940 files and directories currently installed.)
Removing rsync (3.1.3-6) ...
/usr/bin/deb-systemd-helper: 1: /usr/bin/deb-systemd-helper: shape:EventCategoriesMapList,: not found
/usr/bin/deb-systemd-helper: 2: /usr/bin/deb-systemd-helper: documentation:<p>A list of EventCategoriesMap data types.</p>: not found
/usr/bin/deb-systemd-helper: 3: /usr/bin/deb-systemd-helper: Syntax error: "}" unexpected
dpkg: error processing package rsync (--remove):
 installed rsync package post-removal script subprocess returned error exit status 2
Errors were encountered while processing:
 rsync
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2020-05-10:
@carlosspre 
Many thanks for your report.

Looks like `/usr/bin/deb-systemd-helper` is corrupt. Could you please run and paste the content of `dmesg -l err,crit,alert,emerg` to check for file system I/O errors.

For a quick fix the following should work:
```
cd /tmp
wget http://raspbian.raspberrypi.org/raspbian/pool/main/i/init-system-helpers/init-system-helpers_1.56+nmu1_all.deb
dpkg -i init-system-helpers_1.56+nmu1_all.deb
rm init-system-helpers_1.56+nmu1_all.deb
apt-get -f install
```

carlosspre, 2020-05-10:
resolved SAMBA MORE UPDATE OF DIETEPI NOT YET

apt update OK
apt upgrade PROBLEM

E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.

how do I resolve this issue from updating dietpi v6.28.0 to v6.29.2.

Joulinar, 2020-05-10:
Did you run `dpkg --configure -a` as show in your error message?

carlosspre, 2020-05-10:
problem solved my dietpi was used ..... thanks for the help you and good fuck

Joulinar, 2020-05-10:
So all fine now?

carlosspre, 2020-05-13:
yes it's okay now thanks to you friend

Joulinar, 2020-05-13:
Many thanks for confirmation. I'm going to close the issue. Feel free to reopen if needed.

closed 2020-05-13 by Joulinar