MichaIng/DietPi #3529: Sonarr - Access to the path is denied.

Chublo, 2020-05-11:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=30
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'` 
- Distro version | `buster` 
- Kernel version | `Linux DietPi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux`
- SBC model | `RPi 2 Model B (armv7l)` 
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (Sonarr)
- Was the software title installed freshly or updated/migrated? Freshly installed
- Can this issue be replicated on a fresh installation of DietPi? Unknown

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Sonarr has stopped importing completed downloads to my CIFS mounted network share

#### Expected behaviour

- ...It should import

#### Actual behaviour

- ...It's not

#### Extra details

- ... Everything was working fine before 6.29.2, suffered the FSTAB issues with that version, have now upgraded to 6.30.0 but now have noticed that Sonarr is not able to import any files. 

The message I get is:
`Couldn't import episode /mnt/Downloads/Completed/Filename.mkv: Access to the path is denied.`

MichaIng, 2020-05-11:
@Chublo 
Many thanks for your report.

Could you please the the output of the following commands to verify permissions:
```
ls -al /mnt/Downloads{,/Completed}
```
Of course mask file names, only thing I want to check is if final dir and all parent dirs are world read+executable and completed downloads are readable (and writable, if Sonarr should remove important downloads) for the sonarr user or dietpi group.

And to clarify:
- `/mnt/Downloads/Completed/` is where the downloader backend moves completed downloads to? This is a local drive, right?
- Sonarr shall import the completed downloads to a CIFS share, hence copy or move them to a CIFS mount?

Importing into a CIFS share should indeed fail because of this: https://github.com/MichaIng/DietPi/issues/3179
A mono package upgrade should hence be the culprit in your case, which broke internally copying/creating files on file systems which do not support UNIX permissions, hence where chmod/chown cannot be used. In case of CIFS the permissions are fixed, based on mount options, so likely it is affected by this bug as well.
As you can read, a workaround has been done for Sonarr v3, but it is still in beta and hence we still install Sonarr v2 within DietPi.

To verify that this known issue is the reason you import fails, you could simply to a test import into native DietPi download dir `/mnt/dietpi_userdata/downloads`. Also verify that the CIFS mount has 77X/66X modes and "dietpi" group, so Sonarr has write permissions for sure. However since it worked before, I guess this is the case. Last but not least you can install Sonarr v3 manually and see if this fixes the issue: https://sonarr.tv/#downloads-v3-linux
Make a backup of Sonarr database+settings: `cp -a /mnt/dietpi_userdata/sonarr{,_bak}`

Chublo, 2020-05-12:
Here is a sample of 
`ls -al /mnt/Downloads{,/Completed}'

`drwxrwx--- 2 dietpi dietpi     0 May 11 22:19  .`
`drwxrwx--- 2 dietpi dietpi     0 May 11 16:00  ..'
`drwxrwx--- 2 dietpi dietpi     0 May  7 13:40  Filename'
`drwxrwx--- 2 dietpi dietpi     0 May  7 02:43  Filename'
'-rwxrwx--- 1 dietpi dietpi  4096 Dec 16 10:39  ._.DS_Store'
'-rwxrwx--- 1 dietpi dietpi  6148 Dec 16 10:39  .DS_Store'

But sonarr should be moving to my Media Folder. Both Downloads and Media are the cifs mount.

`'root@DietPi:~# ls -al /mnt/Media/TV'
`total 0
`'drwxrwx--- 2 dietpi dietpi 0 May 11 07:47  .`
`'drwxrwx--- 2 dietpi dietpi 0 May 11 22:19  ..`
`'drwxrwx--- 2 dietpi dietpi 0 Aug 17  2019  A****`
`'drwxrwx--- 2 dietpi dietpi 0 Jan 20 06:25 A****`
`'drwxrwx--- 2 dietpi dietpi 0 Aug 17  2019 B****`
'`drwxrwx--- 2 dietpi dietpi 0 Dec 17 18:40 B****`
`'drwxrwx--- 2 dietpi dietpi 0 Apr  2 05:55 B****`

These are the lines from the FSTAB
`//192.168.86.34/Media /mnt/Media cifs username=*****,password=******,iocharset=utf8,uid=dietpi,gid=dietpi,file_mode=0770,dir_mode=0770,vers=3.0,_netdev,nofail,noauto,x-systemd.automount
`//192.168.86.34/Downloads /mnt/Downloads cifs `username=*****,password=******,iocharset=utf8,uid=dietpi,gid=dietpi,file_mode=0770,dir_mode=0770,vers=3.0,_netdev,nofail,noauto,x-systemd.automount`

Assuming all looks good here and nothing to fix, I will try V3. 

Thoughts?

Chublo, 2020-05-12:
Just an update: I updated to V3 and it's working! Thanks again for the advice. 
I'd be interested to know if there is anything odd in the above though!

MichaIng, 2020-05-12:
@Chublo 
Okay, then I think we can be quite sure that it is basically this bug: https://github.com/MichaIng/DietPi/issues/3179
Actually obvious that it affects CIFS as well, but I didn't think about that so far.

MichaIng, 2021-04-25:
This has been solved with the migration to Sonarr v3 (and Radarr v3) which have a workaround implemented until that issue is solved .NET core wise.

closed 2021-04-25 by MichaIng