MichaIng/DietPi #3530: DietPi-Software | rTorrent: Set dht.mode.set = auto

Joulinar, 2020-05-11:
Hi @MichaIng 

it might make sense to set `dht.mode.set = auto` on `/mnt/dietpi_userdata/rtorrent/.rtorrent.rc` by default, based on following forum post. 

https://dietpi.com/phpbb/viewtopic.php?p=24532#p24532

Otherwise rTorrent will not be able to start. At least on v6.30 Buster. 

On a fresh Stretch image, something seems to be wrong with `Screen` as it doesn't work at all. Therefore I'm not able to verify `dht.mode.set` setting atm.

MichaIng, 2020-05-12:
Thanks for forwarding.

Screen is not required anymore since Buster, where the rTorrent version has an internal daemon mode. I'll try to replicate on Stretch.

Okay so DHT seems to require much allocated/mapped memory. Currently we use `pieces.memory.max.set = $RAM_TOTAL` while `RAM_TOTAL` is the total physical memory of the device:
```
2020-05-12 11:21:49 root@micha:~# free -m | mawk '/^Mem:/{print $2;exit}'
975
2020-05-12 11:21:56 root@micha:~# free -m
              total        used        free      shared  buff/cache   available
Mem:            975         325         203          66         447         518
Swap:          4095         175        3920
```
...ayayay now I see the bug, there is an `M` missing, not sure if without it the default is KiB. It works still fine on testing VM, but probably because they have 2 GiB memory and with only 1-2 torrents added for testing.

PR up: #3531

bbsixzz, 2020-05-13:
I see you got lost in the rTorrent syntax jungle @MichaIng 

Welcome to my world ;]

closed 2020-06-30 by MichaIng