MichaIng/DietPi #3533: I installed in the dietpi a SSD 1T and a HD 1T how do I access them on my network in Windows ... access by Samba but only appears the micro sd where is the dietpi ... how do I

carlosspre, 2020-05-13:
### Creating a software request

#### Formal software information
- Software title | 
- Short description | 
- Official URL (if available) | 
- GitHub URL (if available) | 
- (Official) install documentation | 

#### Are there similar/alternative software titles available with DietPi-Software?

- ...

#### What makes your requested software better than the above solutions, if available?

- ...

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- ...

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. ...

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- ...

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/

Joulinar, 2020-05-13:
Hi,

many thanks for your question. Basically you would have 2 ways to get access to your HDD using Samba.

1. Create additional Samba shares pointing to your HDD. This can be done by adding new shares into `/etc/samba/smb.conf `

2. mount your HDD inside `/mnt/dietpi_userdata/`

small user guide from the forum https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=10#p56

carlosspre, 2020-05-14:
erro unistall samba
#### Details:
- Date           | Thu May 14 01:01:02 BST 2020
- DietPi version | v6.30.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux NetPlexPi4 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Setting up samba-common-bin (2:4.9.5+dfsg-5+deb10u1+rpi1) ...
Checking smb.conf with testparm
Load smb config files from /etc/samba/smb.conf
Error loading services.
dpkg: error processing package samba-common-bin (--configure):
 installed samba-common-bin package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of samba:
 samba depends on samba-common-bin (= 2:4.9.5+dfsg-5+deb10u1+rpi1); however:
  Package samba-common-bin is not configured yet.

dpkg: error processing package samba (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 samba-common-bin
 samba
E: Sub-process /usr/bin/dpkg returned an error code (1)

carlosspre, 2020-05-14:
I want to give up and install samba again this is giving error in giving up

carlosspre, 2020-05-14:
dpkg: error processing package samba (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 samba-common-bin
 samba
E: Sub-process /usr/bin/dpkg returned an error code (1)

Joulinar, 2020-05-14:
What exactly you are trying to do? Did you try to install SAMBA or are you trying to remove it?

MichaIng, 2020-05-14:
@carlosspre 
Can you run the following to isolate the error a bid:
```
dpkg --configure samba-common-bin
```

closed 2020-09-27 by MichaIng