MichaIng/DietPi #3534: Update error message

arroznegro, 2020-05-13:
### Creating a bug report/issue
#### Details:
- Date           | Wed May 13 11:26:11 EDT 2020
- DietPi version | v6.30.0 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi-Bottom 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -q update
- Exit code      | 134
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
```

MichaIng, 2020-05-13:
@arroznegro 
Many thanks for your report.

Strange as there is no error message, like it just stuck at "Reading package lists...". I'll check what exit code `134` means.

MichaIng, 2020-05-13:
Very strange, I could not find anything about apt-get exit code 134.

Please try to manually run the command from console: `apt-get update`

MichaIng, 2020-06-01:
Marking as closed due to missing reply. Feel free to reopen if issue is still present.

closed 2020-06-01 by MichaIng