MichaIng/DietPi #3535: tried to update dietpi and this happened.....HELP.

blindtrust, 2020-05-13:
#### Details:
- Date           | Wed 13 May 17:03:21 BST 2020
- Bug report     | e228a8e5-9f96-49b6-a40f-e420fca823ab
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Preconfiguring packages ...
(Reading database ... 25084 files and directories currently installed.)
Preparing to unpack .../base-files_10.3+rpi1+deb10u4_armhf.deb ...
Unpacking base-files (10.3+rpi1+deb10u4) over (10.3+rpi1+deb10u2) ...
dpkg: error processing archive /var/cache/apt/archives/base-files_10.3+rpi1+deb10u4_armhf.deb (--unpack):
 unable to create '/usr/share/lintian/overrides/base-files.dpkg-new' (while processing './usr/share/lintian/overrides/base-files'): Structure needs cleaning
Errors were encountered while processing:
 /var/cache/apt/archives/base-files_10.3+rpi1+deb10u4_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

blindtrust, 2020-05-13:
DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v6.28.0
[ INFO ] DietPi-Update | Latest version  : v6.30.0

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=25850 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.3-fpm

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | wget https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/pre-patch_file -O pre-patch_file
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
[  OK  ] DietPi-Update | G_AGUP
[ INFO ] DietPi-Update | APT upgrade, please wait...
Preconfiguring packages ...
(Reading database ... 25084 files and directories currently installed.)
Preparing to unpack .../base-files_10.3+rpi1+deb10u4_armhf.deb ...
Unpacking base-files (10.3+rpi1+deb10u4) over (10.3+rpi1+deb10u2) ...
dpkg: error processing archive /var/cache/apt/archives/base-files_10.3+rpi1+deb10u4_armhf.deb (--unpack):
 unable to create '/usr/share/lintian/overrides/base-files.dpkg-new' (while processing './usr/share/lintian/overrides/base-files'): Structure needs cleaning
Errors were encountered while processing:
 /var/cache/apt/archives/base-files_10.3+rpi1+deb10u4_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] DietPi-Update | G_AGUG
[ INFO ] DietPi-Update | If problems persist, please report at https://github.com/MichaIng/DietPi/issues for investigation.
[ INFO ] DietPi-Update | APT upgrade, please wait...
Preconfiguring packages ...
(Reading database ... 25084 files and directories currently installed.)
Preparing to unpack .../base-files_10.3+rpi1+deb10u4_armhf.deb ...
Unpacking base-files (10.3+rpi1+deb10u4) over (10.3+rpi1+deb10u2) ...
dpkg: error processing archive /var/cache/apt/archives/base-files_10.3+rpi1+deb10u4_armhf.deb (--unpack):
 unable to create '/usr/share/lintian/overrides/base-files.dpkg-new' (while processing './usr/share/lintian/overrides/base-files'): Structure needs cleaning
Errors were encountered while processing:
 /var/cache/apt/archives/base-files_10.3+rpi1+deb10u4_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] DietPi-Update | G_AGUG
[ INFO ] DietPi-Update | If problems persist, please report at https://github.com/MichaIng/DietPi/issues for investigation.
[ INFO ] DietPi-Bugreport | Packing upload archive, please wait...
[  OK  ] DietPi-Bugreport | Connection test: ssh.dietpi.com
[  OK  ] DietPi-Bugreport | Bug report sent, reference code: e228a8e5-9f96-49b6-a40f-e420fca823ab

MichaIng, 2020-05-13:
@blindtrust 
Many thanks for your report.
> Structure needs cleaning

This means you have a file system error.

You have two options:
1. Do a reboot and force a fsck at early boot stage. Generally this allows to relatively safely fix errors while no system service disturbs or gets disturbed. But there is a tiny risk that some boot-critical files are corrupted as well which prevents the system from booting up. Also as long as you do not have monitor and keyboard attached (e.g. headless device), or a serial console, you cannot follow the process for details:
   ```
   > /forcefsck
   reboot
   ```
2. Remount the file system read-only and run fsck manually and interactively. You have full control about what to fix and can follow the process, but the otherwise running system might disturb the process or those are disturbed as they cannot write the file system anymore:
   ```
   mount -o remount,ro /
   fsck /dev/mmcblk0p2
   mount -o remount,rw /
   ```

Sometimes the fsck needs to be redone multiple times to resolve everything, sometimes it might be not able to resolve some. Also after done have a look into `/lost+found` which might contain some files with names like `#234234`, which are files where meta data (e.g. the name and the path/location in file system) has been lost. By opening them with editor you might identify what they are and where they belong to. Many files on UNIX systems contain this info in the first content lines, as e.g. DietPi scripts do.
Also when finished, do another reboot and check `dmesg -l err,crit,alert,emerg` if you see some errors about file system I/O, "Structure needs cleaning" or similar. On RPi there are some errors during early boot expected about vcms, btw.

blindtrust, 2020-05-13:
### I ssh into my box and tried your commands: this is what I got... (sorry, not great at command line stuff but can follow directions) 

root@DietPi:~ /forcefsck <-- mabye this can't be done at SSH? seem like it should work.
-bash: /forcefsck: No such file or directory
root@DietPi:~ 

tried: sudo touch /forcefsck. <-- no error after the return
reboot

mabybe adding sudo and touch worked so I then ran this after reboot.

root@DietPi:~ dmesg -l err,crit,alert,emerg
[    0.890164] vc_vchi_sm_init: failed to open VCHI service (-1)
[    0.890173] [vc_sm_connected_init]: failed to initialize shared memory service
[   31.794872] vc_sm_cma_vchi_init: failed to open VCHI service (-1)
[   31.794893] [vc_sm_connected_init]: failed to initialize shared memory service
root@DietPi:~

Not sure what to do next. will try the sudo touch /forcefsck if you think that will work.

blindtrust, 2020-05-13:
Tried running fsck in the dietpi disk manager on the root drive.. with a reboot below is the log file... screen shot... 
[image]

root@DietPi:/run# cd /initramfs
-bash: cd: /initramfs: No such file or directory
root@DietPi:/run# journalctl -t systemd-fsck
-- Logs begin at Wed 2020-05-13 19:08:40 BST, end at Wed 2020-05-13 19:20:01 BST. --
May 13 19:08:40 DietPi systemd-fsck[122]: Please pass 'fsck.mode=force' on the kernel command line rather than creating /forcefsck on th
e root file system.
May 13 19:08:40 DietPi systemd-fsck[122]: e2fsck 1.44.5 (15-Dec-2018)
May 13 19:08:40 DietPi systemd-fsck[122]: Pass 1: Checking inodes, blocks, and sizes
May 13 19:09:03 DietPi systemd-fsck[122]: Pass 2: Checking directory structure
May 13 19:09:04 DietPi systemd-fsck[122]: Pass 3: Checking directory connectivity
May 13 19:09:04 DietPi systemd-fsck[122]: Pass 4: Checking reference counts
May 13 19:09:05 DietPi systemd-fsck[122]: Pass 5: Checking group summary information
May 13 19:09:07 DietPi systemd-fsck[122]: rootfs: 25408/1822992 files (0.4% non-contiguous), 814528/7619584 blocks
May 13 19:09:09 DietPi systemd-fsck[222]: Please pass 'fsck.mode=force' on the kernel command line rather than creating /forcefsck on th
e root file system.
May 13 19:09:10 DietPi systemd-fsck[222]: fsck.fat 4.1 (2017-01-24)
May 13 19:09:10 DietPi systemd-fsck[222]: /dev/mmcblk0p1: 347 files, 110426/516191 clusters
root@DietPi:/run# touch fsckmond-force
root@DietPi:/run# reboot
root@DietPi:/run# Connection to 192.168.1.4 closed by remote host.
Connection to 192.168.1.4 closed.

now I just tried fsck.mosde-force just a guess is that will work, no error when I hit return and rebooted....

blindtrust, 2020-05-13:
results after fsck.mosde-force.

root@DietPi:~# journalctl -t systemd-fsck
-- Logs begin at Wed 2020-05-13 19:22:21 BST, end at Wed 2020-05-13 19:26:41 BST. --
May 13 19:22:21 DietPi systemd-fsck[120]: e2fsck 1.44.5 (15-Dec-2018)
May 13 19:22:21 DietPi systemd-fsck[120]: rootfs: clean, 25405/1822992 files, 814533/7619584 blocks
May 13 19:22:23 DietPi systemd-fsck[222]: fsck.fat 4.1 (2017-01-24)
May 13 19:22:23 DietPi systemd-fsck[222]: /dev/mmcblk0p1: 347 files, 110426/516191 clusters
root@DietPi:~# 

retrying update now!

failed

[ INFO ] DietPi-Update | APT update, please wait...
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Fetched 15.0 kB in 1s (13.9 kB/s)
Reading package lists...
[  OK  ] DietPi-Update | G_AGUP
[ INFO ] DietPi-Update | APT upgrade, please wait...
[FAILED] DietPi-Update | G_AGUG
[ INFO ] DietPi-Update | If problems persist, please report at https://github.com/MichaIng/DietPi/issues for investigation.
[ INFO ] DietPi-Update | APT upgrade, please wait...
[FAILED] DietPi-Update | G_AGUG
[ INFO ] DietPi-Update | If problems persist, please report at https://github.com/MichaIng/DietPi/issues for investigation.
[ INFO ] DietPi-Bugreport | Packing upload archive, please wait...
[  OK  ] DietPi-Bugreport | Connection test: ssh.dietpi.com
[  OK  ] DietPi-Bugreport | Bug report sent, reference code: e228a8e5-9f96-49b6-a40f-e420fca823ab

---------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue         -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket              -
---------------------------------------------------------------------

#### Details:
- Date           | Wed 13 May 19:29:22 BST 2020
- Bug report     | e228a8e5-9f96-49b6-a40f-e420fca823ab
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGUG
- Exit code      | 139
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```

```

[image]

MichaIng, 2020-05-13:
@blindtrust 
> May 13 19:08:40 DietPi systemd-fsck[122]: Please pass 'fsck.mode=force' on the kernel command line rather than creating /forcefsck on the root file system.

This is btw expected. However the old method is so nice because the file is removed automatically, hence nothing needs to be done do revert/disable fsck on next reboot. Adding `fsck.mode=force` to the kernel cmdline, which means that you need to add it to `/boot/cmdline.txt`, has the disadvantage that fsck runs on every boot until you manually remove the entry again. Especially on RPi with SDcard this is rarely wanted since it delays boot quite much on larger SDcards. For this reason we use the old method as long as it works, and the warning can be safely ignored 😉.
DietPi-Drive_Manager does the same btw, so that is fine.

You forgot the redirect on your first attempt to create the file: `> /forcefsck`, note the leading `>`, which is the most native method to create an empty file.

Your dmesg looks fine, the error messages you see are expected when running the RPi headless with less then 32M GPU memory share. Some GPU features are then disabled, which is pretty fine as long as no media decoding, RPi camera interface or such is used.

Okay now fsck went through nicely, but APT upgrade still fails. Please try to run it manually from console:
```
apt update
apt upgrade
```

blindtrust, 2020-05-14:
ran the last two commands and this is what i got....

root@DietPi:~# **apt update**
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done                             
Building dependency tree       
Reading state information... Done
56 packages can be upgraded. Run 'apt list --upgradable' to see them.
root@DietPi:~# apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  apt apt-utils base-files curl e2fsprogs firmware-atheros firmware-brcm80211 firmware-iwlwifi firmware-misc-nonfree firmware-realtek git git-man idn2
  iputils-ping libapt-inst2.0 libapt-pkg5.0 libcom-err2 libcurl3-gnutls libcurl4 libext2fs2 libgnutls30 libicu63 libidn2-0 libldap-2.4-2 libldap-common
  libpam-modules libpam-modules-bin libpam-runtime libpam0g libraspberrypi-bin libraspberrypi0 libss2 libssl1.1 libsystemd0 libtiff5 libudev1 openssl
  php7.3-cli php7.3-common php7.3-curl php7.3-fpm php7.3-gd php7.3-json php7.3-mbstring php7.3-opcache php7.3-readline php7.3-sqlite3 php7.3-xml php7.3-zip
  raspberrypi-bootloader raspberrypi-kernel sudo systemd systemd-sysv tzdata udev
56 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/128 MB of archives.
After this operation, 2,755 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Segmentation fault

root@DietPi:~# apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done                            
Building dependency tree       
Reading state information... Done
56 packages can be upgraded. Run 'apt list --upgradable' to see them.

root@DietPi:~# apt list --upgradeable
Listing... Done
apt-utils/stable 1.8.2.1 armhf [upgradable from: 1.8.2]
apt/stable 1.8.2.1 armhf [upgradable from: 1.8.2]
base-files/stable 10.3+rpi1+deb10u4 armhf [upgradable from: 10.3+rpi1+deb10u2]
curl/stable 7.64.0-4+deb10u1 armhf [upgradable from: 7.64.0-4]
e2fsprogs/stable 1.44.5-1+deb10u3 armhf [upgradable from: 1.44.5-1+deb10u2]
firmware-atheros/testing 1:20190114-1+rpt6 all [upgradable from: 1:20190114-1+rpt4]
firmware-brcm80211/testing 1:20190114-1+rpt6 all [upgradable from: 1:20190114-1+rpt4]
firmware-iwlwifi/testing 1:20190114-1+rpt6 all [upgradable from: 1:20190114-1+rpt4]
firmware-misc-nonfree/testing 1:20190114-1+rpt6 all [upgradable from: 1:20190114-1+rpt4]
firmware-realtek/testing 1:20190114-1+rpt6 all [upgradable from: 1:20190114-1+rpt4]
git-man/stable 1:2.20.1-2+deb10u3 all [upgradable from: 1:2.20.1-2+deb10u1]
git/stable 1:2.20.1-2+deb10u3 armhf [upgradable from: 1:2.20.1-2+deb10u1]
idn2/stable 2.0.5-1+deb10u1 armhf [upgradable from: 2.0.5-1]
iputils-ping/stable 3:20180629-2+deb10u1 armhf [upgradable from: 3:20180629-2]
libapt-inst2.0/stable 1.8.2.1 armhf [upgradable from: 1.8.2]
libapt-pkg5.0/stable 1.8.2.1 armhf [upgradable from: 1.8.2]
libcom-err2/stable 1.44.5-1+deb10u3 armhf [upgradable from: 1.44.5-1+deb10u2]
libcurl3-gnutls/stable 7.64.0-4+deb10u1 armhf [upgradable from: 7.64.0-4]
libcurl4/stable 7.64.0-4+deb10u1 armhf [upgradable from: 7.64.0-4]
libext2fs2/stable 1.44.5-1+deb10u3 armhf [upgradable from: 1.44.5-1+deb10u2]
libgnutls30/stable 3.6.7-4+deb10u3 armhf [upgradable from: 3.6.7-4]
libicu63/stable 63.1-6+deb10u1 armhf [upgradable from: 63.1-6]
libidn2-0/stable 2.0.5-1+deb10u1 armhf [upgradable from: 2.0.5-1]
libldap-2.4-2/stable 2.4.47+dfsg-3+rpi1+deb10u2 armhf [upgradable from: 2.4.47+dfsg-3+rpi1+deb10u1]
libldap-common/stable 2.4.47+dfsg-3+rpi1+deb10u2 all [upgradable from: 2.4.47+dfsg-3+rpi1+deb10u1]
libpam-modules-bin/testing 1.3.1-5+rpt2 armhf [upgradable from: 1.3.1-5+rpt1]
libpam-modules/testing 1.3.1-5+rpt2 armhf [upgradable from: 1.3.1-5+rpt1]
libpam-runtime/testing 1.3.1-5+rpt2 all [upgradable from: 1.3.1-5+rpt1]
libpam0g/testing 1.3.1-5+rpt2 armhf [upgradable from: 1.3.1-5+rpt1]
libraspberrypi-bin/testing 1.20200212-1 armhf [upgradable from: 1.20190925+1-1]
libraspberrypi0/testing 1.20200212-1 armhf [upgradable from: 1.20190925+1-1]
libss2/stable 1.44.5-1+deb10u3 armhf [upgradable from: 1.44.5-1+deb10u2]
libssl1.1/testing 1.1.1d-0+deb10u3+rpt1 armhf [upgradable from: 1.1.1d-0+deb10u2]
libsystemd0/stable 241-7~deb10u4+rpi1 armhf [upgradable from: 241-7~deb10u2+rpi1]
[... 33 lines cut ...]
  raspberrypi-bootloader raspberrypi-kernel sudo systemd systemd-sysv tzdata udev
56 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/128 MB of archives.
After this operation, 2,755 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Segmentation fault

**looked around on line and found this as a fix, did not work.....same sgement fault error..
sudo apt-get install --reinstall apt**

root@DietPi:~# sudo apt-get install --reinstall apt
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  apt-utils
Suggested packages:
  apt-doc aptitude | synaptic | wajig dpkg-dev powermgmt-base
The following packages will be upgraded:
  apt apt-utils
2 upgraded, 0 newly installed, 0 to remove and 54 not upgraded.
Need to get 0 B/1,748 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Segmentation fault

MichaIng, 2020-05-14:
@blindtrust 
Hmm, if apt is somehow broken, of course you cannot use it to fix itself 😉. Try the following:
```
cd /tmp
wget http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/apt_1.8.2.1_armhf.deb
dpkg -i apt_1.8.2.1_armhf.deb
rm apt_1.8.2.1_armhf.deb
apt upgrade
```

blindtrust, 2020-05-18:
Still get that… segmentation fault…

root@DietPi:/tmp# dpkg -i apt_1.8.2.1_armhf.deb
(Reading database ... 25084 files and directories currently installed.)
Preparing to unpack apt_1.8.2.1_armhf.deb ...
Unpacking apt (1.8.2.1) over (1.8.2) ...
Setting up apt (1.8.2.1) ...
apt-daily-upgrade.timer is a disabled or a static unit not running, not starting it.
apt-daily.timer is a disabled or a static unit not running, not starting it.
Processing triggers for libc-bin (2.28-10+rpi1) ...
root@DietPi:/tmp# rm apt_1.8.2.1_armhf.deb
root@DietPi:/tmp# apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 apt-utils : Depends: apt (= 1.8.2) but 1.8.2.1 is installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
root@DietPi:/tmp# apt --fix-broken install
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  apt-utils
The following packages will be upgraded:
  apt-utils
1 upgraded, 0 newly installed, 0 to remove and 54 not upgraded.
Need to get 0 B/399 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Segmentation fault

MichaIng, 2020-05-18:
@blindtrust 
Please try the same with apt-utils package:
```
cd /tmp
wget http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/apt-utils_1.8.2.1_armhf.deb
dpkg -i apt-utils_1.8.2.1_armhf.deb
rm apt-utils_1.8.2.1_armhf.deb
apt upgrade
```

blindtrust, 2020-05-18:
Thank you so much for you help on this…. It seemed to do more but again failed with the  Segmentation fault ???

root@DietPi:~# cd /tmp
root@DietPi:/tmp# wget http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/apt-utils_1.8.2.1_armhf.deb
--2020-05-18 16:49:17--  http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/apt-utils_1.8.2.1_armhf.deb
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... 93.93.128.193, 2a00:1098:0:80:1000:75:0:3
Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: http://mirror.sjc02.svwh.net/raspbian/raspbian/pool/main/a/apt/apt-utils_1.8.2.1_armhf.deb [following]
--2020-05-18 16:49:18--  http://mirror.sjc02.svwh.net/raspbian/raspbian/pool/main/a/apt/apt-utils_1.8.2.1_armhf.deb
Resolving mirror.sjc02.svwh.net (mirror.sjc02.svwh.net)... 72.5.72.15, 2606:c680:0:b:3830:34ff:fe66:6663
Connecting to mirror.sjc02.svwh.net (mirror.sjc02.svwh.net)|72.5.72.15|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 399376 (390K) [application/octet-stream]
Saving to: ‘apt-utils_1.8.2.1_armhf.deb’

apt-utils_1.8.2.1_armhf.deb             100%[===============================================================================>] 390.02K  1.24MB/s    in 0.3s    

2020-05-18 16:49:18 (1.24 MB/s) - ‘apt-utils_1.8.2.1_armhf.deb’ saved [399376/399376]

root@DietPi:/tmp# dpkg -i apt-utils_1.8.2.1_armhf.deb
(Reading database ... 25084 files and directories currently installed.)
Preparing to unpack apt-utils_1.8.2.1_armhf.deb ...
Unpacking apt-utils (1.8.2.1) over (1.8.2) ...
Setting up apt-utils (1.8.2.1) ...
root@DietPi:/tmp# rm apt-utils_1.8.2.1_armhf.deb
root@DietPi:/tmp# apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  base-files curl e2fsprogs firmware-atheros firmware-brcm80211 firmware-iwlwifi firmware-misc-nonfree firmware-realtek git git-man idn2 iputils-ping
  libapt-inst2.0 libapt-pkg5.0 libcom-err2 libcurl3-gnutls libcurl4 libext2fs2 libgnutls30 libicu63 libidn2-0 libldap-2.4-2 libldap-common libpam-modules
  libpam-modules-bin libpam-runtime libpam0g libraspberrypi-bin libraspberrypi0 libss2 libssl1.1 libsystemd0 libtiff5 libudev1 openssl php7.3-cli
  php7.3-common php7.3-curl php7.3-fpm php7.3-gd php7.3-json php7.3-mbstring php7.3-opcache php7.3-readline php7.3-sqlite3 php7.3-xml php7.3-zip
  raspberrypi-bootloader raspberrypi-kernel sudo systemd systemd-sysv tzdata udev
54 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 127 MB of archives.
After this operation, 2,755 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf base-files armhf 10.3+rpi1+deb10u4 [70.0 kB]
Get:3 https://archive.raspberrypi.org/debian buster/main armhf libpam0g armhf 1.3.1-5+rpt2 [121 kB]
Get:2 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf libsystemd0 armhf 241-7~deb10u4+rpi1 [305 kB]
Get:5 https://archive.raspberrypi.org/debian buster/main armhf libpam-modules-bin armhf 1.3.1-5+rpt2 [103 kB]    
Get:4 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf systemd armhf 241-7~deb10u4+rpi1 [3,255 kB]        
Get:6 https://archive.raspberrypi.org/debian buster/main armhf libpam-modules armhf 1.3.1-5+rpt2 [290 kB]                       
Get:7 https://archive.raspberrypi.org/debian buster/main armhf libpam-runtime all 1.3.1-5+rpt2 [211 kB]                         
Get:8 https://archive.raspberrypi.org/debian buster/main armhf libssl1.1 armhf 1.1.1d-0+deb10u3+rpt1 [1,275 kB]
Get:10 https://archive.raspberrypi.org/debian buster/main armhf firmware-atheros all 1:20190114-1+rpt6 [3,887 kB]
Get:13 https://archive.raspberrypi.org/debian buster/main armhf firmware-brcm80211 all 1:20190114-1+rpt6 [4,358 kB]
Get:17 https://archive.raspberrypi.org/debian buster/main armhf firmware-iwlwifi all 1:20190114-1+rpt6 [5,318 kB]
Get:9 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf udev armhf 241-7~deb10u4+rpi1 [1,248 kB]                     
Get:20 https://archive.raspberrypi.org/debian buster/main armhf firmware-misc-nonfree all 1:20190114-1+rpt6 [3,069 kB]           
Get:11 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf libudev1 armhf 241-7~deb10u4+rpi1 [144 kB]
Get:12 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf systemd-sysv armhf 241-7~deb10u4+rpi1 [100 kB]
Get:14 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf libidn2-0 armhf 2.0.5-1+deb10u1 [74.0 kB]
Get:15 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf libgnutls30 armhf 3.6.7-4+deb10u3 [1,047 kB]
Get:23 https://archive.raspberrypi.org/debian buster/main armhf firmware-realtek all 1:20190114-1+rpt6 [495 kB]                               
Get:24 https://archive.raspberrypi.org/debian buster/main armhf raspberrypi-kernel armhf 1.20200212-1 [71.4 MB]                     
Get:16 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf libext2fs2 armhf 1.44.5-1+deb10u3 [202 kB]
Get:18 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf e2fsprogs armhf 1.44.5-1+deb10u3 [532 kB]                     
Get:19 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf libapt-pkg5.0 armhf 1.8.2.1 [855 kB]                        
[... cut ...]
Get:53 https://archive.raspberrypi.org/debian buster/main armhf raspberrypi-bootloader armhf 1.20200212-1 [5,581 kB]                                            
Get:54 https://archive.raspberrypi.org/debian buster/main armhf openssl armhf 1.1.1d-0+deb10u3+rpt1 [807 kB]                                                    
Fetched 127 MB in 33s (3,821 kB/s)                                                                                                                              
Segmentation fault
root@DietPi:/tmp#

blindtrust, 2020-05-18:
> @blindtrust <https://github.com/blindtrust>
> Please try the following:
>
> cd /tmp
> wget http://raspbian.raspberrypi.org/raspbian/pool/main/s/strace/strace_4.26-0.2_armhf.deb
> dpkg -i strace_4.26-0.2_armhf.deb
> rm strace_4.26-0.2_armhf.deb
> strace -f apt-get upgrade &> strace.log
> more strace.log
> Hopefully we see on which command/binary it exactly fails.

I don’t think it worked correctly output below…

root@DietPi:~# cd /tmp
root@DietPi:/tmp# 
root@DietPi:/tmp# 
root@DietPi:/tmp# wget http://raspbian.raspberrypi.org/raspbian/pool/main/s/strace/strace_4.26-0.2_armhf.deb
--2020-05-18 17:12:05--  http://raspbian.raspberrypi.org/raspbian/pool/main/s/strace/strace_4.26-0.2_armhf.deb
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... 93.93.128.193, 2a00:1098:0:80:1000:75:0:3
Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80... connected.
HTTP request sent, awaiting response... 
302 Found
Location: http://mirror.sjc02.svwh.net/raspbian/raspbian/pool/main/s/strace/strace_4.26-0.2_armhf.deb [following]
--2020-05-18 17:12:05--  http://mirror.sjc02.svwh.net/raspbian/raspbian/pool/main/s/strace/strace_4.26-0.2_armhf.deb
Resolving mirror.sjc02.svwh.net (mirror.sjc02.svwh.net)... 72.5.72.15, 2606:c680:0:b:3830:34ff:fe66:6663
Connecting to mirror.sjc02.svwh.net (mirror.sjc02.svwh.net)|72.5.72.15|:80... 
connected.
HTTP request sent, awaiting response... 200 OK
Length: 754964 (737K) [application/octet-stream]
Saving to: ‘strace_4.26-0.2_armhf.deb’

strace_4.26-0.2_armhf.deb               100%[===============================================================================>] 737.27K  2.30MB/s    in 0.3s    

2020-05-18 17:12:05 (2.30 MB/s) - ‘strace_4.26-0.2_armhf.deb’ saved [754964/754964]

root@DietPi:/tmp# 
root@DietPi:/tmp# 
root@DietPi:/tmp# dpkg -i strace_4.26-0.2_armhf.deb
(Reading database ... 25095 files and directories currently installed.)
Preparing to unpack strace_4.26-0.2_armhf.deb ...
Unpacking strace (4.26-0.2) over (4.26-0.2) ...
Setting up strace (4.26-0.2) ...
root@DietPi:/tmp# 
root@DietPi:/tmp# 
root@DietPi:/tmp# 
root@DietPi:/tmp# rm strace_4.26-0.2_armhf.deb
root@DietPi:/tmp# strace -f apt-get upgrade &> strace.log

Segmentation fault
root@DietPi:/tmp# 

root@DietPi:/tmp# more strace.log
execve("/usr/bin/apt-get", ["apt-get", "upgrade"], 0x7eb90d98 /* 15 vars */) = 0
brk(NULL)                               = 0xde9000
uname({sysname="Linux", nodename="DietPi", ...}) = 0
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76ff9000
access("/etc/ld.so.preload", R_OK)      = 0
openat(AT_FDCWD, "/etc/ld.so.preload", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=54, ...}) = 0
mmap2(NULL, 54, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0x76ff8000
close(3)                                = 0
readlink("/proc/self/exe", "/usr/bin/apt-get", 4096) = 16
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\254\3\0\0004\0\0\0"..., 512) = 512
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
fstat64(3, {st_mode=S_IFREG|0644, st_size=17708, ...}) = 0
mmap2(NULL, 81964, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76fb6000
mprotect(0x76fba000, 61440, PROT_NONE)  = 0
mmap2(0x76fc9000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x76fc9000
close(3)                                = 0
munmap(0x76ff8000, 54)                  = 0
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=18552, ...}) = 0
mmap2(NULL, 18552, PROT_READ, MAP_PRIVATE, 3, 0) = 0x76ff4000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libapt-private.so.0.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3

blindtrust, 2020-05-18:
I made a new log file I think strace02.log and here is the full output.

root@DietPi:/tmp# strace -f apt-get upgrade &> strace02.log
root@DietPi:/tmp# more strace02.log
execve("/usr/bin/apt-get", ["apt-get", "upgrade"], 0x7e873d98 /* 15 vars */) = 0
brk(NULL)                               = 0x102f000
uname({sysname="Linux", nodename="DietPi", ...}) = 0
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76f3b000
access("/etc/ld.so.preload", R_OK)      = 0
openat(AT_FDCWD, "/etc/ld.so.preload", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=54, ...}) = 0
mmap2(NULL, 54, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0x76f3a000
close(3)                                = 0
readlink("/proc/self/exe", "/usr/bin/apt-get", 4096) = 16
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\254\3\0\0004\0\0\0"..., 512) = 512
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
fstat64(3, {st_mode=S_IFREG|0644, st_size=17708, ...}) = 0
mmap2(NULL, 81964, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76ef8000
mprotect(0x76efc000, 61440, PROT_NONE)  = 0
mmap2(0x76f0b000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x76f0b000
close(3)                                = 0
munmap(0x76f3a000, 54)                  = 0
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=18552, ...}) = 0
mmap2(NULL, 18552, PROT_READ, MAP_PRIVATE, 3, 0) = 0x76f36000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libapt-private.so.0.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\360\264\0\0004\0\0\0"..., 512) = 512
_llseek(3, 348556, [348556], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
_llseek(3, 348188, [348188], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 348556, [348556], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
_llseek(3, 348188, [348188], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 348556, [348556], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
_llseek(3, 348188, [348188], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 348556, [348556], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
_llseek(3, 348188, [348188], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 348556, [348556], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
[... 841 lines cut ...]
[pid  1641] _llseek(3, 129016, [129016], SEEK_SET) = 0
[pid  1641] read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1400) = 1400
[pid  1641] _llseek(3, 94664, [94664], SEEK_SET) = 0
[pid  1641] read(3, "A.\0\0\0aeabi\0\1$\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\23\1\24"..., 47) = 47
[pid  1641] _llseek(3, 129016, [129016], SEEK_SET) = 0
[pid  1641] read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1400) = 1400
[pid  1641] _llseek(3, 94664, [94664], SEEK_SET) = 0
[pid  1641] read(3, "A.\0\0\0aeabi\0\1$\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\23\1\24"..., 47) = 47
[pid  1641] _llseek(3, 129016, [129016], SEEK_SET) = 0
[pid  1641] read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1400) = 1400
[pid  1641] _llseek(3, 94664, [94664], SEEK_SET) = 0
[... cut ...]
[pid  1641] mmap2(0x76d6f000, 4720, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x76d6f000
[pid  1641] close(3)                    = 0
[pid  1641] mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76fb7000
[pid  1641] set_tls(0x76fb74d0)         = 0
[pid  1641] mprotect(0x76f43000, 8192, PROT_READ) = 0
[pid  1641] mprotect(0x76d6d000, 4096, PROT_READ) = 0
[pid  1641] mprotect(0x76d82000, 4096, PROT_READ) = 0
[pid  1641] mprotect(0x76df9000, 4096, PROT_READ) = 0
[pid  1641] mprotect(0x76f78000, 4096, PROT_RE

MichaIng, 2020-05-18:
@blindtrust 
AFAIK there is still something missing, could you check if there is more at the end of the log file? Actually the last lines should be enough, as so far everything looks fine:
```
tail -50 /tmp/strace02.log
```

MichaIng, 2020-06-24:
I mark this as closed. Feel free to reopen if issue is still present.

closed 2020-06-24 by MichaIng