MichaIng/DietPi #3537: Unable to install rtorrent/rutorrent

officiallor, 2020-05-14:
#### Details:
- Date           | Thu May 14 10:41:02 EEST 2020
- Bug report     | 1005d594-2a24-4b10-9638-be0c02f67a01
- DietPi version | v6.30.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | Linux localhost 4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019 armv6l GNU/Linux
- Distro         | stretch (ID=4)
- Command        | apt-get -qq --allow-change-held-packages install rtorrent mediainfo screen
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:
Install rtorrent from dietpi-software
1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:
Before this (a few months ago) there was a problem with a package that couldn't be found.
- ...
#### Additional logs:
```
E: Unable to correct problems, you have held broken packages.
```

Joulinar, 2020-05-14:
Hi,

many thanks. Looks like you have some issues on running `apt-get`. Pls can you try to execute it manually

```
apt update
apt upgrade
```

officiallor, 2020-05-14:
I tried apt update and apt upgrade.

root@localhost:~# apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Get:3 https://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Fetched 25.4 kB in 5s (4,822 B/s)                                           Reading package lists... Done
Building dependency tree
Reading state information... Done
9 packages can be upgraded. Run 'apt list --upgradable' to see them.
root@localhost:~# apt list --upgradable
Listing... Done
apt/oldstable 1.4.10 armhf [upgradable from: 1.4.9]
apt-transport-https/oldstable 1.4.10 armhf [upgradable from: 1.4.9]
apt-utils/oldstable 1.4.10 armhf [upgradable from: 1.4.9]
kodi-bin/stable 2:18.2-1~stretch armhf [upgradable from: 2:17.6-2~stretch]
libapt-inst2.0/oldstable 1.4.10 armhf [upgradable from: 1.4.9]              libapt-pkg5.0/oldstable 1.4.10 armhf [upgradable from: 1.4.9]
libavformat57/stable 7:3.2.14-1~deb9u1+rpt1 armhf [upgradable from: 7:3.2.12-1~deb9u1+rpt1]
mpd/stable 0.21.5-3 armhf [upgradable from: 0.20.23-1]
pcmanfm/stable 1.2.5-3+rpi5 armhf [upgradable from: 1.2.5-3]
root@localhost:~# apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libcurl4 : Conflicts: libcurl3 but 7.52.1-5+deb9u10 is to be installed
E: Broken packages
root@localhost:~#

Joulinar, 2020-05-14:
looks like you have Buster and Stretch within your source file configuration. Can you post

`ls -la /etc/apt/sources.list.d`

officiallor, 2020-05-14:
root@localhost:~# ls -al /etc/apt/sources.list.d/                           total 16
drwxr-xr-x 2 root root 4096 Apr  2 04:05 .
drwxr-xr-x 6 root root 4096 May 15 01:57 ..
-rw-r--r-- 1 root root   57 May 14 10:38 dietpi-php.list
-rw-r--r-- 1 root root   60 Oct 28  2018 raspi.list

Thanks for helping 😄

Joulinar, 2020-05-14:
can you post following

```
cat /etc/apt/sources.list
cat /etc/apt/sources.list.d/raspi.list
cat /etc/apt/sources.list.d/dietpi-php.list
```

officiallor, 2020-05-14:
`root@localhost:/etc/apt# cat sources.list
deb http://raspbian.raspberrypi.org/raspbian stretch main contrib non-free rpi
root@localhost:/etc/apt# cd sources.list.d/
root@localhost:/etc/apt/sources.list.d# cat raspi.list
deb https://archive.raspberrypi.org/debian/ stretch main ui
root@localhost:/etc/apt/sources.list.d# cat dietpi-php.list
deb http://raspbian.raspberrypi.org/raspbian buster main
root@localhost:/etc/apt/sources.list.d#`

Joulinar, 2020-05-15:
ok strange that `dietpi-php.list` pointing to Buster. That should not be the case as you are on Stretch. So it would need to be changed. Anyway I guess it will not fix you issue. Let's check what version of `libcurl*` you have installed. pls post following:

`dpkg -l libcurl*`

However it might be better if @MichaIng could have a look. Because having Stretch as well as Buster on source list, doesn't seem healthy 😉

MichaIng, 2020-05-15:
RPi ARMv6 Stretch, yeah this is were we went a risky way to enable PHP7.3 support. There was sadly no other option, otherwise we'd had to disable a lot of web applications which dropped PHP7.0 support.

Found it:
The new `kodi-bin` depends on `libcurl3` while installing PHP7.3 via Buster pulled `libcurl4`. I'll check how to resolve this. Meanwhile you can upgrade all other packages:
```
apt install apt apt-transport-https apt-utils libapt-inst2.0 libapt-pkg5.0 libavformat57 mpd pcmanfm
```

officiallor, 2020-05-15:
I was trying to remove deluge because I wanted to install rutorrent since the beginning. dietpi-software couldn't remove it and I tried manually but no luck. Now I tried apt install apt... (separately each) but for some reason I can't do much. 
I have posted the error from terminal (on ssh dietpi com) if you're interested for more details. I'm not at home now but if it can't be fixed, I will reinstall dietpi cause I don't have many things on my raspberry and it will be a quick solution.
Note: I switched to another mirror because I was thinking it would fix something but I was wrong 😅

`-bash: ​apt install apt apt-transport-https apt-utils libapt-inst2.0 libapt-pkg5.0 libavformat57 mpd: command not found
root@localhost:~# apt install apt
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  cpio initramfs-tools initramfs-tools-core klibc-utils kodi-bin libaacs0
  libao-common libao4 libass5 libasyncns0 libbluetooth3
  libboost-chrono1.61.0 libboost-python1.61.0 libboost-random1.61.0
  libboost-system1.61.0 libfribidi0 libklibc liblzo2-2 libmariadbclient18
  libmicrohttpd12 libpcrecpp0v5 libplymouth4 libpulse-mainloop-glib0
  libpulse0 libshairplay0 libsndfile1 libssh-4 libtag1v5 libtag1v5-vanilla
  libtinyxml2.6.2v5 libtorrent-rasterbar9 libwebpdemux2 linux-base
  mysql-common plymouth python-bluez python-imaging python-libtorrent
  python-pil python-simplejson
Use 'apt autoremove' to remove them.
The following additional packages will be installed:
  apt-utils
Suggested packages:
  apt-doc aptitude | synaptic | wajig dpkg-dev powermgmt-base python-apt
The following packages will be upgraded:
  apt apt-utils
2 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
2 not fully installed or removed.
Need to get 1,583 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian stretch/main armhf apt armhf 1.4.10 [1,190 kB]
Get:2 http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian stretch/main armhf apt-utils armhf 1.4.10 [392 kB]
Fetched 1,583 kB in 12s (127 kB/s)
(Reading database ... 45001 files and directories currently installed.)
Preparing to unpack .../archives/apt_1.4.10_armhf.deb ...
Unpacking apt (1.4.10) over (1.4.9) ...
Setting up apt (1.4.10) ...
Failed to try-restart apt-daily-upgrade.timer: Unit apt-daily-upgrade.timer is masked.
Failed to try-restart apt-daily.timer: Unit apt-daily.timer is masked.
(Reading database ... 45001 files and directories currently installed.)
Preparing to unpack .../apt-utils_1.4.10_armhf.deb ...
Unpacking apt-utils (1.4.10) over (1.4.9) ...
Setting up apt-utils (1.4.10) ...
Setting up deluge-web (1.3.13+git20161130.48cedf63-3) ...
Traceback (most recent call last):
  File "/usr/bin/pycompile", line 32, in <module>
    from subprocess import PIPE, STDOUT, Popen
EOFError: EOF read where object expected
dpkg: error processing package deluge-web (--configure):
 subprocess installed post-installation script returned error exit status 1
Processing triggers for libc-bin (2.28-10+rpi1) ...
Processing triggers for man-db (2.7.6.1-2) ...
Setting up deluge-console (1.3.13+git20161130.48cedf63-3) ...
Traceback (most recent call last):
  File "/usr/bin/pycompile", line 32, in <module>
    from subprocess import PIPE, STDOUT, Popen
EOFError: EOF read where object expected
dpkg: error processing package deluge-console (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 deluge-web
 deluge-console
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@localhost:~#`

Joulinar, 2020-05-15:
Well if you are open for a new installation, it might be the point to switch to Buster and do a clean install on the current Debian Release, which might be more future proof than continue running Stretch :wink:

MichaIng, 2020-05-15:
Jep, a clean install or a Buster distro upgrade is actually the best solution for those boards, otherwise either a different webserver must be used, or Kodi 17 instead of new Kodi 18, or Kodi 18 from Buster as well, although this would mean, due to dependencies, a half distro-upgrade I think. Either way is possible to apply via APT preferences, let me know which you prefer and I'll assist to apply it.

However the current issue is that deluge-web and deluge-console fail to configure.
> dietpi-software couldn't remove it

Please try to do it manually:
```
apt purge deluge-web deluge-console
userdel -rf debian-deluged
systemctl disable deluge-web
systemctl disable deluged
[[ -f '/etc/systemd/system/deluged.service' ]] && rm -R /etc/systemd/system/deluged.service*
[[ -f '/etc/systemd/system/deluge-web.service' ]] && rm -R /etc/systemd/system/deluge-web.service*
[[ -d $G_FP_DIETPI_USERDATA/deluge ]] && rm -R $G_FP_DIETPI_USERDATA/deluge
```

officiallor, 2020-05-16:
I will go with the clean install as it will take much less time to backup some files and reinstall. I don't know what Busster distro upgrade is. Im thinking of just downloading the Raspberry image and reflashing my SD card. Will I face any problems like this or should I do any configuration? 
I tried to purge deluge but it wasn't possible.
Thank you guys

MichaIng, 2020-05-16:
@officiallor 
Okay yes flashing the current image is the cleanest solution definitely and has several other benefits as much newer software/package versions.

Not sure what happened with Deluge, some internal compiling fails which breaks the package configuration and I guess then using APT, even purging them fails as long as they were not configured. It is kinda nasty that APT is able to break itself. I guess dpkg would be able to resolve it then:
```
dpkg -P deluge-web
dpkg -P deluge-console
```

However yes, if you can bear the time, go for the new Buster image. Buster is btw the code name for the new Debian release: https://en.wikipedia.org/wiki/Debian#Code_names

officiallor, 2020-05-16:
When I get back home I will go for the new image. Thank you again @MichaIng @Joulinar !

closed 2020-09-27 by MichaIng