MichaIng/DietPi #3538: dietpi-upgrade failure from 6.29.2 to 6.30 on RPI 3 B+

mocod67, 2020-05-14:
#### Details:
- Date           | Thu May 14 14:23:09 CEST 2020
- DietPi version | v6.29.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux mediapi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:
Just execute "dietpi-upgrade" command
#### Expected behaviour:
Normal completion of the upgrade
#### Actual behaviour:
Unexpected end without upgrade
#### Extra details:

#### Additional logs:
```
debconf: Perl may be unconfigured (Can't locate Debconf/Log.pm in @INC (you may need to install the Debconf::Log module) (@INC contains: /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.28.1 /usr/local/share/perl/5.28.1 /usr/lib/arm-linux-gnueabihf/perl5/5.28 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.28 /usr/share/perl/5.28 /usr/local/lib/site_perl /usr/lib/arm-linux-gnueabihf/perl-base) at (eval 1) line 4.
BEGIN failed--compilation aborted at (eval 1) line 4.
) -- aborting
Setting up libpam0g:armhf (1.3.1-5+rpt2) ...
Can't locate Debconf/Db.pm in @INC (you may need to install the Debconf::Db module) (@INC contains: /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.28.1 /usr/local/share/perl/5.28.1 /usr/lib/arm-linux-gnueabihf/perl5/5.28 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.28 /usr/share/perl/5.28 /usr/local/lib/site_perl /usr/lib/arm-linux-gnueabihf/perl-base) at /usr/share/debconf/frontend line 6.
BEGIN failed--compilation aborted at /usr/share/debconf/frontend line 6.
dpkg: error processing package libpam0g:armhf (--configure):
 installed libpam0g:armhf package post-installation script subprocess returned error exit status 2
Errors were encountered while processing:
 libpam0g:armhf
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2020-05-14:
@mocod67 
Many thanks for your report.

Looks like file corruption. Please try to run the configuration isolated:
```
dpkg --configure libpam0g
```

mocod67, 2020-05-14:
Thanks for your prompt reply.
I've tried your command but without success:

```
root@mediapi:~# dpkg --configure libpam0g
Paramétrage de libpam0g:armhf (1.3.1-5+rpt2) ...
Can't locate Debconf/Db.pm in @INC (you may need to install the Debconf::Db module) (@INC contains: /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.28.1 /usr/local/share/perl/5.28.1 /usr/lib/arm-linux-gnueabihf/perl5/5.28 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.28 /usr/share/perl/5.28 /usr/local/lib/site_perl /usr/lib/arm-linux-gnueabihf/perl-base) at /usr/share/debconf/frontend line 6.
BEGIN failed--compilation aborted at /usr/share/debconf/frontend line 6.
dpkg: erreur de traitement du paquet libpam0g:armhf (--configure) :
 installed libpam0g:armhf package post-installation script subprocess returned error exit status 2
Traitement des actions différées (« triggers ») pour libc-bin (2.28-10+rpi1) ...
Traitement des actions différées (« triggers ») pour install-info (6.5.0.dfsg.1-4+b1) ...
Des erreurs ont été rencontrées pendant l'exécution :
 libpam0g:armhf
```
Currently I cannot install or update anything while the error is still there.
I remember that for the same reason, a month or two ago, I've re-installed from scratch my media player with dietp.
I'm looking for a backup to find if the issue was the same one and let you know if I can reproduce it with another release.

MichaIng, 2020-05-14:
@mocod67 
I think it is corrupted files. Please check: `dmesg -l err,crit,alert,emerg`
If it contains file system I/O errors, then you could try to fix them:
```
echo '' > /forcefsck
reboot
```
However likely some files are broken or lost, you can try to install related packages manually:
```
cd /tmp
wget http://raspbian.raspberrypi.org/raspbian/pool/main/d/debconf/debconf_1.5.71_all.deb
wget http://raspbian.raspberrypi.org/raspbian/pool/main/p/perl/perl-base_5.28.1-6_armhf.deb
dpkg -i *.deb
rm *.deb
dpkg --configure libpam0g
```

mocod67, 2020-05-15:
@MichaIng 
I've checked for the I/O errors in the logs as requested. There's nothing.

I've also tried your other dpkg commands. I've got the following errors (as part of the comments are in French, I've translated them):
`root@mediapi:/tmp# dpkg -i *.deb
(Read of the database... 40845 files and directories already installed)
Preparing to unpack debconf_1.5.71_all.deb ...
Unpacking of debconf (1.5.71) sur (1.5.71) ...
dpkg: error while processing the archive debconf_1.5.71_all.deb (--install) :
 Not possible to create « /usr/share/perl5/Debconf/AutoSelect.pm.dpkg-new » (while processing « ./usr/share/perl5/Debconf/AutoSelect.pm »): The structure needs a clean up.
Can't locate Debconf/Db.pm in @INC (you may need to install the Debconf::Db module) (@INC contains: /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.28.1 /usr/local/share/perl/5.28.1 /usr/lib/arm-linux-gnueabihf/perl5/5.28 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.28 /usr/share/perl/5.28 /usr/local/lib/site_perl /usr/lib/arm-linux-gnueabihf/perl-base) at /usr/share/debconf/frontend line 6.
BEGIN failed--compilation aborted at /usr/share/debconf/frontend line 6.
dpkg: error while cleaning up:
 installed debconf package post-installation script subprocess returned error exit status 2
Preparing to unpack perl-base_5.28.1-6_armhf.deb ...
Unpacking of perl-base (5.28.1-6) sur (5.28.1-6) ...
Configuration of perl-base (5.28.1-6) ...
Errors were encounterred during execution:
 debconf_1.5.71_all.deb`

By the way, I've checked the backup of my previous media player. It was a dietpi 6.28.
I had the same perl output error while trying to upgrade it to 6.30. The processing of the issue was different though as libpam0g was not impacted.

MichaIng, 2020-05-15:
@mocod67 
> The structure needs a clean up.

Hmm this is again clearly a file corruption issue. `dmesg -l err,crit,alert,emerg` now still doesn't show anything? What happens if you try to remove the affected file?
```
rm /usr/share/perl5/Debconf/AutoSelect.pm
ls -l /usr/share/perl5/Debconf/AutoSelect.pm
```

> Can't locate Debconf/Db.pm

This file is part of the debconf package, so as long as this fails, further APT and dpkg installs can fail as well: https://packages.debian.org/buster/all/debconf/filelist

mocod67, 2020-05-15:
@MichaIng 
Ok, it seems that I now have I/O error:
`root@mediapi:/tmp# dmesg -l err,crit,alert,emerg
[  239.298563] EXT4-fs error (device mmcblk0p2): ext4_find_dest_de:1815: inode #36517: block 11954: comm dpkg: bad entry in directory: rec_len % 4 != 0 - offset=0, inode=4294967295, rec_len=65535, name_len=255, size=4096`

I tried
`echo '' > /forcefsck
reboot`
and don't have anymore error message.

What should I do then?

MichaIng, 2020-05-15:
@mocod67 
Please try again to install the debconf package:
```
cd /tmp
wget http://raspbian.raspberrypi.org/raspbian/pool/main/d/debconf/debconf_1.5.71_all.deb
wget http://raspbian.raspberrypi.org/raspbian/pool/main/p/perl/perl-base_5.28.1-6_armhf.deb
dpkg -i *.deb
rm *.deb
dpkg --configure libpam0g
```

mocod67, 2020-05-16:
@MichaIng 
Everything went smoothly. I then retried the dietpi-update without success:

`#### Details:
- Date           | Sat May 16 07:16:30 CEST 2020
- DietPi version | v6.29.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux mediapi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update

#### Additional logs:
```
Setting up systemd (241-7~deb10u4+rpi1) ...
Can't locate Debian/AdduserCommon.pm in @INC (you may need to install the Debian::AdduserCommon module) (@INC contains: /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.28.1 /usr/local/share/perl/5.28.1 /usr/lib/arm-linux-gnueabihf/perl5/5.28 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.28 /usr/share/perl/5.28 /usr/local/lib/site_perl /usr/lib/arm-linux-gnueabihf/perl-base) at /usr/sbin/addgroup line 34.
BEGIN failed--compilation aborted at /usr/sbin/addgroup line 34.
dpkg: error processing package systemd (--configure):
 installed systemd package post-installation script subprocess returned error exit status 2
Errors were encountered while processing:
 systemd
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
`
I'm preparing another SDCard to make sure it doesn't come from it.

MichaIng, 2020-05-16:
@mocod67 
Indeed looks like fixing the first creates a corruption in the second. Might be a bad block on the SDcard. How does the whole directory of trouble look like, actually?
```
ls -Al /usr/share/perl5/Debconf/
```

mocod67, 2020-05-16:
@MichaIng 
Here's the content of the directory:

`
root@mediapi:~# ls -Al /usr/share/perl5/{Debconf,Debian}/
/usr/share/perl5/Debconf/:
total 124
-rw-r--r-- 1 root root  1948 févr. 26  2019 AutoSelect.pm
-rw-r--r-- 1 root root   499 févr. 26  2019 Base.pm
drwxr-xr-x 2 root root  4096 mai   16 07:09 Client
-rw-r--r-- 1 root root  7020 févr. 26  2019 Config.pm
-rw-r--r-- 1 root root 15639 févr. 26  2019 ConfModule.pm
drwxr-xr-x 2 root root  4096 mai   16 07:09 DbDriver
-rw-r--r-- 1 root root  2359 févr. 26  2019 DbDriver.pm
-rw-r--r-- 1 root root  1202 févr. 26  2019 Db.pm
drwxr-xr-x 8 root root  4096 mai   16 07:09 Element
-rw-r--r-- 1 root root   196 févr. 26  2019 Element.pm
-rw-r--r-- 1 root root  1487 févr. 26  2019 Encoding.pm
drwxr-xr-x 2 root root  4096 mai   16 07:09 Format
-rw-r--r-- 1 root root   133 févr. 26  2019 Format.pm
drwxr-xr-x 2 root root  4096 mai   16 07:09 FrontEnd
-rw-r--r-- 1 root root  2864 févr. 26  2019 FrontEnd.pm
-rw-r--r-- 1 root root   301 févr. 26  2019 Gettext.pm
-rw-r--r-- 1 root root   198 févr. 26  2019 Iterator.pm
-rw-r--r-- 1 root root   914 févr. 26  2019 Log.pm
-rw-r--r-- 1 root root   291 févr. 26  2019 Path.pm
-rw-r--r-- 1 root root   642 févr. 26  2019 Priority.pm
-rw-r--r-- 1 root root  5867 févr. 26  2019 Question.pm
drwxr-xr-x 2 root root  4096 mai   16 07:09 Template
-rw-r--r-- 1 root root  8201 févr. 26  2019 Template.pm
-rw-r--r-- 1 root root   374 févr. 26  2019 TmpFile.pm

/usr/share/perl5/Debian/:
total 4
drwxr-xr-x 3 root root 4096 mai   16 07:09 DebConf
`

I've also tried the reinstallation of adduser package and then the dietpi-update.
It went smoothly.
I've also flashed another SDCard with a backup and tried the upgrade that completed without error.
You were definitely right about the corruption.

Thank you very much for your help

closed 2020-05-16 by mocod67

MichaIng, 2020-05-16:
@mocod67 
Great that it finally worked. Yes typical issue on SBCs with OS on SDcard sadly. Ran into this myself more than once, funnily once as well with broken perl module files 😄.

mocod67, 2020-05-17:
Once again, thanks for your support.
Have a nice weekend.