MichaIng/DietPi #3539: Can you please add software SNAPCAST in software list in upcoming updates

sealteam666, 2020-05-14:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- ...

#### Describe the solution you'd like:

- ...

#### Describe alternatives you've considered:

- ...

#### Additional context

- ...

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2020-05-14:
@sealteam666 
Many thanks for your request.

A few more information, some links and what it is, would have been helpful. I guess you mean: https://github.com/badaix/snapcast

MichaIng, 2021-06-04:
The client side is currently worked on: #4465
I mark this issue as closed in favour of #2244, where some more info has been collected already.

closed 2021-06-04 by MichaIng