MichaIng/DietPi #3540: dietpi-recallbox retro game

carlosspre, 2020-05-15:
I know that you can install the retropier on the dietpi now the question is whether it is possible to install the recallbox on the dietpi :::

MichaIng, 2020-05-15:
@carlosspre 
Many thanks for your question.

It is possible to install RetroPie and any other piece of software on DietPi that you can install on Debian/Raspbian. But note that we removed the RetroPie install option from DietPi-Software since its installer does a bunch of deeper setup steps that turns the system into a single-purpose device. So it works (following the official install instructions), but it breaks certain DietPi-Software installs and probably DietPi-Config options, according to my last knowledge.

Not sure about RecalBox, but it seems to be installed via Docker: https://gitlab.com/recalbox/recalbox#how-to-build
This means it is basically isolated from the OS and should work fine beside other installs.

MichaIng, 2021-09-24:
We decided to cleanup open issues a bit and close those which will obviously not be addressed soon. If anyone wants to start working on implementing this software, please feel free to to a start, and we'll help where we can.

Here you find instructions about how to implement a new software title to `dietpi-software`: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

But not that we likely won't accept pure Docker containers in `dietpi-software` which are single command installs. For such cases it makes more sense to extend our Docker documentation, if required, to show how Docker containers can be installed, updated and uninstalled.

closed 2021-09-24 by MichaIng