MichaIng/DietPi #3541: Service dependency problem

Vec7or, 2020-05-16:
### Service dependency problem

#### Required Information
- DietPi version | CORE=6, SUB=29, RC=2, GITBRANCH=master, GITOWNER=MichaIng
- Distro version | buster
- Kernel version | Linux DietPi 5.4.14-sunxi #rc1 SMP Sat Jan 25 15:19:01 CET 2020 armv7l GNU/Linux
- SBC model | NanoPi Neo
- Power supply used | 5V 2A
- SDcard used | SanDisk Micro SDHC I 16GB

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi? --> Yes

#### Steps to reproduce
1. Install redis with the dietpi installer
2. Add own service via `dietpi-services` which has a dependency on redis
3. Try to start the service with `dietpi-services`

#### Expected behaviour
- The service should be able to run

#### Actual behaviour
```
Unit redis.service not found
```
#### Extra details
I do not know if this is a known limitation of `dietpi-services` but without using
```
Requires=redis.service
After=network.target redis.service
```
within my services it is very difficult to guarantee service order.

MichaIng, 2020-05-16:
@Vec7or 
Many thanks for your report.

Note that the service name is `redis-server.service`. Custom services are generally added to the end of the list of what DietPi-Services starts, hence is Redis is present, it is started first. If you add `Requires=redis-server.service`. However not that those ordering is only relevant for systemd, while DietPi-Services has an internal order anyway. But it does not hurt to add it, we do it as well, and I have the plan to switch to systemd-based service starts on boot.

Vec7or, 2020-05-17:
You are right. What a silly mistake to get the service name wrong. It did work on another system so the name was different in that reposiotory it seems. But my problem is solved thanks.

closed 2020-05-17 by Vec7or