MichaIng/DietPi #3542: [?] USB Boot coming natively (rpi-update)

Hukuma1, 2020-05-16:
Didn't know how to ask, so apologies.

Yesterday a beta firmware was uploaded by RPi team (https://github.com/raspberrypi/rpi-eeprom/) to enable USB booting. I currently use USB booting with the help of DietPi's Drive Manager. Is there anything to do in the coming weeks if I wanted to switch to official USB boot support? (no sd card) Trying to avoid starting fresh with my NVME mount. ;)

Joulinar, 2020-05-16:
Hi,

many thanks for your message. Well DietPi's Drive Manager doesn't do anything special as it's managing mount options within `/etc/fstab`. Best to my knowledge, USB boot is managed/enabled via `/boot/cmdline.txt`.

Hukuma1, 2020-05-16:
Cool! Will there be a guide possibly to convert it to the native USB boot once it is released? I am not sure how I would migrate from `/etc/fstab` to `/boot/cmdline.txt` method I mean. Or do you mean I can continue using DietPi solution and simply update `cmdline.txt` later to skip SD card altogether?

Joulinar, 2020-05-16:
I guess we misunderstood. There is no DietPi solution. DietPi for Raspberry Pi is based on Raspbian Lite. Means all features and kernel functions are same because it's still as Raspbian Lite. 

Regarding the USB boot, this is currently enabled via `/boot/cmdline.txt` as this is the workaround on RPi4. This is not a DietPi specific.

`/etc/fstab` on the other hand is managing all your mounts on your device. But not really something dedicated to USB boot and there is nothing to migrate. 

depending on how USB boot will work on RPi4, you maybe need to adjust `/boot/cmdline.txt` later on. Once available it will be described at Raspberry web site.

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

MichaIng, 2020-05-17:
@Hukuma1 
Many thanks for this information, that is great that RPi4 finally gets USB boot, something I was waiting for before even considering to buy one.

DietPi-Drive_Manager allows to move the root file system only, not the boot file system for now. Especially when the external drive does already contain the root file system, moving the boot file system + bootloader requires some repartitioning and manually cloning the bootloader bits, as it is located outside of the partition. I don't even know a command line tool that allows moving a partition together with contained data, but `gparted` only, which is an interactive GUI util.

Long story short, if you want to boot from USB drive without manual tinkering, flash a fresh DietPi image onto it, that is much safer and cleaner 🙂. In theory you could recover a `dietpi-backup` to quickly re-apply your old data (move the backup to the exact same location on the new system), rerun `dietpi-drive_manager` or remove the obsolete /boot mount from `/etc/fstab` manually then. bootloader and EEPROM are not covered by dietpi-backup, and it does not case about partitions/mounts either, so that should not cause problems.

Hukuma1, 2020-05-17:
Ah thank you! Yes, I was wondering the procedure if I did not want to start fresh. I think I started fresh like 6 times now with all the stuff I tinkered with over the past few months. 🤣 Using Drive_Manager has been incredible to move it to USB drive. So, so much faster than the SD card. Will close this out as it's been 'solved.' Thanks all!

closed 2020-05-17 by Hukuma1