MichaIng/DietPi #3543: dietpi-update 6.28 - 6.30 source error

ibksturm, 2020-05-17:
Hi There

I'm Running DP on asus Tinkerboard 2 S (build in MMC)

When i try dietpi-update from 6.28 - 6.30 theres a depenci error

[image]

so the problem is, on webserver theres no folder ...`/downloads/binaries/unknow/`.... but the vmtouch_armv7l.deb file is locatet in `https://dietpi.com/downloads/binaries/stretch/`

please update location path

Joulinar, 2020-05-17:
Hi,

Many thanks for your report. The path to the binary is build based on detection of your device and debian version. Seems the update script is not able to detect your system correctly. Therefore its giving you an `unknown` with the path.

ibksturm, 2020-05-17:
wow that was fast 

i found a work around, if i'd run `Change command : Adjust and rerun the command` i could change the path on `wget https://dieptpi.com......./strech/....`

aaah funny, you post the same workaround at the same moment :)

MichaIng, 2020-05-17:
@ibksturm 
Your distro version is not detected correctly. That will cause issues throughout our scripts so before applying any workaround you should definitely fix the root, to not run into a bunch of follow-up issues.

Please paste the output of:
```
cat /etc/os-release /etc/debian_version
```

ibksturm, 2020-05-17:
```
NAME="Ubuntu"
VERSION="20.04 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
bullseye/sid

```

aah i understand
-> source of the problem is, that i manualy add ubuntu/focal sources in the apt

---> that's not a bug, that's a feature :P *smile*

closed 2020-05-17 by ibksturm

MichaIng, 2020-05-17:
@ibksturm 
Okay that is a pretty good reason 😄. Note that this will most likely break DietPi scripts as well as other APT packages/programs as you mix two distros. As of your output the `base-files` package is from Ubuntu, so now every software that checks this things you run an Ubuntu system and will behave accordingly while the core binaries are still Debian.

Debian and Ubuntu are similar but not identical, and their differences raise with every distro version. Do not mix their sources, as a general recommendation! 😉

ibksturm, 2020-05-17:
:smile_cat: 

aaaah it would run :+1: