MichaIng/DietPi #3544: Image update | Odroid XU3/XU4/MC1/HC1/HC2

MichaIng, 2020-05-17:
Many thanks for testing our new image ❤️: [Download](https://dietpi.com/downloads/images/testing/DietPi_OdroidXU4-ARMv7-Buster.7z)

Please help us to estimate the stability state and give us a short report here if boot and first run setup went fine or if you faced any issue.

camatthew88, 2020-05-18:
Does not work. Stuck after crng init done. Also wifi does not work

MichaIng, 2020-05-18:
@camatthew88 
Many thanks for testing. Looks like the entropy daemon does not start as expected. I'll have a look. Meanwhile boot should continue when waiting a while (can be a few minutes) and you can speed it up by attaching a keyboard and typing some random keys, moving an attached mouse etc.

MichaIng, 2020-05-21:
Info by @Doctorbeefy: #3552

MichaIng, 2020-05-21:
If someone has time and mood for testing, I would like to know if the modern entropy daemon works fine on Odroid XU4:
1. Check if hardware random generator device exists:
```
ls -l /dev/hwrng

Doctorbeefy, 2020-05-21:
Not sure what any of this means but here is this. 

Also, after updating to 6.30 the other day I woke up this morning and tried to update my plex server when running sudo apt-get upgrade it told me my storage was full. I looked at my cloud shell screen and I noticed all of my storage values for my SD card and external storage did not have a "storage cap value" it just had how much space was used. (IE. 400mb / )Normally it shows 400mb / 29.8gig a reboot resolved this issue.

```
root@DietPi:~# dmesg | grep random
[    0.000000] random: get_random_bytes called from start_kernel+0x90/0x424 with crng_init=0
[    0.688124] random: fast init done
[    6.778971] random: systemd: uninitialized urandom read (16 bytes read)
[    6.831290] random: systemd: uninitialized urandom read (16 bytes read)
[    6.860373] random: systemd: uninitialized urandom read (16 bytes read)
[    8.824201] random: crng init done
[    8.826648] random: 7 urandom warning(s) missed due to ratelimiting
root@DietPi:~# ls -l /dev/hwrng
crw------- 1 root root 10, 183 May 21 09:16 /dev/hwrng
root@DietPi:~# G_AGP haveged
[ INFO ] APT purge for: haveged, please wait... 
(Reading database ... 22061 files and directories currently installed.)
Removing haveged (1.9.1-7) ...
(Reading database ... 22052 files and directories currently installed.)
Purging configuration files for haveged (1.9.1-7) ...
Processing triggers for systemd (241-7~deb10u4) ...
[  OK  ] APT purge for: haveged
root@DietPi:~# G_AGA
[ INFO ] APT autopurge, please wait... 
(Reading database ... 22048 files and directories currently installed.)
Removing libhavege1:armhf (1.9.1-7) ...
Processing triggers for libc-bin (2.28-10) ...
[  OK  ] APT autopurge
root@DietPi:~# G_AGI rng-tools5
[ INFO ] APT install for: rng-tools5, please wait... 
Selecting previously unselected package rng-tools5.
(Reading database ... 22042 files and directories currently installed.)
Preparing to unpack .../rng-tools5_5-4_armhf.deb ...
Unpacking rng-tools5 (5-4) ...
Setting up rng-tools5 (5-4) ...
Created symlink /etc/systemd/system/multi-user.target.wants/rngd.service → /lib/systemd/system/rngd.service.
Processing triggers for systemd (241-7~deb10u4) ...
[  OK  ] APT install for: rng-tools5

Reboot

root@DietPi:~# dmesg | grep random
[    0.000000] random: get_random_bytes called from start_kernel+0x90/0x424 with crng_init=0
[    0.688161] random: fast init done
[    6.798712] random: systemd: uninitialized urandom read (16 bytes read)
[    6.850304] random: systemd: uninitialized urandom read (16 bytes read)
[    6.876240] random: systemd: uninitialized urandom read (16 bytes read)
[    9.096501] random: crng init done
[    9.098624] random: 7 urandom warning(s) missed due to ratelimiting
```

MichaIng, 2020-05-21:
@Doctorbeefy 
> Not sure what any of this means but here is this.

Many thanks for testing. This basically means that the new version of the pure hardware random generator daemon works fine with the XU4. This is less RAM, less CPU and less disk space (both tiny anyway) expensive, and probably a better source for real randomness compared to the software-based HAVEGE algorithm. Randomness is required e.g. to create (temporary) security keys for secure communication or storage, passwords and such, which became more important for recent distro versions. It is bad if you instruct the system to create a random password or key pair and the way the random bits are derived is somehow reproducible, hence not "really" random. The entropy daemons task is to use certain hardware resources (or otherwise a proven algorithm) to produce as-real-as-possible random number and fill a pool with this (i.e. `/dev/random`) that other software can then use.

> Also, after updating to 6.30 the other day I woke up this morning and tried to update my plex server when running sudo apt-get upgrade it told me my storage was full. I looked at my cloud shell [...]

Do you mean DietPi-Cloudshell?
And you mean with the new Odroid XU4 beta image? Because there should be no update done since it is already on DietPi v6.30 when flashing it 🤔.

Doctorbeefy, 2020-05-21:
@MichaIng 

> Do you mean DietPi-Cloudshell?
Yes, I was talking about my LCD screen using it. 

> And you mean with the new Odroid XU4 beta image? Because there should be no update done since it is already on DietPi v6.30 when flashing it 🤔.

I updated from 6.29 to 6.30 and this is when the issue happened. I will check on it again tomorrow to see if it does it again and let you know. It was weird seeing it acted like every storage attached (SD card/ External HD's) were all full when they weren't.

MichaIng, 2020-05-21:
@Doctorbeefy 
Ah probably you used the previous beta image then, as I repacked it on Mai 18. However this should not influence functionality since it is based on the same firmware/kernel.

> I will check on it again tomorrow to see if it does it again and let you know.

I just checked the cloudshell script and it's output is based on `df`. If it really happens again, would be great if you could check `df` directly to see if it is a low-level issue, probably as well the `dmesg -l err,crit,alert,emerg` again to see if there are any errors with later timestamp then lets say `[    500.000000]` so that it is not something from the boot process but happened during normal operation.

Doctorbeefy, 2020-05-26:
``` ─────────────────────────────────────────────────────
 DietPi v6.30.0 (beta) : 15:01 - Tue 26/05/20
 ─────────────────────────────────────────────────────
 - Device model : Odroid XU3/XU4/MC1/HC1/HC2 (armv7l)
 - CPU temp : /boot/dietpi/func/dietpi-globals: line 1519: cannot create temp file for here-document: No space left on device
0'C : 32'F (Who put me in the freezer!)
 - LAN IP : 10.0.0.244 (eth0)
 - MOTD : Check your kernel error logs from time to time: dmesg -l err,crit,alert,emerg
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 Image           : DietPi Core Team (pre-image: Meveric)
 Web             : https://dietpi.com | https://twitter.com/dietpi_
 Patreon Legends : Bryce
 Donate          : https://dietpi.com/#donate
 DietPi Hosting  : Powered by https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@DietPi:~# df
Filesystem      1K-blocks      Used  Available Use% Mounted on
udev               951628         0     951628   0% /dev
tmpfs              204244     21892     182352  11% /run
/dev/mmcblk1p2   29894439   3415100   26475243  12% /
tmpfs             1021216         8    1021208   1% /dev/shm
tmpfs                5120         0       5120   0% /run/lock
tmpfs             1021216         0    1021216   0% /sys/fs/cgroup
tmpfs             1048576   1048576          0 100% /tmp
tmpfs               51200         8      51192   1% /var/log
/dev/mmcblk1p1     255744     45716     210028  18% /boot
/dev/sda1      1921720520 735905360 1185798776  39% /mnt/storage
root@DietPi:~# sudo apt-get upgrade
Reading package lists... Error!
E: Write error - write (28: No space left on device)
E: Write error - write (28: No space left on device)
E: The package lists or status file could not be parsed or opened.
root@DietPi:~# dmesg -l err,crit,alert,emerg
[    0.156303] CPU4: Spectre v2: firmware did not set auxiliary control register IBE bit, system vulnerable
[    0.176284] CPU5: Spectre v2: firmware did not set auxiliary control register IBE bit, system vulnerable
[    0.184612] CPU6: Spectre v2: firmware did not set auxiliary control register IBE bit, system vulnerable
[    0.192608] CPU7: Spectre v2: firmware did not set auxiliary control register IBE bit, system vulnerable
[    1.874393] exynos-hdmi 14530000.hdmi: Failed to get supply 'vdd': -517
[    1.957251] devfreq 11800000.mali: Couldn't update frequency transition information.
[    3.470420] s5p_mfc_load_firmware:73: Firmware is not present in the /lib/firmware directory nor compiled in kernel
[    3.889464] OF: graph: no port node found in /soc/hdmi@14530000
[    8.614423] s5p_mfc_load_firmware:73: Firmware is not present in the /lib/firmware directory nor compiled in kernel
[    8.633804] s5p_mfc_load_firmware:73: Firmware is not present in the /lib/firmware directory nor compiled in kernel
[    8.898491] sd 0:0:0:0: [sda] No Caching mode page found
[    8.902944] sd 0:0:0:0: [sda] Assuming drive cache: write through
[  219.762490] s5p_mfc_load_firmware:73: Firmware is not present in the /lib/firmware directory nor compiled in kernel
[  219.784279] s5p_mfc_load_firmware:73: Firmware is not present in the /lib/firmware directory nor compiled in kernel
root@DietPi:~# ```

Doctorbeefy, 2020-05-26:
@MichaIng 
I noticed the temp is way low as well.  Here are screenshots from the cloudshell LCD as well (Don't Mind the dust)

Also this started happening with DietPi version: 6.30.0. It didn't seem to happen with DietPi version: 6.29

[image]
[image]
[image]
[image]

Joulinar, 2020-05-26:
Hi,

did you remove some files and tried to clean `/tmp` ??

Doctorbeefy, 2020-05-26:
@Joulinar No I haven't done anything, this only happens randomly after running for a day or so.

Joulinar, 2020-05-26:
maybe something is filling up `/tmp` and did not get remove. can you check what is located in `/tmp`?

Doctorbeefy, 2020-05-26:
@Joulinar A ton of crap is in /tmp such as 

```0000706a-e090-4d85-955f-4d5f8859d1f9  55d7f3df-4ed3-4e8c-b37f-bf3046ce0f83  aafe1b29-8bcd-4b41-b3b6-fd87cec43f75
000a5373-993d-41b5-846d-49141381cf77  55debdcc-e838-44e7-aa68-2d63981d8d03  ab03697f-aab6-4da1-8ce9-238002e3d996
00111bd6-79d2-46ed-a0b4-1d6b19d8b541  55e4c56e-d634-4cf9-8594-60ac63cefc8d  ab04c667-a76c-45be-ad09-452fbe1b0302
0016fc18-737e-4d3e-859a-82fac5b99952  55e955d0-be16-48f4-ab62-17c966ee7f98  ab096d88-624f-4ff5-937f-321b43074368
001c174f-589d-4fd8-adbc-29d2a7631a70  55edc049-018f-45f3-9ff9-9d02a1028e7d  ab115248-0720-43e2-9f9f-5a209e6c700a
001d84f1-5646-4bfa-9605-be17b1fc266b  55f875e1-db48-4661-ae36-24bf98c1c73c  ab188456-3b1d-4a4b-aad1-1aefdd22a9d3
00255bc0-3135-4b9a-9b66-2f66e906e7cb  55f8c893-2a4a-4001-92c8-0c722560e888  ab32e2c8-31bf-46e0-b66c-c613141dae29
```

Doctorbeefy, 2020-05-26:
@Joulinar I would like to note as well a reboot will fix this issue, I have been waiting though to do this until troubleshooting is done.

Joulinar, 2020-05-26:
for sure a reboot will fix it, because `/tmp` is not a persistent directory. It's a `tmpfs`, located in memory and will be freshly created during boot. 

hmm plex is creating these files. Seems to be used for transcoding. I found an interesting discussion on plex forum about `/tmp` usage. It seems some others having similar issues 

https://forums.plex.tv/t/pms-stopped-honoring-temporary-directory-setting-now-uses-tmp/594501/20

A possible solutions is offered as well, to create an additional `override.conf` and to specify an alternative path for the temp file location to avoid filling up `/tmp`

Doctorbeefy, 2020-05-26:
@Joulinar great catch looks like they messed up something. Should I delete all my extra crap here?

Joulinar, 2020-05-26:
I would go to try to implement the workaround using the additional `override.conf` and reboot your system afterwards. You can point with the plex tmp dir to some other location. maybe on your external device.

Doctorbeefy, 2020-05-26:
@Joulinar it looks like the easiest fix is to disable the new intro skip they implemented. It looks like that is storing data in tmp for every intro.

Joulinar, 2020-05-26:
ok, honestly I'm not that familiar with plex functionality and usage. at least you could disable the into function and check how `/tmp` will behave.

MichaIng, 2020-05-27:
I've just overseen the 100% /tmp usage. Good find, seems Plex currently has an issue of not cleaning up temporary files. Regardless if `/tmp` is a tmpfs or not, this must not happen of course. Using an alternative location could make it even worse, filling up the SDcard which might make the whole system break + adds SDcard wear.

I'll have a look into the other error, but I guess it is minor or expected when not having certain feature packages installed:
```
s5p_mfc_load_firmware:73: Firmware is not present in the /lib/firmware directory nor compiled in kernel
```

Joulinar, 2020-05-27:
@MichaIng 
You are right, moving Plex temp to SD card could make thinks worse if it's filling up completely. As of now we don't know how much space would be needed. Anyway we hijacked this topic now as it was initially intent for the new image but turned into the Plex discussion. Should me move to a new one?

MichaIng, 2020-05-27:
Lets go on with this issue here: https://github.com/MichaIng/DietPi/issues/3568

Sudrien, 2020-05-28:
Got a new HC1 for the cluster -

Oddly, the (three?) IPv6 dns9.quad9.net pings were failing for me, but forcing 9.9.9.9 did not. I presume it's due to my network VPN setup. Otherwise setup went smoothly.

Any particular reason dietpi.com isn't being used for those?

MichaIng, 2020-05-29:
@Sudrien 
Many thanks for testing.

Not sure, somehow pinging any domain that resolves with IPv6 currently fails from my network as well with timeout, while running curl to connect to the webserver on the same IPv6 works pretty fine. I tested the same from our server and observed a significant time delay when pinging to IPv6 before the first answer arrives, around 3 seconds.

Just to verify, if you run
```
ping dns9.quad9.net
```
it resolves to an IPv6 IP?
And does it receive any package at all when waiting long, else which error?
```
ping -4 dns9.quad9.net
```
instead works fine?

Generally when IPv6 works slow I would simply disable it, but ping/ICMPv6 seems to behave somehow different then HTTP(S).

Another try while playing around from dietpi.com server:
```
2020-05-29 09:40:59 root@dietpi:~# ping -v6 google.com
ping: socket: Permission denied, attempting raw socket...
PING google.com(ams15s30-in-x0e.1e100.net (2a00:1450:400e:807::200e)) 56 data bytes
From dietpi.com (XXX) icmp_seq=9 Destination unreachable: Address unreachable
From dietpi.com (XXX) icmp_seq=10 Destination unreachable: Address unreachable
From dietpi.com (XXX) icmp_seq=11 Destination unreachable: Address unreachable
64 bytes from ams15s30-in-x0e.1e100.net (2a00:1450:400e:807::200e): icmp_seq=22 ttl=54 time=2062 ms
64 bytes from ams15s30-in-x0e.1e100.net (2a00:1450:400e:807::200e): icmp_seq=23 ttl=54 time=1038 ms
64 bytes from ams15s30-in-x0e.1e100.net (2a00:1450:400e:807::200e): icmp_seq=24 ttl=54 time=14.3 ms
```

Another interesting attempt:
```
2020-05-29 09:45:14 root@dietpi:~# curl -vIL google.com
* Rebuilt URL to: google.com/
*   Trying 2a00:1450:400e:807::200e...
* TCP_NODELAY set
*   Trying 172.217.17.142...
* TCP_NODELAY set
* Connected to google.com (172.217.17.142) port 80 (#0)
> HEAD / HTTP/1.1
> Host: google.com
> User-Agent: curl/7.52.1
> Accept: */*
>
< HTTP/1.1 301 Moved Permanently
HTTP/1.1 301 Moved Permanently
< Location: http://www.google.com/
Location: http://www.google.com/
< Content-Type: text/html; charset=UTF-8
Content-Type: text/html; charset=UTF-8
< Date: Fri, 29 May 2020 08:45:28 GMT
Date: Fri, 29 May 2020 08:45:28 GMT
< Expires: Sun, 28 Jun 2020 08:45:28 GMT
Expires: Sun, 28 Jun 2020 08:45:28 GMT
< Cache-Control: public, max-age=2592000
Cache-Control: public, max-age=2592000
< Server: gws
Server: gws
< Content-Length: 219
Content-Length: 219
< X-XSS-Protection: 0
X-XSS-Protection: 0
< X-Frame-Options: SAMEORIGIN
X-Frame-Options: SAMEORIGIN

<
* Curl_http_done: called premature == 0
* Connection #0 to host google.com left intact
* Issue another request to this URL: 'http://www.google.com/'
*   Trying 2a00:1450:400e:804::2004...
* TCP_NODELAY set
*   Trying 172.217.17.36...
* TCP_NODELAY set
* Connected to www.google.com (172.217.17.36) port 80 (#1)
> HEAD / HTTP/1.1
> Host: www.google.com
> User-Agent: curl/7.52.1
> Accept: */*
>
< HTTP/1.1 200 OK
HTTP/1.1 200 OK
< Content-Type: text/html; charset=ISO-8859-1
Content-Type: text/html; charset=ISO-8859-1
< P3P: CP="This is not a P3P policy! See g.co/p3phelp for more info."
P3P: CP="This is not a P3P policy! See g.co/p3phelp for more info."
< Date: Fri, 29 May 2020 08:45:28 GMT
Date: Fri, 29 May 2020 08:45:28 GMT
< Server: gws
Server: gws
< X-XSS-Protection: 0
X-XSS-Protection: 0
< X-Frame-Options: SAMEORIGIN
X-Frame-Options: SAMEORIGIN
< Transfer-Encoding: chunked
Transfer-Encoding: chunked
< Expires: Fri, 29 May 2020 08:45:28 GMT
Expires: Fri, 29 May 2020 08:45:28 GMT
< Cache-Control: private
Cache-Control: private
< Set-Cookie: 1P_JAR=2020-05-29-08; expires=Sun, 28-Jun-2020 08:45:28 GMT; path=/; domain=.google.com; Secure
Set-Cookie: 1P_JAR=2020-05-29-08; expires=Sun, 28-Jun-2020 08:45:28 GMT; path=/; domain=.google.com; Secure
< Set-Cookie: NID=204=Lex1GjQ9NTGU1r7OKBmE25r_KEACqtEcP-BYcXcGbaCFM490EHMw1Psr7ElEwadUigmYkRrEgrN2CD2H39jSP0Hi7sGZa3KJl0E4kmguYGAPyrS1UgsMKU-lRJx1x5fgIIgm4i4d-xGEPOByHglPYg_oF6nprzTV_HEoRS2uAJ4; expires=Sat, 28-Nov-2020 08:45:28 GMT; path=/; domain=.google.com; HttpOnly
Set-Cookie: NID=204=Lex1GjQ9NTGU1r7OKBmE25r_KEACqtEcP-BYcXcGbaCFM490EHMw1Psr7ElEwadUigmYkRrEgrN2CD2H39jSP0Hi7sGZa3KJl0E4kmguYGAPyrS1UgsMKU-lRJx1x5fgIIgm4i4d-xGEPOByHglPYg_oF6nprzTV_HEoRS2uAJ4; expires=Sat, 28-Nov-2020 08:45:28 GMT; path=/; domain=.google.com; HttpOnly

<
* Curl_http_done: called premature == 0
* Connection #1 to host www.google.com left intact
```
Host resolves to IPv6 but curl quickly switches to IPv6. Forcing IPv6 (`-6`) breaks the connection:
```
2020-05-29 09:43:24 root@dietpi:~# curl -IL -6 google.com
curl: (7) Failed to connect to google.com port 80: No route to host
```

It's a pain, the DNS lookup to get the IPv6 address has no issue and works immediately but connections are somewhat broken or unstable or slow. I'll do further tests...

closed 2020-06-01 by MichaIng