MichaIng/DietPi #3548: /boot/dietpi/func/dietpi-banner: line 222: /run/dietpi/.dietpi_motd: Permission denied

Sopor, 2020-05-19:
### Creating a bug report/issue
I updated DietPi today and now i will see this message when i run `diet-pi-login`
```
 ─────────────────────────────────────────────────────
 DietPi v6.30.0 : 00:18 - Wed 20/05/20
 ─────────────────────────────────────────────────────
 - Device model : RPi 2 Model B (armv7l)
 - Uptime : up 27 weeks, 1 day, 21 hours, 55 minutes
 - CPU temp : 44'C : 111'F (Optimal temperature)
 - LAN IP : 192.168.0.10 (eth0)
/boot/dietpi/func/dietpi-banner: line 222: /run/dietpi/.dietpi_motd: Permission denied
 ─────────────────────────────────────────────────────
```
#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=30
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
stretch | 9.11

- Kernel version | `uname -a`
Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
< EMPTY > | RPi2
- Power supply used | (EG: 5V 1A RAVpower)
I don't know
- SDcard used | (EG: SanDisk ultra)
SanDisk

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Run `/DietPi/dietpi/dietpi-login`
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2020-05-19:
hi,

many thanks for your report. Pls can you try to login as user `root`

Sopor, 2020-05-19:
If i run as sudo it seems to work
```
 MOTD : Check your kernel error logs from time to time: dmesg -l err,crit,alert,emerg
```

Pisgah, 2020-05-20:
An answer may be found here : 
https://dietpi.com/phpbb/viewtopic.php?f=10&t=7637&p=24644&hilit=banner#p24644

See the links inside.

MichaIng, 2020-05-20:
The issue is solved with a reboot, indeed the correct directory permissions were not applied with update directly (but via tmpfiles from next boot on). This solves it (otherwise a reboot):
```
chmod 777 /run/dietpi
```
Enhanced for next update: https://github.com/MichaIng/DietPi/commit/99caec8ea8cd043a16fe8e59319d729acec75d28

Sopor, 2020-05-22:
@MichaIng Now when i run `diet-pi-login` i do get a motd. I haven't done anything and the chmode is 0644 on `.dietpi_motd` and 0666 on `.network`. Why does it work now?

MichaIng, 2020-05-22:
@Sopor 
Yes it is only a one-time issue during the update. I hence mark this issue as closed. Feel free to reopen if required.

closed 2020-05-22 by MichaIng

Sopor, 2020-05-24:
@MichaIng Yes, but i haven't rebooted or even logged out from my terminal, so how can it suddenly work now?

MichaIng, 2020-05-25:
@Sopor 
I guess the MOTD was not yet downloaded when banner was called with non-root user. Then you called it with sudo, so the MOTD was downloaded and the non-root user didn't need to anymore. Daily at night the current MOTD is removed and freshly downloaded via cron job as root user, so then the issue cannot occur anymore. On next reboot the tmpfiles creation applies the correct permissions.

Sopor, 2020-05-25:
Thanks for explaining, @MichaIng ...