MichaIng/DietPi #3549: bluetooth config for HCIUART.service i have console ttyAMA0 opened but i cannot connect

sealteam666, 2020-05-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2020-05-20:
@sealteam666 
I guess you're on RPi3 or RPi4 or RPi Zero W, right?

Generally a UART device can be used only EITHER as serial console OR for Bluetooth. By default on RPi3/4/Zero W the full UART device ttyAMA0 is used for Bluetooth, hence a serial console must not be attached to it in parallel. Please do the following to disable the serial console on ttyAMA0, respectively assure it is disabled:
```
dietpi-config > Advanced Options > Serial/UART > ttyAMA0 > disable
```
Then enable Bluetooth:
```
dietpi-config > Advanced Options > Bluetooth > enable
```

sealteam666, 2020-05-20:
After lot of research and waking up entire night i found out that pi bluetooth try to start hciuart and fail with error dependency fail . I am  using raspberry pi 2 with bluetooth dongle. I thought hciuart is used to turn on bluetooth dongle but it doesn't because it is for UART . Later i found that my dongle is fake made with CSR 4.0 DRIVERS with `btmon` command i found on some ubuntu stack exchange . So my request is please do something about pi-bluetooth so that noobs like me can understand when error occurs with hciuart. Thanks for replying

closed 2020-05-20 by sealteam666