MichaIng/DietPi #3550: Uninstalling Docker deletes volumes

kpine, 2020-05-20:
### Creating a bug report/issue
When using `dietpi-software` to uninstall Docker, it purges all the packages and [deletes ](https://github.com/MichaIng/DietPi/blob/d29a5af277aa1e6b76b4b3057968f0885894e61d/dietpi/dietpi-software#L14098) the userdata (`$G_FP_DIETPI_USERDATA/docker-data`). The userdata also includes volumes that store persistent data for the containers.

This only applies to volumes, not bind mounts.

#### Required Information
- DietPi version
```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=30
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `buster`
- Kernel version | `Linux hasspi.int.chabo.io 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux`
- SBC model | `RPi 3 Model B (armv7l)`
- Power supply used | OEM
- SDcard used | Samsung

#### Additional Information (if applicable)
- Software title | Docker
- Was the software title installed freshly or updated/migrated? Originally installed using `dietpi-software`.
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `42daab74-e359-43cd-ae4b-1f818b10210d`

#### Steps to reproduce
1. Install docker via dietpi-software
2. Create a named volume, notice the volume is created in the userdata directory
3. Uninstall docker via dietpi-software
4. Notice the volume has been completed removed

#### Expected behaviour
The uninstall should not delete the volumes. Or maybe dietpi-software could provide an extra warning that uninstalling will permanently remove user data, ideally by checking for the existence of any named volumes.

#### Actual behaviour
dietpi-software deletes all of the volumes.

#### Extra details
I noticed my docker repo was set to buster nightly, which is not updated any longer. This was probably the result of installing it long ago. Instead of just modifying the source list file, I thought I should uninstall and re-install using the official method which uses the stable repo. I was not aware that uninstalling would also delete the data.

MichaIng, 2020-05-20:
@kpine 
Many thanks for your report and I'm sorry for your possible data loss.

Indeed in most cases, uninstalling the program via `dietpi-software` removes all related data per design. I agree that there should be an info about this, since e.g. removing an APT package (instead of purging) keeps configs and data, even purging keeps userdata often.

When you want to repair an instance, reinstalling it would be the way to go (i.e. `dietpi-software reinstall 162` for Docker), but that is not jep available through the GUI.

Ideally when uninstalling software, one could be presented an additional menu to select (or unselect) removing userdata per-software.

I'll add the information for now, with DietPi v6.31,

kpine, 2020-05-20:
Thank you for the courteous reply. I agree with all your points.

If it's normal and consistent for Diet-Pi to remove all user data upon uninstall than that is fair. As you said, I was expecting it to be like removing the packages via apt. So anything to clarify that behavior would be helpful. In the end, I have backups (as one should!), so nothing much was lost, just a bit of time.

The `reinstall` command is good to know, would have been perfect here.

MichaIng, 2020-07-15:
I added the info: https://github.com/MichaIng/DietPi/commit/032f9262bb371a5e0d57812e53495e42c010b1db
Minor format tweak: https://github.com/MichaIng/DietPi/commit/9559ca28fcf5ee056e4eea578b98544600c91bee
Changelog: https://github.com/MichaIng/DietPi/commit/d6130d37eebdbee1f7aa5a175c3547b6e55e64d7
[image]

closed 2020-08-31 by MichaIng