MichaIng/DietPi #3551: Plex.service still trying to run after unistall

thecarlosv, 2020-05-20:
After unistalling plex, the service still tries to go active. Already tried 
root@CarlosPi:~# systemctl stop plexmediaserver
root@CarlosPi:~# apt-get purge plexmediaserver -y
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
E: No se ha podido localizar el paquete plexmediaserver

but i get that error. 

9e34f6c6-7ed5-4c77-9758-11706481f1e7

MichaIng, 2020-05-20:
@thecarlosv 
Many thanks for your report.

Indeed I see `/lib/systemd/system/plexmediaserver.service` is still present, which of course purging the package should have removed it. We have to forward this to Plex devs and implement a workaround until it has been resolved upstream.

Please do the following:
```
systemctl unmask plexmediaserver
systemctl disable --now plexmediaserver
rm -R /lib/systemd/system/plexmediaserver.service*
```

thecarlosv, 2020-05-20:
Thanks, it works!

closed 2020-05-20 by thecarlosv