MichaIng/DietPi #3552: Odroid Xu4 6.30.0 Beta Build Kernel Error Logs

Doctorbeefy, 2020-05-21:
Just posting them here as I saw the MOTD

```root@DietPi:~# dmesg -l err,crit,alert,emerg
[    0.156291] CPU4: Spectre v2: firmware did not set auxiliary control register IBE bit, system vulnerable
[    0.176285] CPU5: Spectre v2: firmware did not set auxiliary control register IBE bit, system vulnerable
[    0.184626] CPU6: Spectre v2: firmware did not set auxiliary control register IBE bit, system vulnerable
[    0.192592] CPU7: Spectre v2: firmware did not set auxiliary control register IBE bit, system vulnerable
[    1.873071] exynos-hdmi 14530000.hdmi: Failed to get supply 'vdd': -517
[    1.955956] devfreq 11800000.mali: Couldn't update frequency transition information.
[    3.470432] s5p_mfc_load_firmware:73: Firmware is not present in the /lib/firmware directory nor compiled in kernel
[    3.889272] OF: graph: no port node found in /soc/hdmi@14530000
[    8.435748] s5p_mfc_load_firmware:73: Firmware is not present in the /lib/firmware directory nor compiled in kernel
[    8.455010] s5p_mfc_load_firmware:73: Firmware is not present in the /lib/firmware directory nor compiled in kernel
[    8.739386] sd 0:0:0:0: [sda] No Caching mode page found
[    8.745534] sd 0:0:0:0: [sda] Assuming drive cache: write through```

Doctorbeefy, 2020-05-21:
I do want to note I haven't noticed any issues.

MichaIng, 2020-05-21:
I'm closing this in favour of #3544, many thanks.

closed 2020-05-21 by MichaIng