MichaIng/DietPi #3553: Auto Setup not running

DanrwAU, 2020-05-21:
### Creating a bug report/issue

#### Required Information
- DietPi version | Latest Buster Rpi image
- SBC model | Rpi 4+ 2GB
- Power supply used | Official Pi4 PSU
- SDcard used | Sandisk something

#### Steps to reproduce
Using AUTO_SETUP_AUTOMATED=1 results in the setup hanging until i SSH in. 

Some part of this process completes however, as my password set in the file is able to be used to login 

The first steps completed after login are visible in the screenshot 
[image]

#### Expected behaviour
Setup completes automatically without requiring SSH

#### Extra details

Copy of my dietpi.txt 
````
# IMPORTANT:
# - This is intended for advanced users, unless you know what you are doing, do not edit this file. Please use the DietPi programs instead.
# - Modifications to /boot/dietpi.txt will not be preserved on reboot.
# - Please ensure you edit from the DietPi-RAMdisk location: /DietPi/dietpi.txt
# - Do not remove uncommented lines, as the items are scraped by DietPi programs, on demand.

#------------------------------------------------------------------------------------------------------
# D I E T - P I
# DietPi-Automation settings, applied on first boot of DietPi only, ONCE!
#------------------------------------------------------------------------------------------------------
##### Language/Regional Options #####
# Locale: eg: "en_GB.UTF-8" / "de_DE.UTF-8" | One entry and UTF-8 ONLY!
AUTO_SETUP_LOCALE=en_AU.UTF-8

# Keyboard Layout eg: "gb" / "us" / "de" / "fr"
AUTO_SETUP_KEYBOARD_LAYOUT=us

# Timezone eg: "Europe/London" / "America/New_York" | Full list (TZ*): https://en.wikipedia.org/wiki/List_of_tz_database_time_zones
AUTO_SETUP_TIMEZONE=Australia/Melbourne

##### Networking Options #####
# Enable Ethernet or WiFi adapter: 1=enable | 0=disable
# - If both Ethernet and WiFi are enabled, WiFi will take priority and Ethernet will be disabled.
# - If using WiFi, please edit dietpi-wifi.txt to pre-enter credentials.
AUTO_SETUP_NET_ETHERNET_ENABLED=0
AUTO_SETUP_NET_WIFI_ENABLED=1

# Force ethernet speeds: 0=automatic speed | 10 = 10Mbit | 100 = 100Mbit etc
# - This is mainly aimed at Pine A64 which may have an HW issue that causes unstable 1Gbit link.
AUTO_SETUP_NET_ETH_FORCE_SPEED=0

# Enter your Static Network details below, if applicable.
AUTO_SETUP_NET_USESTATIC=0
AUTO_SETUP_NET_STATIC_IP=192.168.0.100
AUTO_SETUP_NET_STATIC_MASK=255.255.255.0
AUTO_SETUP_NET_STATIC_GATEWAY=192.168.0.1
AUTO_SETUP_NET_STATIC_DNS=8.8.8.8

# Hostname
AUTO_SETUP_NET_HOSTNAME=DietPi-6

##### Misc Options #####
# Swapfile size to generate: 0=disable | 1=auto (2GB-RAM = size) | 2+=manual (MB)
AUTO_SETUP_SWAPFILE_SIZE=1
# Swapfile location
AUTO_SETUP_SWAPFILE_LOCATION=/var/swap

[... 184 lines cut ...]

# Chromium
SOFTWARE_CHROMIUM_RES_X=1280
SOFTWARE_CHROMIUM_RES_Y=720
SOFTWARE_CHROMIUM_AUTOSTART_URL=https://dietpi.com

# Home Assistant
# - Optional Python build dependencies and modules, possibly required for certain HA components
#	Space separated list (no quotation!), will be installed together with Home Assistant automatically, if present
SOFTWARE_HOMEASSISTANT_APT_DEPS=
#	Add Python modules with version string at best, e.g.: firstModule==1.2.3 secondModule==4.5.6
SOFTWARE_HOMEASSISTANT_PIP_DEPS=

#------------------------------------------------------------------------------------------------------
# D I E T - P I
# Dev settings
#------------------------------------------------------------------------------------------------------
DEV_GITBRANCH=master
DEV_GITOWNER=MichaIng

#------------------------------------------------------------------------------------------------------
# D I E T - P I
# Settings, automatically added by dietpi-update
#------------------------------------------------------------------------------------------------------
````

MichaIng, 2020-05-21:
@DanrwAU 
Many thanks for your report.

Indeed by design while the first non-network part of the first boot setup is done completely non-interactive, the update and software installs are only done after logging in one time. The request to automate it fully already exists: #2520

It basically requires some careful testing as commands might behave different when not being in an actual user session. Otherwise we could try to auto-login but this requires changing a systemd unit, hence reloading the init daemon during its init process, which might be not 100% failsafe. However lets keep discussion in the linked thread.

DanrwAU, 2020-05-21:
Interesting, I thought this was working previously.

I guess I am nuts 😂

MichaIng, 2020-05-21:
@DanrwAU 
No problem, it is IMO indeed a flaw since automated really should mean automated and not "manual login still required" 😉.

closed 2020-05-21 by MichaIng