MichaIng/DietPi #3554: shairport-sync now crashes when invoked (v6.30)

brucefikkers, 2020-05-21:
apt-get install shairport-sync 

fixes the issue....

MichaIng, 2020-05-21:
@audiobruce 
Can you please give some more information:
1. Which SBC/device are you using, which CPU architecture?
2. What was the error message?
3. Please add some error log: `journalctl -u shairport-sync`

audiobruce, 2020-05-22:
Hi,

1. RPi 4B
2. shairport was available on network but as soon as I connected to send audio it disappeared. restarted pi / shairport several times and it did the same thing.
3. sorry no logs and I've now fixed it by reinstalling shairport!

Previous DietPi version (not sure which one - around a month ago) was fine.

Best regards,
BRuce

MichaIng, 2020-05-22:
@audiobruce 
Okay, we cannot debug something now but I'll leave this issue open if someone else faces the same and since you provide a solution, although this means a version downgrade.

MichaIng, 2020-07-05:
I mark this as closed, feel free to reopen if required.

closed 2020-07-05 by MichaIng