MichaIng/DietPi #3556: Squeezelite 1.9 on dietpi by default

MattL0, 2020-05-22:
curently dietpi install an old 1.8 package. Could this be bumped to squeezelite 1.9.7?

Thanks

https://github.com/ralph-irving/squeezelite

MichaIng, 2020-05-22:
@MattL0 
Many thanks for your request.

I just checked the Debian repo and there no upstream release has been merged since Stretch as well: https://packages.debian.org/bullseye/squeezelite

Looks like they provide pre-compiled binaries for x86_64 and RPi at least: https://sourceforge.net/projects/lmsclients/files/squeezelite/linux/
The others (ARMv7/armhf + ARMv8/arm64) need to be compiled. I see if I find time to do that for v6.31. If anyone else finds time, feel free to do a start 😉.

MattL0, 2020-05-28:
It was pretty easy to change the squeezéite version on the pi.

Now my speaker are more in sync then before .

MichaIng, 2020-07-01:
New squeezelite 1.9.7 packages are ready for testing:
```
systemctl disable --now squeezelite
mv /etc/systemd/system/squeezelite.service{,_bak}
cd /tmp
wget https://dietpi.com/downloads/binaries/$G_DISTRO_NAME/squeezelite_$G_HW_ARCH_NAME.deb
apt install ./squeezelite_$G_HW_ARCH_NAME.deb
rm squeezelite_$G_HW_ARCH_NAME.deb
```
- SSL support has been additionally enabled, OPUS would be additionally possible and especially for RPi there is GPIO support possible. Let me know if you think it should be compiled in. For now its like the Debian packages + SSL.
- The service runs now as own user "squeezelite", so especially check if any permissions are missing. It is added to "audio" group, so generally audio device access is granted.
- Logging to file has been disabled btw. Generally I prefer to use logging to journal where possible since journald is running anyway and is a native RAMlog. Logs are available hence via: `journalctl -u squeezelite`

MattL0, 2020-07-01:
Works great!! Thanks

I have both squeezelite and shairport sync ( compiled it myself) on my 3 raspberry pi. 
1. **I had to rename each one here naturally /lib/systemd/system/squeezelite.service**
2 **I had to remove avahi-daemon** ( was able to see every squeezelite on squeezebox like that) **and then reinstalled it** ( to be able to see my shairport device  in a HomeSeer plugin ( airplayspeak.)

MichaIng, 2020-07-01:
> I had to rename each one here naturally /lib/systemd/system/squeezelite.service

Yes that is required indeed. Btw best is to create a drop-in config in `/etc/systemd/system/squeezelite.service.d/your.conf`. Easiest can be done via `dietpi-services` > `squeezelite` > `Edit`.
Once released I'll update the docs accordingly. Probably for next release we can add a config file like `/etc/default/squeezelite` to allow a more comfortable adjustment of command line options, and the package postinst script could automatically add some unique device name like `DietPi-SqueezeLite_8acf9` so simply `SqueezeLite_8acf9` with unique random (or taken from UUID) suffix.

> I had to remove avahi-daemon and then reinstalled it

Can you explain a bid more that the issue is with enabled Avahi daemon? I don't use any actively myself but generally SqueezeLite should be able to detect and connect to the LMS automatically, else there is an `-s` option to add the IP/hostname of the LMS to SqueezeLite via command line option. Since Avahi only broadcasts the hostname, I guess otherwise there is a hostname conflict?

closed 2020-07-02 by MichaIng

MattL0, 2020-07-24:
Just curious, will this work on the x64 version of dietpi on the pi?

MichaIng, 2020-07-24:
Yes, we have binaries for arm64 as well: https://dietpi.com/downloads/binaries/buster/squeezelite_aarch64.deb

To get the correct one regardless of distro version and architecture: `https://dietpi.com/downloads/binaries/$G_DISTRO_NAME/squeezelite_$G_HW_ARCH_NAME.deb`
Or install via dietpi-software.