MichaIng/DietPi #3557: KVM problem

Dreaming-Codes, 2020-05-22:
I have no other way of accessing vps than ssh and vnc so I can't know what goes wrong with the boot

Joulinar, 2020-05-22:
Hi,

many thanks for your report. Well, if we don't have any logs and you are not able to access the system, it will be quite hard to assist. Maybe you can ask your VPS provider of there are any other console to access the VM layer.

MichaIng, 2020-05-23:
@Dreaming-Codes 
Many thanks for your report.

Have you gone through the steps on the other issue already? https://github.com/MichaIng/DietPi/issues/1285#issuecomment-630510940

It is basically about LVM support: #2995
Lets keep the discussion there, so I mark this as closed.

closed 2020-05-23 by MichaIng