MichaIng/DietPi #3558: dietpi-update failed rp 4

jokoren, 2020-05-23:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...
#### Details:
- Date           | Fri May 22 22:36:51 EDT 2020
- DietPi version | v6.29.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux rv 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
(Reading database ... 32192 files and directories currently installed.)
Preparing to unpack .../00-bind9-host_1%3a9.11.5.P4+dfsg-5.1+deb10u1_armhf.deb ...
Unpacking bind9-host (1:9.11.5.P4+dfsg-5.1+deb10u1) over (1:9.11.5.P4+dfsg-5.1) ...
Preparing to unpack .../01-libbind9-161_1%3a9.11.5.P4+dfsg-5.1+deb10u1_armhf.deb ...
Unpacking libbind9-161:armhf (1:9.11.5.P4+dfsg-5.1+deb10u1) over (1:9.11.5.P4+dfsg-5.1) ...
Preparing to unpack .../02-libisccfg163_1%3a9.11.5.P4+dfsg-5.1+deb10u1_armhf.deb ...
Unpacking libisccfg163:armhf (1:9.11.5.P4+dfsg-5.1+deb10u1) over (1:9.11.5.P4+dfsg-5.1) ...
Preparing to unpack .../03-libisccc161_1%3a9.11.5.P4+dfsg-5.1+deb10u1_armhf.deb ...
Unpacking libisccc161:armhf (1:9.11.5.P4+dfsg-5.1+deb10u1) over (1:9.11.5.P4+dfsg-5.1) ...
Preparing to unpack .../04-libdns1104_1%3a9.11.5.P4+dfsg-5.1+deb10u1_armhf.deb ...
Unpacking libdns1104:armhf (1:9.11.5.P4+dfsg-5.1+deb10u1) over (1:9.11.5.P4+dfsg-5.1) ...
Preparing to unpack .../05-libisc1100_1%3a9.11.5.P4+dfsg-5.1+deb10u1_armhf.deb ...
Unpacking libisc1100:armhf (1:9.11.5.P4+dfsg-5.1+deb10u1) over (1:9.11.5.P4+dfsg-5.1) ...
Preparing to unpack .../06-liblwres161_1%3a9.11.5.P4+dfsg-5.1+deb10u1_armhf.deb ...
Unpacking liblwres161:armhf (1:9.11.5.P4+dfsg-5.1+deb10u1) over (1:9.11.5.P4+dfsg-5.1) ...
Preparing to unpack .../07-libisc-export1100_1%3a9.11.5.P4+dfsg-5.1+deb10u1_armhf.deb ...
Unpacking libisc-export1100:armhf (1:9.11.5.P4+dfsg-5.1+deb10u1) over (1:9.11.5.P4+dfsg-5.1) ...
Preparing to unpack .../08-libdns-export1104_1%3a9.11.5.P4+dfsg-5.1+deb10u1_armhf.deb ...
Unpacking libdns-export1104 (1:9.11.5.P4+dfsg-5.1+deb10u1) over (1:9.11.5.P4+dfsg-5.1) ...
Preparing to unpack .../09-rpi-eeprom-images_7.0-1_all.deb ...
Unpacking rpi-eeprom-images (7.0-1) over (6.0-1) ...
Preparing to unpack .../10-rpi-eeprom_7.0-1_all.deb ...
Unpacking rpi-eeprom (7.0-1) over (6.0-1) ...
Setting up libisc1100:armhf (1:9.11.5.P4+dfsg-5.1+deb10u1) ...
Setting up liblwres161:armhf (1:9.11.5.P4+dfsg-5.1+deb10u1) ...
Setting up libisc-export1100:armhf (1:9.11.5.P4+dfsg-5.1+deb10u1) ...
Setting up libisccc161:armhf (1:9.11.5.P4+dfsg-5.1+deb10u1) ...
Setting up htpdate (1.2.0-3) ...
Job for htpdate.service failed because the control process exited with error code.
See "systemctl status htpdate.service" and "journalctl -xe" for details.
invoke-rc.d: initscript htpdate, action "start" failed.

jokoren, 2020-05-23:
htpdate.service - HTTP based time synchronization tool
   Loaded: loaded (/lib/systemd/system/htpdate.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-05-22 22:36:46 EDT; 49ms ago
     Docs: man:htpdate
  Process: 3138 ExecStart=/usr/sbin/htpdate $HTP_OPTIONS $HTP_PROXY -i /run/htpdate.pid $HTP_SERVERS (code=exited, status=226/NAMESPACE)

May 22 22:36:46 rv systemd[1]: Starting HTTP based time synchronization tool...
May 22 22:36:46 rv systemd[3138]: htpdate.service: Failed to set up mount namespacing: No such file or directory
May 22 22:36:46 rv systemd[3138]: htpdate.service: Failed at step NAMESPACE spawning /usr/sbin/htpdate: No such file or directory

MichaIng, 2020-05-23:
@jokoren 
Many thanks for your report.

> /usr/sbin/htpdate: No such file or directory

Hmm, does this file exist?
```
ls -l /usr/sbin/htpdate
```

And is there a special reason you use this HTTP-based time sync daemon? Usually `systemd-timesyncd` (default on DietPi) works pretty fine to sync time with usual NTP servers, at best the local network router/gateway which often provides cached NTP for the whole network.

If you want or need htpdate, try to reinstall it to recover the file that seems to be missing:
```
apt install --reinstall htpdate
```
Else, I suggest to purge it and fallback to systemd-timesyncd:
```
apt purge htpdate
/DietPi/dietpi/func/dietpi-set_software ntpd-mode 2 # Time sync daily and on boot
```

jokoren, 2020-05-23:
Fixed it by 

dietpi-config
  advanced options
    Time sync mode
      [custom]

Fresh install on 3 pi's was set for boot+daily

Jim O'Koren 
Retired software engineer

Hmm, does this file exist?
ls -l /usr/sbin/htpdate

And is there a special reason you use this HTTP-based time sync daemon? Usually systemd-timesyncd (default on DietPi) works pretty fine to sync time with usual NTP servers, at best the local network router/gateway which often provides cached NTP for the whole network.

If you want or need htpdate, try to reinstall it to recover the file that seems to be missing:
apt install --reinstall htpdate

Else, I suggest to purge it and fallback to systemd-timesyncd:
apt purge htpdate
/DietPi/dietpi/func/dietpi-set_software ntpd-mode 2 # Time sync daily and on boot

Joulinar, 2020-05-23:
@jokoren 
`boot+daily` is default setting for time sync service, which is working very well usually.

MichaIng, 2020-05-25:
Hmm actually strange that the DietPi time sync setting breaks htpdate. "Boot + daily" means that there is not even a daemon running that could block anything, and furthermore in case of time sync/NTP nothing is bound to any port + htpdate does not even use the NTP protocol.

It is actually the other way round: If `systemd-timesyncd` sees any other active NTP/time sync daemon, it denies to start up (with condition failure), which breaks DietPi time sync, when "Custom" has not been chosen. I'll run a quick test to see if/how it works out.

MichaIng, 2020-05-25:
Just found that `/media` is "required" to fulfil FHS: https://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch03s02.html
For this reason probably it is expected in other cases as well and we should leave it existing in our images!

closed 2020-06-30 by MichaIng

jokoren, 2020-07-02:
broke everything

will try install of official images

Jim O'Koren 
Computer scientist, retired

Closed #3558.

jokoren, 2020-07-02:
`Be careful my friend"

Jim, former TS/SCI NSA Consultant US NAVY

Be SAFE
Jim O'K

On Thursday, July 2, 2020, 1:51:40 AM EDT, james o'koren <jokoren@yahoo.com> wrote: 

broke everything

will try install of official images

Jim O'Koren 
Computer scientist, retired