MichaIng/DietPi #3559: Trying to format a just expanded partition. (Hyper-V VM)

jakebullet70, 2020-05-23:
#### Details:
- Date           | Sat May 23 08:26:12 BST 2020
- Bug report     | c0bc3e13-5367-47f5-b707-d881614cbcb2
- DietPi version | v6.30.0 (MichaIng/master)
- Image creator  | yumiris
- Pre-image      | Debian
- Hardware       | Virtual Machine (x86_64) (ID=20)
- Kernel version | Linux DietPi 4.19.0-6-amd64 #1 SMP Debian 4.19.67-2+deb10u2 (2019-11-11) x86_64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | dd if=/dev/zero of=/dev/fd0 bs=4K count=1337
- Exit code      | 1
- Software title | DietPi-Drive_Manager
#### Steps to reproduce:

1. Stop VM. 
2. Expand the drive
3. Re-Start VM
4. Run Diet PI drive manager
5. Select new partition and try to format
#### Expected behaviour:

- Create new drive
#### Actual behaviour:

- Error and this on the screen with note to create ticket
#### Extra details:

- Hyper-V VM
#### Additional logs:
```
dd: failed to open '/dev/fd0': No such device or address

![2020-05-23_10-33-11](https://user-images.githubusercontent.com/14124210/82724844-90416180-9ce1-11ea-8d71-6f371d7697e3.jpg)
![2020-05-23_10-33-27](https://user-images.githubusercontent.com/14124210/82724845-90d9f800-9ce1-11ea-87b4-b5f6569c9acf.jpg)
![2020-05-23_10-33-40](https://user-images.githubusercontent.com/14124210/82724846-91728e80-9ce1-11ea-9057-0c271ad3bdaf.jpg)
![2020-05-23_10-33-55](https://user-images.githubusercontent.com/14124210/82724847-91728e80-9ce1-11ea-8967-3d3e56bf0e29.jpg)
![2020-05-23_10-34-13](https://user-images.githubusercontent.com/14124210/82724848-920b2500-9ce1-11ea-8403-de2346308c85.jpg)

```

MichaIng, 2020-05-23:
@jakebullet70 
Many thanks for your report.

Hmm, this is a virtual floppy device. I am wondering why modern VMs still bother with this... I'll try to replicate and see if floppy devices in general have limitations, e.g. write-protection enabled by default or such. Meanwhile is there any special reason why you attach the volume as floppy device instead of e.g. SATA hard disk?

jakebullet70, 2020-05-23:
Not a big Linux guy. What I was trying to do was expand the file base
system from 8GB to 100GB, I did that in hyper-v and all was OK.
Then restarted the VM and just tried to format what I thought was the new
free space.

I did get around it by adding a new SCSI drive and adding that and moving
dietpi_userdata to the new drive.

Though though of making the base image larger then 8GB as it is an
expanding VHDX it really wont matter.

jakebullet70, 2020-05-23:
Just a note to say **thanks** for the Hyper-V image.

MichaIng, 2020-05-23:
Ah okay, the `/dev/fd0` device definitely is not the additional free space, I guess you expended `/dev/sda`.

The problem is that, beside the main data partition `/dev/sda1` there is a swap partition `/dev/sda5` on the same drive which blocks `/dev/sda1` from being expended. Also `dietpi-drive_manager` currently does not support creating or (re)moving multiple partitions on a drive. This would need to be done manually.

Let me see if I can guide you through, please paste the following:
```
resize2fs /dev/sda1
fdisk -l /dev/sda
```

Joulinar, 2020-05-23:
In general I'm using a bootable usb stick running GParted to manage drive and partitions. But yeah never tested it on a VM. Maybe I will give it a try 😉

jakebullet70, 2020-05-23:
Picture is worth a 1000 words...  :)  
Verified the VHDX was expanded to 80GB

This looks like the error: Partition 2 does not start on physical sector boundary.
Thinking maybe the swap file is in the way????   

[image]
[image]

Joulinar, 2020-05-23:
yep, GParted is working as well on my VM (Synology VMM). Quite easy to boot from the ISO and do the magic to resize and format partitions. 

[image]
[image]

jakebullet70, 2020-05-23:
I will get a bootible GParted ISO and give it a shot.

Joulinar, 2020-05-23:
Simply download the iso from their website and select it as boot media on your VM.

MichaIng, 2020-05-23:
> In general I'm using a bootable usb stick running GParted to manage drive and partitions.

Yes `gparted` is the most flexible partitioning tool I know, giving a nice graphical overview and as well allowing to move partitions (with data) physically around in the disk. The command line tools are not that flexible according to my tests, moving partitions with data at least is not possible form all I know, just deleting them and recreating at a different position which makes the data being lost usually.

> This looks like the error: Partition 2 does not start on physical sector boundary.
Thinking maybe the swap file is in the way????

Not sure if it is an error since it is a swap partition. Also the swap partition is a logical partition located inside the "extended" partition `/dev/sda2`. Not sure how sector boundaries apply here.

So there are two possibilities:
In both cases first assure that you have more usable physical memory `available` than what is currently `used` by swap. Then remove the swap partition and underlying extended partition:
```
free -m
swapoff -a # Disable swap partition
parted /dev/sda rm 5 # Remove swap partition
parted /dev/sda rm 2 # Remove extended partition
```
#### 1. Use a swap file instead of a swap partition and use the whole drive as system partition
```
sfdisk --no-reread /dev/sda -fN1 <<< ',+,,,' # Expand main partition until end of drive
resize2fs /dev/sda1 # Expand file system until the end of the new partition size
/DietPi/dietpi/func/dietpi-set_swapfile 1024 # Create and use swapfile with 1 GiB size
```
#### 2. Stay with a swap partition, but move it to the end of the drive
```
parted /dev/sda unit s resizepart 1 165675007 # Expend main partition until 1 GiB before drive end
fdisk /dev/sda # Open fdisk terminal
n # Create new partition
p # Create primary partition
2 # Give it partition number 2
# Hit return two times to accept the defaults for first and last sector, which will fill the remaining free space correctly
t # Select partition type
2 # Select partition 2
L # List hex codes for partition types and check which number is used for "Linux swap" usually it is 82 but on my VirtualBox system it is "19"
82 # Type in the number that matches "Linux swap" in the above table
w # Write partition table to disk
partprobe /dev/sda # Reread partition table
fdisk -l /dev/sda # Check that everything is as it should be
resize2fs /dev/sda1 # Expand file system until the end of the new partition size
mkswap /dev/sda2 # Create swap "file system"
swapon /dev/sda2 # Enable swap partition
echo '/dev/sda2 swap swap default' >> /etc/fstab # Mount swap on boot
```

jakebullet70, 2020-05-23:
I can see the swap file at the end of the original partition so I understand now why I could not just extend the original partition. 

As you can see I got GParted running and I see what I need to do. 
I am going to close the ticket. 

Thanks all! 

[image]

yumiris, 2020-05-24:
Howdy all, just a heads-up! This issue has raised a discussion regarding the disk capacity of future DietPi VHDX images - feel free to contribute with thoughts and suggestions [here](https://github.com/yumiris/DietPi.Hyper-V/issues/6).

MichaIng, 2020-05-25:
I'll reopen this, next larger rework could include adding support for small drives, or generally choosing whether to add a partition table at all or not. E.g. I have my external drives without any partition table, which works fine on Linux at least. AFAIK Windows is not able to detect the volumes then.