MichaIng/DietPi #3560: dietpi-update on Rock64 board+EMMC

linusky, 2020-05-23:
#### Details:
- Date           | Sat 23 May 10:15:42 EDT 2020
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | Rock64 (aarch64) (index=43)
- Kernel version | #1 SMP Thu Oct 12 09:14:48 UTC 2017
- Distro         | stretch (index=4)
- Command        | /DietPi/dietpi/func/dietpi-ramdisk 1
- Exit code      | 127
- Software title | DietPi-Update

#### Steps to reproduce:

1. dietpi-update

#### Expected behaviour:

- ...

#### Actual behaviour:

- One command failed during the update of DietPi

#### Extra details:

- Board Rock64 + EMMC

#### Additional logs:
```
Log file contents:
/DietPi/dietpi/func/dietpi-globals: line 1316: /DietPi/dietpi/func/dietpi-ramdisk: No such file or directory
```

Joulinar, 2020-05-23:
Hi,

many thanks for your report. I guess it will be same like #3532 

Following this comment, issue can be ignored
https://github.com/MichaIng/DietPi/issues/3532#issuecomment-628528041

MichaIng, 2020-05-24:
Jep exactly, simply ignore it, the first boot setup will finish as intended.

ToDo: Redo all images from around Summer 2019, Rock64 is already in beta, will move them to stable during the next week.

linusky, 2020-05-24:
Hi, Thank you for the answer. I ignore it so.
After the reboot all seems to be good.

closed 2020-05-24 by linusky