MichaIng/DietPi #3562: DietPi-Software | Squeezelite update

Joulinar, 2020-05-24:
Hi @MichaIng 

not sure if there is a reason, but it might be good to install Sonarr from `master` branch and not from `developer`.

https://github.com/MichaIng/DietPi/blob/d29a5af277aa1e6b76b4b3057968f0885894e61d/dietpi/dietpi-software#L5396
https://github.com/MichaIng/DietPi/blob/d29a5af277aa1e6b76b4b3057968f0885894e61d/dietpi/dietpi-software#L5404

Basically users will be notified within Sonarr anyway about a new version available if they install from `developer` branch. I found this due to the discussion on the forum https://dietpi.com/phpbb/viewtopic.php?f=11&t=7598

MichaIng, 2020-05-24:
Not sure why we used developer once, probably it solved a particular issue. However I actually planned to migrate to Sonarr v3 anyway with the next release, as v2 is not really developed anymore and v3 works around a nasty mono bug: #3179 #3529

MichaIng, 2020-05-25:
Okay I switched the branch for now, makes totally sense. Sonarr v3 implementation is a larger task, so better to have this merged for now: https://github.com/MichaIng/DietPi/commit/7a4350f648d26215d46acb555223d31b6efc9d15

closed 2020-06-30 by MichaIng

Dr0bac, 2021-03-10:
@MichaIng sonarr 3 is out from beta, they released it few days ago.

Joulinar, 2021-03-10:
means we would need to create a migration from v2 to v3 as they are moved to apt repository now

https://sonarr.tv/#downloads-v3-linux-debian

MichaIng, 2021-03-10:
Thanks for the hint, so this is another task for the next release. Not sure if we should force the migration or integrate it into `dietpi-software` to be done manually? On the other hand from all I read it should be smooth without any data/configs lost or so.

MichaIng, 2021-03-10:
The new package install automatically removes/replaces the old package (nzbdrone) right?

Jep, the simple reinstall should be enough since Sonarr v2 is also installed via their APT repository, so with a changed repo+key+package name all previous files are cleanly replaced.

Joulinar, 2021-03-10:
hmm I did a small test and for me whole config was gone after upgrading to V3. As well first time update failed
```
sonarr.service, stopping and disabling...
Failed to mask unit: File /etc/systemd/system/sonarr.service already exists.
dpkg: error processing archive /var/cache/apt/archives/sonarr_3.0.5_all.deb (--unpack):
 new sonarr package pre-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/sonarr_3.0.5_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
it was needed to rerun. 2nd time it was working well. However some package configuration is needed as you need to specify user/group

yes it seems `nzbdrone` is going to be removed
```
The following package was automatically installed and is no longer required:
  mediainfo
Use 'apt autoremove' to remove it.
The following packages will be REMOVED:
  nzbdrone
The following NEW packages will be installed:
  sonarr
0 upgraded, 1 newly installed, 1 to remove and 4 not upgraded.
```

And finally our service hardening is removed
```
root@DietPi4:~# systemctl cat sonarr.service
# /lib/systemd/system/sonarr.service
[Unit]
Description=Sonarr Daemon
After=network.target

[Service]
User=sonarr
Group=dietpi
UMask=002

Type=simple
ExecStart=/usr/bin/mono --debug /usr/lib/sonarr/bin/Sonarr.exe -nobrowser -data=/var/lib/sonarr
TimeoutStopSec=20
KillMode=process
Restart=on-failure

[Install]
WantedBy=multi-user.target
root@DietPi4:~#
```