MichaIng/DietPi #3563: dietpi-login outputs syntax errors

WasdeAsde, 2020-05-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
`G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=30
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
`buster 10.4`
- Kernel version | `uname -a`
`Linux dietpi-sys 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`RPi 2 Model B (armv7l)`
- Power supply used | (EG: 5V 1A RAVpower)
`5V 2A Official Raspberry Pi Power supply`
- SDcard used | (EG: SanDisk ultra)
`SanDisk ultra 32 GB`

#### Additional Information (if applicable)
- Software title | `dietpi-login`
- Was the software title installed freshly or updated/migrated?
migrated from Dietpi version 6.27.3
- Can this issue be replicated on a fresh installation of DietPi?
yes

- Bug report ID | `echo $G_HW_UUID`
`a13f1d35-977b-429b-a76f-4dbceac95ddd`

#### Steps to reproduce

1. install dietpi v6.30.0 onto a new micro sd card
2. boot with a raspberry pi 2
3. login with root or dietpi user

#### Expected behaviour

- no syntax error messages

#### Actual behaviour

- terminal output after loging in:
`
/boot/dietpi/dietpi-login: Zeile 234: Warnung: Kommansosubstitution: NULL byte in der Eingabe ignoriert.
/boot/dietpi/dietpi-login: Zeile 235: ((: > 0 : Syntax Fehler: Operator erwartet. (Fehlerverursachendes Zeichen ist \"> 0 \").
`
translated:
`
/boot/dietpi/dietpi-login: Line 234: Warning: Command substitution: NULL byte in input ignored.
/boot/dietpi/dietpi-login: Line 235: ((: > 0 : Syntax Error: Operator expected. (The character causing the error is \"> 0 \").
`

#### Extra details

- I have an additional Raspberry Pi 4 Model B in use. There the errors do not occur. Same settings, same SD card.

MichaIng, 2020-05-25:
Many thanks for your report.

These are those two lines: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-login#L234-L235
So `/boot/dietpi/.dietpi-autostart_index` seems to exist but is empty. That should not happen. When choosing an autostart option via `dietpi-autostart` it is assured to be a non-zero integer: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-autostart#L182
Also first boot setup handles this failsafe. Not sure how it can happen, however solution is as simple:
```
echo 0 > /boot/dietpi/.dietpi-autostart_index
```

MichaIng, 2020-06-24:
I mark this as closed. Feel free to reopen if issue is still present.

closed 2020-06-24 by MichaIng