MichaIng/DietPi #3564: Upgrade 2.28 - 2.30 fails

balexandrov, 2020-05-26:
Hi guys!
I'm running DietPi  on RPi 3 Model B+ (armv7l)
There is error when tried to upgrade with dietpi-update to 2.30. Bug report reference code: 4f9dc39a-bf20-450d-81bc-f76263e0b4a8

The installation is fairly clean, nothing custom as far I remember :)
Tried to touch here and there but the error stays the same.

Screen log
```
[ INFO ] DietPi-Patch | APT install for: ./vmtouch_armv7l.deb, please wait...
E: Method file has died unexpectedly!
E: Method /usr/lib/apt/methods/file did not start correctly
[FAILED] DietPi-Patch | APT install for: ./vmtouch_armv7l.deb
 - Command: apt-get -qq --allow-change-held-packages install ./vmtouch_armv7l.deb

```
Info about the system
```
root@DietPi:~# cat /etc/os-release /etc/debian_version
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
10.4
```

```
root@DietPi:~# cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
```
root@DietPi:~# uname -a
Linux DietPi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
```

Thanks!
Bojo

MichaIng, 2020-05-27:
Many thanks for your report.

I remember "`Method https has died unexpectedly!`" which occurred due to some GnuTLS error from what I remember, but this one is new. First idea is corrupted files. Please run the following commands and paste their output here:
```
dmesg -l err,crit,alert,emerg
ls -l /usr/lib/apt/methods/file
apt install --reinstall apt
cd /tmp
wget https://dietpi.com/downloads/binaries/buster/vmtouch_armv7l.deb
apt-get install ./vmtouch_armv7l.deb
rm vmtouch_armv7l.deb
```

balexandrov, 2020-05-27:
Hmm it seems that this manual install went without error. dietpi-upgrade also this time completed successfully...hmm... Don't know what the cause could be but this workaround worked :)

```
root@DietPi:~# dmesg -l err,crit,alert,emerg
[    0.619408] vc_vchi_sm_init: failed to open VCHI service (-1)
[    0.619411] [vc_sm_connected_init]: failed to initialize shared memory service
[    5.110737] vc_sm_cma_vchi_init: failed to open VCHI service (-1)
[    5.110740] [vc_sm_connected_init]: failed to initialize shared memory service
[   10.807123] sd 0:0:0:0: [sda] No Caching mode page found
[   10.807128] sd 0:0:0:0: [sda] Assuming drive cache: write through

```

```
root@DietPi:~# ls -l /usr/lib/apt/methods/file
-rwxr-xr-x 1 root root 34312 May 12 10:57 /usr/lib/apt/methods/file
```

```
root@DietPi:~# apt install --reinstall apt
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 1,349 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://quantum-mirror.hu/mirrors/pub/raspbian/raspbian buster/main armhf apt armhf 1.8.2.1 [1,349 kB]
Fetched 1,349 kB in 1s (2,293 kB/s)
(Reading database ... 37158 files and directories currently installed.)
Preparing to unpack .../archives/apt_1.8.2.1_armhf.deb ...
Unpacking apt (1.8.2.1) over (1.8.2.1) ...
Setting up apt (1.8.2.1) ...
apt-daily-upgrade.timer is a disabled or a static unit not running, not starting it.
apt-daily.timer is a disabled or a static unit not running, not starting it.
Processing triggers for libc-bin (2.28-10+rpi1) ...
```

```
root@DietPi:/tmp# apt-get install ./vmtouch_armv7l.deb
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'vmtouch' instead of './vmtouch_armv7l.deb'
The following NEW packages will be installed:
  vmtouch
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/14.3 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 /tmp/vmtouch_armv7l.deb vmtouch armhf 1.3.1-dietpi1 [14.3 kB]
Selecting previously unselected package vmtouch.
(Reading database ... 37158 files and directories currently installed.)
Preparing to unpack /tmp/vmtouch_armv7l.deb ...
Unpacking vmtouch (1.3.1-dietpi1) ...
Setting up vmtouch (1.3.1-dietpi1) ...
Configuring vmtouch systemd service ...
Created symlink /etc/systemd/system/local-fs.target.wants/vmtouch.service → /lib/systemd/system/vmtouch.service.
```

closed 2020-05-27 by balexandrov

MichaIng, 2020-05-27:
Great! Probably something was broken with APT and fixed by the apt package reinstall.