MichaIng/DietPi #3566: Disk Encryption

nnovaes, 2020-05-27:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
- It would be nice to have some disk encryption option available through dietpi-software. Currently i'm using file encryption through rsync, but having a full disk encryption option would be good, especially with other security sensitive install options such as backup apps and cloud (nextcloud, owncloud) options.

#### Describe the solution you'd like:

- Incorrectly setting a disk encryption could potentially lock users out of access, so a guided install through dietpi-software would be awesome. I don't have any "prescription" for a tool since i'm not very well versed on security, so the community may discuss. 

#### Describe alternatives you've considered:
I've heard of GnuPG, VeraCrypt, and Tomb, but i'm not sure which one would have the best integration, usability, and above all, smaller footprint for the dietpi type of hardware
- ...

MichaIng, 2020-05-27:
Many thanks for your request.

Would [ecryptfs](https://www.ecryptfs.org/) be a good option? A native Linux method (maintained as part of the Linux project itself) to create an encrypted file system on a drive. But I never actively used or tested it, it just came to my mind as we added "support" to at least preserved fstab entries for them when running `dietpi-drive_manager`: #2078
@johnvick should have some experience and probably knows some pro/cons compared to other methods?

Generally I would add it as part of `dietpi-drive_manager`, i.e. when formatting one can chose to create an ecryptfs as well, probably even convert one, although I'm not sure if there is a native conversion method or it will be "copy data elsewhere, format drive with ecryptfs, copy back".

nnovaes, 2020-05-30:
`Generally I would add it as part of dietpi-drive_manager`
haven't thought about that... it would be very cool.

`Would ecryptfs be a good option?`
sounds good, but tbh, not qualified to answer

seniorm0ment, 2021-07-30:
I would like to see full disk encryption as well.
Personally I use LUKS/cryptsetup for all my stuff, and I believe that is the most common.