MichaIng/DietPi #3567: dietpi-update - considered safe?

baikal, 2020-05-27:
I tried to update 6.28 to 6.30 using dietpi-update on a fairly stock** installation of dietpi on a RaspberryPi 3B+ in a remote location. Seemed to work ok, after selection 'OK' in the curses menu my ssh-connection got lost ... couple of minutes later, I could reach the machine again. But it still said "update from 6.28 to 6.30 available". Tried a second time, also did a reboot - each time lost ssh connection and afterwards, it came up again - still 6.28.
Third try, does not come up again ... I have to get there physically before I can get more information.

** syncthing and ftp installed from dietpi-software, system on SDcard, attached USB-harddrive via USB-Combo(Ethernet)-Hub.

MichaIng, 2020-05-27:
Many thanks from your report.

Did you copy or remember the last output on SSH client before the crash? The active network connection is neither dropped, nor is a reboot triggered, so that should not happen.

baikal, 2020-05-27:
No copy, sorry. From memory, very little output if any at all after hitting enter on the 'OK' in the dietpi-update curses menu for 'update to 6.30 now', ssh dropped immediatly. It might not have rebooted, i just waited some minutes anyways (assuming the drop was normal and it would do update and probably reboot). On my third (last/fatal) try, I think I had selected "do backup before updating" ...

Anything specific to look out for when I do have the machine 'in my hands'?
Are services (like ssh, syncthing) getting stopped and/or restarted before/during/after an update?

Joulinar, 2020-05-27:
usually SSH should not be stopped. You could check `journalctl -u dropbear.service` how SSH server is behaving. I assume you still use dropbear SSH as it's default setting

MichaIng, 2020-05-27:
Syncthing is stopped, and things like the webserver or media servers and such, but SSH, VPN, VNC or any other process which is related to the current shell session or network connectivity is left running of course.

Looks like something seriously crashes. When you get hands on it, would be create to do some general checks like:
```
dmesg -l err,crit,alert,emerg # Kernel errors
df # Free disk space
free -m # Free memory space
```

baikal, 2020-05-31:
Had the device sent to me. Boots up fine using a monitor & keyboard, disk space is certainly not a problem, memory looks fine to (but of course freshly booted) - though its not a RPi 3, but a RPi Zero W  :confused:. 

from dmesg -l err,crit,alert,emerg:

```
[    0.756333] vc_vchi_sm_init: failed to open VCHI service (-1)
[    0.756346] [vc_sm_connected_init]: failed to initialize shared memory service
[    3.342952] systemd-fstab-generator[54]: Failed to create unit file /run/systemd/generator/mnt-..._ALL.mount, as it already exists. Duplicate entry in /etc/fstab?
[    3.432729] systemd[50]: /lib/systemd/system-generators/systemd-fstab-generator failed with exit status 1.
[    9.060549] vc_sm_cma_vchi_init: failed to open VCHI service (-1)
[    9.060558] [vc_sm_connected_init]: failed to initialize shared memory service
```
The ones about the mount seem to be a conflict between a manual bind-mount in /etc/fstab and a conflicting one (added by dietpi-drive_manager ?) though that should not affect updates, I assume.

Anyway, this time the update to 6.30 went flawlessly. Though on a local terminal and not over ssh - as I figured just afterwards that would have been a better test. Is there a way to trigger a dummy update to test over ssh as long as I have the machine in my hands? (yes, dropbear as standard ssh)

After manual bind-mounting and running dietpi-drive_manager, the /etc/fstab seems to be "perfect" and that error does not appear any longer, the VCHI / shared memory ones stay.

Joulinar, 2020-05-31:
`VCHI / shared memory` is as expected and can be ignored

baikal, 2020-05-31:
ok, thanks.
I found the 'dietpi-update -1' thing, running that over ssh keeps the connection alive.
So it stays a mystery how the connection would drop when the machine is in its normal (remote) network ... to be seen next time.

closed 2020-05-31 by baikal