MichaIng/DietPi #3568: DietPi-Software | Plex Media Server: Current versions fills /tmp until it's full

MichaIng, 2020-05-27:
### Creating a bug report/issue

#### Required Information
- DietPi version | `v6.30`
- Distro version | `buster`
- SBC model | `Odroid XU4`

#### Additional Information (if applicable)
- Software title | Plex Media Server
- Was the software title installed freshly or updated/migrated? migrated
- Can this issue be replicated on a fresh installation of DietPi? WIP

#### Steps to reproduce

1. Install/update to current PMS package
2. Play some video files
3. `ls -Al /tmp; du -ms /tmp/*`

#### Expected behaviour

- Temporary files created by PMS should be fair below 1 GiB

#### Actual behaviour

- Temporary files reach 1 GiB before other things crash due to /tmp tmpfs being full

#### Extra details

- https://github.com/MichaIng/DietPi/issues/3544#issuecomment-634293391

MichaIng, 2020-05-27:
I'm trying to replicate. Need some large video files for this, it seems, music files do not lead to any tmp files from what I can see. Need to copy some things over, WIP 😉.

MichaIng, 2020-05-27:
Hmm, I was playing a larger MP4 h.264 video, verified that transcoder is active, two dirs are present in /tmp but nothing gets written to them.

@Doctorbeefy
Which kind of media files did you play or manage with PMS when /tmp gets filled?

Doctorbeefy, 2020-05-27:
@MichaIng Mainly mkv I do have some mp4. I have found going to Settings > Library and setting Generate intro video markers to none resolved the issue. They added intro skipping which then scans for intros when you add content. I have a feeling this is what was causing tmp to fill up.

MichaIng, 2020-05-27:
Interesting, I see the changelog here: https://forums.plex.tv/t/plex-media-server/30447/345?u=michaing
But I do not see intro detection setting, just automated preview and chapter thumbnail generation. Is it probably a premium feature? Because I have a free account only, for testing 😉.

Doctorbeefy, 2020-05-27:
@michalng I have the 1.4 gig file that plex created if that would help. I am unsure if it would help or not. I would also need somewhere to upload it to.

Joulinar, 2020-05-27:
I don't thing it's needed to share then file themselves. More important what did you do to have it created 😉

Doctorbeefy, 2020-05-27:
I'll mess around and see if I can recreate it. I am not sure if it happens when I add something to plex or if someone transcodes while watching it. I have quite a bit of friends and family using it.

MichaIng, 2020-05-27:
Jep, from all I understood now, the new intro skip feature transcodes the media as well to scan for the intro, somehow logic 🤔. While the usual playback transcoding is done into a dedicated dir in /var/lib/.. and can be changed via settings, this intro detection transcoding is written to TMPDIR which is `/tmp` by default, together with other non-transcoding tmp data of PMS.

So I guess the 1.4 GiB file is a transcoded video for the purpose of intro scanning.

But maybe I got it wrong, maybe Chuck can clear things up.

Doctorbeefy, 2020-05-27:
@MichaIng lol have to run another test looks like plex deleted it this time. Maybe they fixed it in the last update. It did add a rather large file though but it did not fill up my storage. I wonder if I add a bunch of stuff to the point where it gets full and it still trys to add stuff if that is causing it. 
[image]

The image is what it does in plex when it adds the large file to tmp

Doctorbeefy, 2020-05-27:
```root@DietPi:/tmp# ls -Al /tmp; du -ms /tmp/*
total 108292
-rw-r--r-- 1 plex dietpi 110886990 May 27 16:49 d32e7a43-96f5-4480-85fc-b66912a085e5
drwxr-xr-x 2 root root          60 May 27 16:49 DietPi-Cloudshell
drwxrwxrwt 2 root root          40 May 26 18:00 .font-unix
drwxrwxrwt 2 root root          40 May 26 18:00 .ICE-unix
drwxr-xr-x 3 plex dietpi        60 May 27 16:17 pms-0122803a-67d6-4a46-b28e-f0bb5b0ecd4a
drwxrwxrwt 2 root root          40 May 26 18:00 .Test-unix
drwxrwxrwt 2 root root          40 May 26 18:00 .X11-unix
drwxrwxrwt 2 root root          40 May 26 18:00 .XIM-unix
106	/tmp/d32e7a43-96f5-4480-85fc-b66912a085e5
1	/tmp/DietPi-Cloudshell
0	/tmp/pms-0122803a-67d6-4a46-b28e-f0bb5b0ecd4a
root@DietPi:/tmp# ls -Al /tmp; du -ms /tmp/*
total 80900
-rw-r--r-- 1 plex dietpi 82837582 May 27 16:49 77869299-dec2-464f-bcb8-8bfbc6cea354
drwxr-xr-x 2 root root         60 May 27 16:49 DietPi-Cloudshell
drwxrwxrwt 2 root root         40 May 26 18:00 .font-unix
drwxrwxrwt 2 root root         40 May 26 18:00 .ICE-unix
drwxr-xr-x 3 plex dietpi       60 May 27 16:17 pms-0122803a-67d6-4a46-b28e-f0bb5b0ecd4a
drwxrwxrwt 2 root root         40 May 26 18:00 .Test-unix
drwxrwxrwt 2 root root         40 May 26 18:00 .X11-unix
drwxrwxrwt 2 root root         40 May 26 18:00 .XIM-unix
80	/tmp/77869299-dec2-464f-bcb8-8bfbc6cea354
1	/tmp/DietPi-Cloudshell
0	/tmp/pms-0122803a-67d6-4a46-b28e-f0bb5b0ecd4a
root@DietPi:/tmp# ls -Al /tmp; du -ms /tmp/*
total 118276
-rw-r--r-- 1 plex dietpi 121110606 May 27 16:49 77869299-dec2-464f-bcb8-8bfbc6cea354
drwxr-xr-x 2 root root          60 May 27 16:49 DietPi-Cloudshell
drwxrwxrwt 2 root root          40 May 26 18:00 .font-unix
drwxrwxrwt 2 root root          40 May 26 18:00 .ICE-unix
drwxr-xr-x 3 plex dietpi        60 May 27 16:17 pms-0122803a-67d6-4a46-b28e-f0bb5b0ecd4a
drwxrwxrwt 2 root root          40 May 26 18:00 .Test-unix
drwxrwxrwt 2 root root          40 May 26 18:00 .X11-unix
drwxrwxrwt 2 root root          40 May 26 18:00 .XIM-unix
116	/tmp/77869299-dec2-464f-bcb8-8bfbc6cea354
1	/tmp/DietPi-Cloudshell
0	/tmp/pms-0122803a-67d6-4a46-b28e-f0bb5b0ecd4a
root@DietPi:/tmp# ls -Al /tmp; du -ms /tmp/*
total 0
drwxr-xr-x 2 root root   60 May 27 16:49 DietPi-Cloudshell
drwxrwxrwt 2 root root   40 May 26 18:00 .font-unix
drwxrwxrwt 2 root root   40 May 26 18:00 .ICE-unix
drwxr-xr-x 3 plex dietpi 60 May 27 16:17 pms-0122803a-67d6-4a46-b28e-f0bb5b0ecd4a
drwxrwxrwt 2 root root   40 May 26 18:00 .Test-unix
drwxrwxrwt 2 root root   40 May 26 18:00 .X11-unix
drwxrwxrwt 2 root root   40 May 26 18:00 .XIM-unix
1	/tmp/DietPi-Cloudshell
0	/tmp/pms-0122803a-67d6-4a46-b28e-f0bb5b0ecd4a
root@DietPi:/tmp# 

![Screen Shot 2020-05-27 at 4 49 25 PM](https://user-images.githubusercontent.com/4505410/83070915-2fe05600-a03a-11ea-9c1f-7c0dfcb46584.png)

Doctorbeefy, 2020-05-27:
That's only two episodes. I will test it a little more next time I add a crap ton of episodes. I am assuming that if temp hits 100% before it is done detecting intros that whatever handles the process in plex might crash causing the files to get left in tmp.

MichaIng, 2020-05-27:
Great tests, isolating and documenting the issue perfectly! Good that in this case the dir didn't get larger then something above 100 MiB, which is okay IMO.
Remember [here](https://github.com/MichaIng/DietPi/issues/3544#issuecomment-634293868) was a single directory ~1 GiB (causing the crash), and due to the larger amount of dirs and changing dir names you show now, it seems to be one dir per media file indeed.

The current intro detection included single series episodes, I guess over-length 4k movies are what could break things, if transcoding does not stop intelligently after a few minutes which are assumed to be enough to catch the intro 😉.

MichaIng, 2020-05-31:
Other PMS users run into related issues and devs are already aware of it, hence there will likely be a solution soon 😃.

MichaIng, 2020-06-24:
Issue has been fixed with recent release: https://forums.plex.tv/t/plex-media-server/30447/351

closed 2020-06-24 by MichaIng