MichaIng/DietPi #3569: Cannot Finish Installation - Cannot Ping 1.1.1.1

RileyKemp, 2020-05-27:
#### Details:
- Date           | ᏦᎢᏁ ᎠᏂ 27 17:11:51 CDT 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | ping -c 1 -W 10 1.1.1.1
- Exit code      | 1
- Software title | DietPi-Login

#### Steps to reproduce:

1. ...Turn On Raspi
2. ...Login

#### Expected behaviour:

- ...Should connect to the network and then finish running setup

#### Actual behaviour:

- ...setup cannot complete, cannot ping 1.1.1.1

#### Extra details:

- ...I'm using SSH, so i know its connected to my network
- it will ping www.google.com when i run the network test, 
- i tried forcing IPv4 only, didnt change anything
- i tried changing the pinged ip, didnt change anything

#### Additional logs:
```
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms
```

---------------------------------------------------------------------

Joulinar, 2020-05-27:
Hi

many thanks for your report. You are not able to `ping 1.1.1.1` which usually is a local network issues as this IP is part of Cloudflares DNS Server. We have seen some reports where Cloudflare was not reachable due to network devices like router or some ISP blocking it. `1.1.1.1` is globally available IP address for Cloudflares DNS resolver:

https://www.dnswatch.info/dns/dnslookup?la=en&host=1.1.1.1&submit=Resolve

Anyway if you like to change this, you can do it as follow. 

Do `nano /DietPi/dietpi.txt` and search for the following 2 lines:

```
# General connection and DNS testing
# - IP to ping when checking network connectivity. Default: 1.1.1.1 (Cloudflare DNS, should be very fast world-wide)
CONFIG_CHECK_CONNECTION_IP=1.1.1.1
# - Domain to ping when checking DNS resolver. Default: one.one.one.one (Cloudflare DNS domain, see above)
CONFIG_CHECK_DNS_DOMAIN=one.one.one.one
```

you can change it to `8.8.8.8` and `dns.google` or alternative `9.9.9.9` and `dns9.quad9.net`

RileyKemp, 2020-05-28:
That fixed it, thanks so much

closed 2020-05-28 by RileyKemp