MichaIng/DietPi #3570: Raspberry PI OS (64bit) support

FusionPlmH, 2020-05-28:
Due to the newest new, Raspberry Pi has been started the offical 64bits image on beta .

Here is the link of raspberry pi news:
https://www.raspberrypi.org/blog/8gb-raspberry-pi-4-on-sale-now-at-75/

Raspberry Pi OS (64 bit) beta test version:
https://github.com/MichaIng/DietPi/issues/new?template=image_request.md

Download:
https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2020-05-28/2020-05-27-raspios-buster-arm64.zip

I wanna know Does dietpi want to follow offical , to make up a 64bits OS ? @MichaIng

Joulinar, 2020-05-28:
Hi,

many thanks for your request. Basically DietPi on Raspberry Pi devices is a Raspbian Lite and it will be support. It's as well planned to create 64bit images for RPi2 PCB1.2 and up. 

See https://dietpi.com/phpbb/viewtopic.php?p=25160#p25160

FusionPlmH, 2020-05-28:
So we may start to have some work on it

MichaIng, 2020-05-28:
Lol, copy&paste from forum:

I<s>'m</s> was just checking the 64bit beta OS. It is not called Raspbian, lets see which repo they use: https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2020-05-28/
...
Lol, it's default Debian:
[quote]deb http://deb.debian.org/debian buster main contrib non-free
deb http://deb.debian.org/debian-security/ buster/updates main contrib non-free
deb http://deb.debian.org/debian buster-updates main contrib non-free[/quote]

Actually makes sense. Raspbian was actually only created since the first RPi1 (and RPi Zero now) are armv6hf which is not supported by Debian. For ARMv7 + ARMv8 boards there is no point in using a different distro. All RPi specific firmware/kernel packages, special board-feature/GPU builds of things like Kodi, ffmpeg etc, are all provided by the archive.raspberrypi.org and have nothing to do with Raspbian.

I think this means that we need to change our code in many places so that we do not assume RPi <=> Raspbian anymore but instead check /etc/os-release for whether it's Raspbian or Debian.

MichaIng, 2020-06-22:
I've started working on this: #3617
1. RPi is now detected based on the existence of a device tree file, which is required to boot any RPi with official kernel. I wanted to allow correct detection as well when chrooting into such image or using systemd-nspawn, hence it must not depend on `/proc` or `/sys` content which one would use usually. Checking for a device tree file looked like the most compatible, still quite accurate solution. Basically it must be an ondisk file (or content) which is **only** present on RPi images (as much assured as we can be) while being present on most of them, even when e.g. customising it, e.g. removing the APT kernel package and installing via `rpi-update` only, or compiling oneself. If someone finds a more accurate/failsafe/better method, I'm open to change it. Note that this check is only reached of no `/etc/.dietpi_hw_model_identifier` is present, which is created during DietPi-PREP for all non-RPi models.
2. On RPi, there is a new global variable `G_RASPBIAN` introduced which equals `1` by default and `0` if `/etc/os-release` contains `ID=debian`. Also here in rare circumstances theoretically this file can be wrong, so the choice is whether to assume Raspbian by default (and check for Debian explicitly) or the other way round. For now, where Raspbian is still the most used image, we'll assume it by default.
3. ToDo: Implement into DietPi-Survey, so we can see how the share between both versions develops over time and could and differentiate benchmark results between both, which might be quite interesting.

Now the new variable needs to be rolled out into our scripts, most importantly the one for APT sources.list creation/change. Means we'll need to go through every `G_HW_MODEL` call, see where we do some Raspbian-only step if we find RPi and in case check for `G_RASPBIAN` instead or additionally. Puhh... will take some hours, lets see 😅.

Joulinar, 2020-06-22:
sounds like quite a lot of `search` > `find` > `replace` > `next` 😂

MichaIng, 2020-06-22:
Good that we have grep and https://github.com/MichaIng/DietPi/search?q=G_HW_MODEL&unscoped_q=G_HW_MODEL 😄.

MichaIng, 2020-06-23:
Support ready for review/testing. I'll now create an image and test DietPi-PREP compatibility by this.

Joulinar, 2020-06-23:
Let me know if you need something tested. Otherwise I will download Raspberry OS 64bit and run PREP script

MichaIng, 2020-06-23:
> Otherwise I will download Raspberry OS 64bit and run PREP script

Already in progress 😃, would be great if you could test the resulting image instead.

The image can run on all RPi models from RPi1 v1.2 on, hence those have an ARMv8-capable CPU.

Joulinar, 2020-06-23:
Currently I have a RPi3B+ and RPi4B 4GB only. Don't have that much SBC's :smile:

MichaIng, 2020-06-23:
Here we go: https://dietpi.com/downloads/images/testing/DietPi_RPi-ARMv8-Buster.7z

Joulinar, 2020-06-24:
ok first observation. On boot, the system tries to login as user `pi` automatically. 

[image]

MichaIng, 2020-06-24:
Oh, there must be some `getty@tty1.service` drop-in left:
```
systemctl cat getty@tty1
```

Joulinar, 2020-06-24:
sorry for delay. I used the old image still. attached the install output from CLI. 

[RPi64bit.txt](https://github.com/MichaIng/DietPi/files/4827632/RPi64bit.txt)

I tried to change NTP to my Router, but this is failing. Same behaviour once installation finished using `dietpi-config`

```
root@DietPi:~# dietpi-config
[  OK  ] DietPi-Config | Setting in /boot/dietpi.txt adjusted: CONFIG_NTP_MIRROR=Gateway
[ SUB1 ] DietPi-Set_software > ntpd-mode (2)
[FAILED] DietPi-Set_software | No local gateway detected. Reverting NTP mirror back to system defaults.
[  OK  ] DietPi-Set_software | Setting in /boot/dietpi.txt adjusted: CONFIG_NTP_MODE=2
[FAILED] ntpd-mode 2 | An issue has occurred
[  OK  ] DietPi-Run_NTPD | systemctl restart systemd-timesyncd
[  OK  ] DietPi-Run_NTPD | systemd-timesyncd synced
[  OK  ] Network time sync | Completed
```

MichaIng, 2020-06-24:
>  I used the old image still.

Jep that is perfectly fine, no difference besides the removed autologin.

> I tried to change NTP to my Router, but this is failing.

Good fine 👍, a syntax issue (`local` does not allow to assign and alter a variable in one call, `;` required): https://github.com/MichaIng/DietPi/commit/66f937ab3e4e2edba74918bd8107162c36078de0

Joulinar, 2020-06-24:
how did you like to continue from here? Not sure what to test currently. Do we need to run thru all software title to ensure they are running on Raspberry OS 64bit? I already found WireGuard not working. I guess due to kernel dependency not fully available yet. Maybe I'm wrong.

MichaIng, 2020-06-25:
- Yes WireGuard currently fails as the is no matching headers package available yet for the arm64 kernel package: http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/
- Amiberry requires as well new `pi64` target builds.
- OpenTyrian can/needs to be pulled from Debian repo (is not available on Raspbian): https://packages.debian.org/search?searchon=names&keywords=opentyrian
- DXX-Rebirth need to be rebuild for arm64 or in general as an update, otherwise disabled for arm64 for now.

And 3rd party installers might not yet be prepared, e.g. Pi-hole, PiVPN, Docker.

Joulinar, 2020-06-25:
ok let's wait on the further development of Raspberry OS 64bit.

camdenorrb, 2020-06-29:
This is awesome! :)

FusionPlmH, 2020-07-01:
Just managed to install on running on raspberrypi 3b+ , with pihole and unbound , it working like a charm performance over better than Ubuntu 20.04 64bit and dietpi 32bit system

MichaIng, 2020-07-01:
Many thanks for testing. Awesome. Please let us know if you face any issues at a later time or if any expected feature is missing.

irousom, 2020-07-06:
How I missed this earlier I don't know, but will share some other results after many weeks' time on and off getting a 64-bit OS to run correctly that would execute the DietPi conversion script successfully.  Summary:

1. - Starting with the new Raspberry Pi OS 64-bit on Raspberry Pi 3 B+, I could run the conversion script - but the amount of bloat that was removed was quite high.  
2. - Starting with the new Raspberry Pi OS 64-bit on Raspberry Pi 4, I could run the conversion script, but the subsequent reboot would fail.
3. - Starting with the vanilla Debian image in UEFI mode vis-a-vis [https://pete.akeo.ie/2019/07/installing-debian-arm64-on-raspberry-pi.html](url) was very fast and ran the conversion script just fine.  The resultant image was small and easy to reproduce.
4. - I plan on testing the alpha image offered here on the Raspberry Pi 4 and compare the results to the process in step #3 above but with the RPi 4 firmware [https://github.com/pftf/RPi4/releases](url) instead.

@MichaIng - will this ultimately help meet your objectives, or is this testing redundant with previous efforts?

MichaIng, 2020-07-06:
> Starting with the new Raspberry Pi OS 64-bit on Raspberry Pi 3 B+, I could run the conversion script - but the amount of bloat that was removed was quite high.

Indeed, the current 64-bit image is a huge fully fledged desktop image, so not the best start. However there will be a Lite version for sure at a later time, when a final release comes closer.

> Starting with the new Raspberry Pi OS 64-bit on Raspberry Pi 4, I could run the conversion script, but the subsequent reboot would fail.

Strange, that worked quite fine for our testing image. Did you face any error messages or some display (or serial console) output on failing boot? Since that image contains future version bootloader/kernel/firmware packages, the essential parts should not be touched.

> Starting with the vanilla Debian image in UEFI mode vis-a-vis https://pete.akeo.ie/2019/07/installing-debian-arm64-on-raspberry-pi.html was very fast and ran the conversion script just fine. The resultant image was small and easy to reproduce.

Not sure if UEFI has any benefit here, however good to know it works. Yes result for sure is cleaner. Cleanest and what I aim for mid/long-term is creating our images from scratch via debootstrap which allows to still install bootloader+kernel+firmware from official repositories, archive.raspberrypi.org in this case.

> I plan on testing the alpha image offered here on the Raspberry Pi 4 and compare the results to the process in step #3 above but with the RPi 4 firmware https://github.com/pftf/RPi4/releases instead.

While this is generally interesting and might bring some future benefits, we don't aim to go with experimental firmware and IMO the use for UEFI here is currently minimal. But at least we can take care that DietPi-PREP is compatible with an RPi UEFI images in general and probably the RPi foundation is gonna pick it up by times as well.
_______________
From my side, most important is that our scripts and software installers work as expected with the new official Raspberry Pi OS 64-bit, as this is a fundamental difference compared to Raspbian and we needed to change a lot of code throughout DietPi 🙂.

magloui, 2020-07-07:
Start test with x64 version
I mount old NTFS drive without issue, and share with samba without problem
add drive to fstab
But after restart the drive is not mounted again, need to mount manually
And in samba I have access to the share but not see the content anymore
I check multiple time the config and all seems correct, and work perfectly on x86 version 
waiting for my new hard drive for testing with Ext4 partition

Joulinar, 2020-07-07:
pls can you have a look to `/etc/fstab` how the entry looks like for your drive. Usually it should be `auto mounted` as soon as someone is trying to access

magloui, 2020-07-07:
> 
> pls can you have a look to `/etc/fstab` how the entry looks like for your drive. Usually it should be `auto mounted` as soon as someone is trying to access

I check in the x86 config fstab
UUID=E69EB5EF9EB5B87F /mnt/plex ntfs noatime,lazytime,rw,permissions,nofail,noauto,x-systemd.automount

In X64 is not the same
UUID=E69EB5EF9EB5B87F /mnt/plex ntfs mft_zone_multiplier=1,fmask=0177,dmask=077,nls=utf8,errors=continue,noatime,noauto      0       0

But after copy x86  in the fstab of x64 and reboot, issue exactly the same, no automount and samba return an empty link

But may be it's mistake work on the system only since 2 week ;-)

MichaIng, 2020-07-07:
> UUID=E69EB5EF9EB5B87F /mnt/plex ntfs mft_zone_multiplier=1,fmask=0177,dmask=077,nls=utf8,errors=continue,noatime,noauto 0 0

But this is not the result of `dietpi-drive_manager`. The `x86` is what `dietpi-drive_manager` adds, containing `x-systemd.automount` which is responsible for automount. Please run `dietpi-drive_manager 3` on the x64 system which should enable automount.

magloui, 2020-07-08:
Ok I manually mount drive, clean up and i mount drive with `dietpi-drive_manager`, I saw system install ntfs-3g and after my drive is mounted all the time, I used and samba correctly show disk content
Thank's

Pillendreher, 2020-07-15:
Since I was reinstalling everything anyway, I decided to give this one a try.

1.Initial flash and install to SD works.
2.Moving everything to a USB SSD with the latest bootloader work as well.
3.SSH is functional as well.

I'll update this as I move along with configuring the Pi4.

MichaIng, 2020-07-15:
Many thanks for testing and feedback!

Pillendreher, 2020-07-15:
One issue I've been able to reproduce: Whenever I upload files to the SSD via FTP, my SSH client suddenly turns very sluggish (there's a noticeable pause between key strokes turning into letters). Connecting via VNC, the system itself is sluggish as well. The cursor is moving slowly, the file manager opens five seconds after clicking on the icon. 

htop has the load average of the last minute at 1.71.

MichaIng, 2020-07-15:
Which file system is on the SSD?
VNC is sluggish only during file upload or in general?
Which VNC server do you use? TigerVNC or RealVNC?
And which desktop?

Sorry for question bombardment, but those could all be relevant 😉.

Pillendreher, 2020-07-15:
> Which file system is on the SSD?

ext4

> VNC is sluggish only during file upload or in general?
only during the file upload. Otherwise is very quick. The same goes for SSH btw.

> Which VNC server do you use? TigerVNC or RealVNC?
RealVNC

> And which desktop?

I'm afraid I don't know what you mean - the Raspberry Pi is not set up with a monitor and I didn't install another UI. Under dietpi-config => Display Options => Display Resolution I'm using the standard "vc4-kms-v3d-pi4   : OpenGL | 1920 x 1080"

> Sorry for question bombardment, but those could all be relevant 😉.

No worries at all, I'm just glad I can be of help.

MichaIng, 2020-07-15:
Okay, I'm not sure how CPU intense FTP usually is (I use SFTP for everything), also of course it depends on transfer rate, as the SSD write speed very likely is not the limiting factor. Have you tried other file transfer methods, just to compare?

ext4 and RealVNC should be best options, although, are those very large files you upload (GiB sizes) or many smaller/usual ones?

With RealVNC, a desktop environment is installed, if you didn't choose one explicitly, its LXDE, so the lightest option. However desktop and VNC do not play a role here since the FTP transfer and/or writes through USB to SSD ext4 fs is the culprit.

Ah and did you try another USB port, respectively compared USB3.0 with USB2.0?

FusionPlmH, 2020-07-16:
[image]
Now just update the pihole to v5.1 result like this , and I have been try to uninstall and install it

Joulinar, 2020-07-16:
I had tue same yesterday after doing the update on a normal 32bit image. Restart your system, close browser and clear cache.

FusionPlmH, 2020-07-16:
Now i have been restall all the dietpi + pihole + unbound

i use unbound as my own primary dns server  

now i face another issues like 

after use pihole v5.1 

i was not able to input my local ipv6 dns server ::1#5335 address at the pihole web admin page

here is my unbound config:
https://github.com/FusionPlmH/unbound/blob/master/pi-hole.conf

Joulinar, 2020-07-16:
I don't think this is related to DietPi. Might be better to check on PiHole board.

FusionPlmH, 2020-07-16:
> I don't think this is related to DietPi. Might be better to check on PiHole board.

just checking on pihole board , the answer look like "No need for ipv6 loopback when ipv4 is available",seems this answer is acceptable for me .

Pillendreher, 2020-07-16:
> Okay, I'm not sure how CPU intense FTP usually is (I use SFTP for everything), also of course it depends on transfer rate, as the SSD write speed very likely is not the limiting factor. Have you [...]

Oh my, this is embarassing....I think I figured out what was  happening....

I was outside yesterday and pushed a movie file (30 gb) to the Pi over Wifi. Naturally, the connection wasn't the best. I think the whole sluggishness was caused due to the simple fact that I was maxing out the connection to the Pi with the upload and therefore ssh/vnc appeared sluggish because there was not enough bandwith left. 

I just tried both downloading and uploading big files to and from the Pi via both Samba and FTP over Gigabit Ethernet and didn't encounter any issues.

Consider this issue irrelevant for now. I'll give it another try with my laptop over wifi in the house just to make sure my theory is correct...

MichaIng, 2020-07-16:
No worries, great that you found the root 😃.

CactiChameleon9, 2020-07-19:
Hello.... I am having a few issues. The main one being it does not boot (rpi4). I wondered why I could not find it on my network (preset before booting), so I reinstalled and tried with no changed settings with a HDMI output (tv). Nothing happens however. Furthermore the CPU stays hot for quite a while after booting. Am I doing something wrong?

CactiChameleon9, 2020-07-20:
Hello... I am having issues with wifi. It does not seem to work by just entering in the details before first boot I could not ssh or find on the network.

On first boot, I set up wifi and then I think it downloads and installs 32 bit stuff (downloads and replaces a lot of stuff in /boot). I setup installing only proftpd and it reboots me. I then cannot connect to wifi, with it being "not found" in the dietpi-config network menu.

Joulinar, 2020-07-20:
I guess related to #3685 and seems to be fixed

MichaIng, 2020-07-21:
Image has been updated: https://dietpi.com/downloads/images/testing/DietPi_RPi-ARMv8-Buster.7z
- Based now on DietPi master/stable branch
- Based on now released stable RPi Linux kernel 5.4

I'd call it beta phase now since there are still feature missing on "Raspberry Pi OS (64-bit)" in general and its the first official Linux 5.4 release.

CactiChameleon9, 2020-07-21:
That's great news. I assume I'll have to start fresh to install. Ill do that as soon as libreoffice online compiles for my nextcloud.

Pillendreher, 2020-07-21:
> Image has been updated: https://dietpi.com/downloads/images/testing/DietPi_RPi-ARMv8-Buster.7z
>
>     * Based now on DietPi master/stable branch
>
>     * Based on now released stable RPi Linux kernel [...]

Any chance to update without having to do a complete re-install?

MichaIng, 2020-07-21:
No need to start fresh, simply do:
```
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /boot/dietpi.txt
```
to switch to master branch. This is effective with next DietPi release.
```
apt update
apt full-upgrade
```
to upgrade kernel packages.

csolisr, 2020-07-21:
> 
> 
> No need to start fresh, simply do:
> 
> ```
> G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /boot/dietpi.txt
> ```
> 
> to switch to master branch. This is effective with next DietPi release.
> 
> ```
> apt update
> apt full-upgrade
> ```
> 
> to upgrade kernel packages.

To clarify: does this allow users of the 32-bit version to upgrade to the 64-bit version in-place, or does this only apply for users of the previous 64-bit image?

Joulinar, 2020-07-21:
This applies for 64bit image only.

MichaIng, 2020-07-21:
You in fact cannot upgrade/switch from 32-bit to 64-bit. Raspbian (Raspberry Pi OS 32-bit) and the new Raspberry Pi OS 64-bit are fundamentally different, the whole system is build up on two different major repositories (Raspbian armhf vs Debian arm64).

The underlying OS is btw independent from DietPi version/branch vise-versa. DietPi simply detects detects if Raspbian or Debian and acts accordingly (since DietPi v6.31).

Pillendreher, 2020-07-21:
> apt full-upgrade

> 
> 
> No need to start fresh, simply do:
> 
> ```
> G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /boot/dietpi.txt
> ```
> 
> to switch to master branch. This is effective with next DietPi release.
> 
> ```
> apt update
> apt full-upgrade
> ```
> 
> to upgrade kernel packages.

`root@DietPi:~# G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /boot/dietpi.txt
[  OK  ] G_CONFIG_INJECT | Setting in /boot/dietpi.txt adjusted: DEV_GITBRANCH=master
root@DietPi:~# apt update
Hit:1 https://deb.debian.org/debian buster InRelease
Get:2 https://deb.debian.org/debian buster-updates InRelease [51.9 kB]
Hit:3 https://download.mono-project.com/repo/debian buster InRelease
Get:4 https://deb.debian.org/debian-security buster/updates InRelease [65.4 kB]
Hit:5 https://archive.raspberrypi.org/debian buster InRelease
Get:6 https://deb.debian.org/debian buster-backports InRelease [46.7 kB]
Get:7 https://deb.debian.org/debian buster-backports/main arm64 Packages.diff/Index [27.8 kB]
Get:8 https://deb.debian.org/debian buster-backports/main armhf Packages.diff/Index [27.8 kB]
Get:9 https://deb.debian.org/debian buster-backports/main arm64 Packages 2020-07-21-0801.04.pdiff [596 B]
Get:10 https://deb.debian.org/debian buster-backports/main arm64 Packages 2020-07-21-1404.14.pdiff [174 B]
Get:10 https://deb.debian.org/debian buster-backports/main arm64 Packages 2020-07-21-1404.14.pdiff [174 B]
Get:11 https://deb.debian.org/debian buster-backports/main armhf Packages 2020-07-21-0801.04.pdiff [588 B]
Get:12 https://deb.debian.org/debian buster-backports/main armhf Packages 2020-07-21-1404.14.pdiff [174 B]
Get:12 https://deb.debian.org/debian buster-backports/main armhf Packages 2020-07-21-1404.14.pdiff [174 B]
Fetched 221 kB in 2s (117 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
root@DietPi:~# apt full-upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@DietPi:~#
`

Did I do something wrong?

MichaIng, 2020-07-21:
Nope everything fine then. The firmware/kernel packages were released already yesterday, so depending on when you booted the image the first time this has been upgraded automatically. I just recognised today 😄.

Pillendreher, 2020-07-21:
> 
> 
> Nope everything fine then. The firmware/kernel packages were released already yesterday, so depending on when you booted the image the first time this has been upgraded automatically. I just recognised today 😄.

Oh, that's it then - my Raspberry Pi 4 was shutd down since yesterday afternoon and the first boot today was me trying to update the image.

MattL0, 2020-07-23:
> > apt full-upgrade
> 
> > No need to start fresh, simply do:
> > ```
> > G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /boot/dietpi.txt
> > ```
> > 
> > 
> > to switch to master branch. This is effective with next DietPi release.
> > ```
> > apt update
> > apt full-upgrade
> > ```
> > 
> > 
> > to upgrade kernel packages.
> 
> ```
> root@DietPi:~# G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /boot/dietpi.txt
> [  OK  ] G_CONFIG_INJECT | Setting in /boot/dietpi.txt adjusted: DEV_GITBRANCH=master
> root@DietPi:~# apt update
> Hit:1 https://deb.debian.org/debian buster InRelease
> Get:2 https://deb.debian.org/debian buster-updates InRelease [51.9 kB]
> Hit:3 https://download.mono-project.com/repo/debian buster InRelease
> Get:4 https://deb.debian.org/debian-security buster/updates InRelease [65.4 kB]
> Hit:5 https://archive.raspberrypi.org/debian buster InRelease
> Get:6 https://deb.debian.org/debian buster-backports InRelease [46.7 kB]
> Get:7 https://deb.debian.org/debian buster-backports/main arm64 Packages.diff/Index [27.8 kB]
> Get:8 https://deb.debian.org/debian buster-backports/main armhf Packages.diff/Index [27.8 kB]
> Get:9 https://deb.debian.org/debian buster-backports/main arm64 Packages 2020-07-21-0801.04.pdiff [596 B]
> Get:10 https://deb.debian.org/debian buster-backports/main arm64 Packages 2020-07-21-1404.14.pdiff [174 B]
> Get:10 https://deb.debian.org/debian buster-backports/main arm64 Packages 2020-07-21-1404.14.pdiff [174 B]
> Get:11 https://deb.debian.org/debian buster-backports/main armhf Packages 2020-07-21-0801.04.pdiff [588 B]
> Get:12 https://deb.debian.org/debian buster-backports/main armhf Packages 2020-07-21-1404.14.pdiff [174 B]
> Get:12 https://deb.debian.org/debian buster-backports/main armhf Packages 2020-07-21-1404.14.pdiff [174 B]
> Fetched 221 kB in 2s (117 kB/s)
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> All packages are up to date.
> root@DietPi:~# apt full-upgrade
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Calculating upgrade... Done
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
> root@DietPi:~#
> ```
> 
> Did I do something wrong?
> 
> EDIT:
> 
> ```
> root@DietPi:~# uname -r
> 5.4.51-v8+
> ```

Lol did the same here. Was trying to update the pi. But it was already updated . Saw it with uname -r too

codeandmedia, 2020-07-23:
Hey team, Thank you for your passion! Just a small feedback. I have installed 64bit image on my PRI4 4GB, to USB-flash drive(a little bit new way). I had got first setup error 'cause of USB-install (I guess). Throught dietpi-drive_manager, I resized usb-flash and all is working on succesfully. 

[image]

Pillendreher, 2020-07-25:
I'm sorry I haven't been able to really test the image recently...my laptop has started acting up (losing its connection to its SSD) and I've spent the last week swapping SSDs and flashdrives back and forth between the Pi4, my laptop and my desktop PC just to figure out what was causing the issue...I've never done so much imaging, cloning and restoring in my whole life 😄 
 
Anything in particular that needs testing with the current image? The only issue I've noticed recently is that for the first boot after shutting the Pi4 down with "shutdown -h now", something always seems to get "stuck" (I don't know what exactly; my router tells me that the Pi is not part of my home network even though the ethernet cable is attached as always) and the Pi doesn't seem to boot. A "reset" (unplugging it and plugging it back in) leads to a flawless boot though...I don't know if that's a side effect of the mass storage boot I'm using though.

MichaIng, 2020-07-25:
@codeandmedia 
You mean the USB drive you flashed DietPi on was not expended on first boot automatically? Could you please paste the output of the following command:
```
cat /var/tmp/dietpi/logs/fs_partition_resize.log
```

@Pillendreher 
Are you able to attach a screen to check the system messages on shutdown? Else you could investigate by enabling boot persistent system logs. I'd recommend to do this via persistent systemd-journal (instead of rsyslog):
1. Uninstall `DietPi-RAMlog` via `dietpi-software`
2. `reboot`
3. `mkdir /var/log/journal`
4. `shutdown -h now` or shorter: `poweroff` 😉
5. After reset `journalctl` should show you the systemd logs from shutdown phase.

To revert:
1. `rm -R /var/log/journal`
2. Install `DietPi-RAMlog` via `dietpi-software`

codeandmedia, 2020-07-25:
@MichaIng yes, I'm using simple USB-flash drive instead of SDcard, and first boot was getting issue with install, but after manually resizing all is ok, and working on correctly. 

```
root@Pi:~# cat /var/tmp/dietpi/logs/fs_partition_resize.log
Removed /etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service.
Disk /dev/sda: 28.9 GiB, 31029460992 bytes, 60604416 sectors
Disk model: DataTraveler 3.0
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x99cc74d8

Old situation:

Device     Boot  Start      End  Sectors  Size Id Type
/dev/sda1         8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/sda2       532480 60604415 60071936 28.7G 83 Linux

/dev/sda2:
New situation:
Disklabel type: dos
Disk identifier: 0x99cc74d8

Device     Boot  Start      End  Sectors  Size Id Type
/dev/sda1         8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/sda2       532480 60604415 60071936 28.7G 83 Linux

The partition table has been altered.
Calling ioctl() to re-read partition table.
Re-reading the partition table failed.: Device or resource busy
The kernel still uses the old table. The new table will be used at the next reboot or after you run partprobe(8) or kpartx(8).
Syncing disks.
resize2fs 1.44.5 (15-Dec-2018)
Filesystem at /dev/sda2 is mounted on /; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 2
The filesystem on /dev/sda2 is now 7508992 (4k) blocks long.
```

MichaIng, 2020-07-25:
Ah since you resized through `dietpi-drive_manager`, it used the same service, overwriting the initial log file. From what we see, the partition got resized successfully before (old and new situation are the same) but the `resize2fs` seems to have failed for some reason as it would otherwise show some "nothing to do!" in its output. I'll keep that in mind in case facing some other similar reports. Probably when using external USB drives, we need to add a second sleep between partition resize and file system resize. I'll also compare with `raspi-config` method, if they have some workaround or different method to cover such cases.

[136 more replies]

closed 2021-04-24 by MichaIng