MichaIng/DietPi #3572: DietPi-Software | NAA daemon (networkaudiod) upgrade to 4.x.x on Buster

Heroldgray, 2020-05-28:
On Buster latest NAA is **4.x.x**: https://www.signalyst.eu/bins/naa/linux/buster/
Still 3.6.x on Strech, ALLO Sparky limitation: https://www.signalyst.eu/bins/naa/linux/stretch/
Is it possible DietPi (Buster) automatically install and update latest 4.x version ?

Joulinar, 2020-05-28:
Hi,

many thanks for your report. I did a quick test and current installed version is 4.1.1

 Mode: Installing NAA daemon: signalyst network audio adaptor (naa)

```
[  OK  ] DietPi-Software | Checking URL: https://www.signalyst.eu/bins/naa/linux/buster/networkaudiod_4.1.1-46_armhf.deb
[  OK  ] DietPi-Software | wget https://www.signalyst.eu/bins/naa/linux/buster/networkaudiod_4.1.1-46_armhf.deb -O 124.deb
```

To update your installation you could try running `dietpi-software reinstall 124`

MichaIng, 2020-05-28:
But the official Stretch builds are indeed stuck at 3.6.0 and no new builds have been done since last October 👎.

So yes the mentioned Sparky SBC is stuck on Stretch due to stuck kernel version (Linux 3.10) (actually Stretch already expects Linux 3.16 at least) so it is stuck on NAA 3.6.0. I though about providing a Buster image for Sparky SBC, but not sure about possible regressions.

I just recognised that it's no open source software, so no own builds possible. The Buster packages depend on libc6 2.28 which clearly breaks using it on Stretch. I see no way to work around this my end.

The hint with the update via reinstall is good to have in online docs. Will add it 🙂.

Heroldgray, 2020-05-29:
Thank's @MichaIng and @Joulinar !

closed 2020-07-09 by MichaIng