MichaIng/DietPi #3573: Network | IPv6 connection issues

MichaIng, 2020-05-29:
@Sudrien
I reopened a new issue for this. Does not hurt to separately investigate IPv6 connection issues since this has been a constant pain.

Not sure, somehow pinging any domain that resolves with IPv6 currently fails from my network as well with timeout, while running curl to connect to the webserver on the same IPv6 works pretty fine. I tested the same from our server and observed a significant time delay when pinging to IPv6 before the first answer arrives, around 3 seconds.

Just to verify, if you run
```
ping dns9.quad9.net
```
it resolves to an IPv6 IP?
And does it receive any package at all when waiting long, else which error?
```
ping -4 dns9.quad9.net
```
instead works fine?

Generally when IPv6 works slow I would simply disable it, but ping/ICMPv6 seems to behave somehow different then HTTP(S).

Another try while playing around from dietpi.com server:
```
2020-05-29 09:40:59 root@dietpi:~# ping -v6 google.com
ping: socket: Permission denied, attempting raw socket...
PING google.com(ams15s30-in-x0e.1e100.net (2a00:1450:400e:807::200e)) 56 data bytes
From dietpi.com (XXX) icmp_seq=9 Destination unreachable: Address unreachable
From dietpi.com (XXX) icmp_seq=10 Destination unreachable: Address unreachable
From dietpi.com (XXX) icmp_seq=11 Destination unreachable: Address unreachable
64 bytes from ams15s30-in-x0e.1e100.net (2a00:1450:400e:807::200e): icmp_seq=22 ttl=54 time=2062 ms
64 bytes from ams15s30-in-x0e.1e100.net (2a00:1450:400e:807::200e): icmp_seq=23 ttl=54 time=1038 ms
64 bytes from ams15s30-in-x0e.1e100.net (2a00:1450:400e:807::200e): icmp_seq=24 ttl=54 time=14.3 ms
```

Another interesting attempt:
```
2020-05-29 09:45:14 root@dietpi:~# curl -vIL google.com
* Rebuilt URL to: google.com/
*   Trying 2a00:1450:400e:807::200e...
* TCP_NODELAY set
*   Trying 172.217.17.142...
* TCP_NODELAY set
* Connected to google.com (172.217.17.142) port 80 (#0)
> HEAD / HTTP/1.1
> Host: google.com
> User-Agent: curl/7.52.1
> Accept: */*
...
```
Host resolves to IPv6 but curl quickly switches to IPv6. Forcing IPv6 (`-6`) breaks the connection:
```
2020-05-29 09:43:24 root@dietpi:~# curl -IL -6 google.com
curl: (7) Failed to connect to google.com port 80: No route to host
```

It's a pain, the DNS lookup to get the IPv6 address has no issue and works immediately but connections are somewhat broken or unstable or slow. I'll do further tests...
____________
Obvious solution is to disable IPv6, but it's somehow contra-productive to do that in nearly every case then (either for fix or for connection performance) and by this break all efforts to establish the future 🤔.
____________
```
2020-05-29 11:32:17 root@dietpi:~# wget --spider google.com
Spider mode enabled. Check if remote file exists.
--2020-05-29 11:37:27--  http://google.com/
Resolving google.com (google.com)... 2a00:1450:400e:807::200e, 172.217.17.142
Connecting to google.com (google.com)|2a00:1450:400e:807::200e|:80... failed: No route to host.
Connecting to google.com (google.com)|172.217.17.142|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: http://www.google.com/ [following]
Spider mode enabled. Check if remote file exists.
--2020-05-29 11:37:45--  http://www.google.com/
Resolving www.google.com (www.google.com)... 2a00:1450:4001:81d::2004, 172.217.22.100
Connecting to www.google.com (www.google.com)|2a00:1450:4001:81d::2004|:80... failed: No route to host.
Connecting to www.google.com (www.google.com)|172.217.22.100|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Remote file exists and could contain further links,
but recursion is disabled -- not retrieving.
```
The google.com IPv6 address is btw correct as can be seen by public reverse IP lookups for a google.com server in Germany, which makes sense as our VPS is located there. Also ping -6 succeeds after a few failing attempts. I guess this initial failures make curl and wget switch to IPv4.
_______
#### Traceroutes
```
2020-05-29 13:17:30 root@dietpi:~# traceroute6 google.com
traceroute to google.com (2a00:1450:400e:807::200e), 30 hops max, 80 byte packets
 1  * * *
 2  * * *
 3  * * *
 4  * * *
 5  * * *
 6  * * *
 7  * * *
 8  * * *
 9  * * *
10  * * *
11  * * dietpi.com (XXX)  3051.680 ms !H
2020-05-29 13:18:01 root@dietpi:~# traceroute6 dns9.quad9.net
traceroute to dns9.quad9.net (2620:fe::9), 30 hops max, 80 byte packets
 1  2a06:1c40:3::1 (2a06:1c40:3::1)  13.032 ms  12.880 ms  12.785 ms
 2  2a06:7f80::1 (2a06:7f80::1)  12.880 ms  12.735 ms  12.720 ms
 3  dns9.quad9.net (2620:fe::9)  12.557 ms !X  12.576 ms !X  12.528 ms !X
2020-05-29 13:18:12 root@dietpi:~# ping -6 dns9.quad9.net
PING dns9.quad9.net(dns9.quad9.net (2620:fe::9)) 56 data bytes
64 bytes from dns9.quad9.net (2620:fe::9): icmp_seq=1 ttl=61 time=4.98 ms
64 bytes from dns9.quad9.net (2620:fe::9): icmp_seq=2 ttl=61 time=4.72 ms
```
- `dns9.quad9.net` works fine here but big old Google is not reachable. The question is why `ping -6` finally succeeds after sending ~10 unanswered packets.
______
Another topic:
```
2020-05-29 13:23:46 root@dietpi:~# ping -v dns9.quad9.net
ping: socket: Permission denied, attempting raw socket...
ping: socket: Permission denied, attempting raw socket...
PING dns9.quad9.net(2620:fe::fe:9 (2620:fe::fe:9)) 56 data bytes
64 bytes from 2620:fe::fe:9 (2620:fe::fe:9): icmp_seq=1 ttl=61 time=4.71 ms
64 bytes from 2620:fe::fe:9 (2620:fe::fe:9): icmp_seq=2 ttl=61 time=4.83 ms
64 bytes from 2620:fe::fe:9 (2620:fe::fe:9): icmp_seq=3 ttl=61 time=4.71 ms
```
By default ICMP sockets are disabled: `sysctl net.ipv4.ping_group_range` returns `1 0`
This is for IPv4:
```
2020-05-29 13:24:19 root@dietpi:~# ping -v4 dns9.quad9.net
ping: socket: Permission denied, attempting raw socket...
PING dns9.quad9.net (9.9.9.9) 56(84) bytes of data.
64 bytes from dns9.quad9.net (9.9.9.9): icmp_seq=1 ttl=54 time=4.86 ms
64 bytes from dns9.quad9.net (9.9.9.9): icmp_seq=2 ttl=54 time=4.85 ms
64 bytes from dns9.quad9.net (9.9.9.9): icmp_seq=3 ttl=54 time=5.51 ms
64 bytes from dns9.quad9.net (9.9.9.9): icmp_seq=4 ttl=54 time=4.91 ms
```
Now I set a "group range" from group 0 to group 0, means only the root group is allowed to use ping sockets then:
```
2020-05-29 13:25:20 root@dietpi:~# sysctl -w net.ipv4.ping_group_range='0 0'
net.ipv4.ping_group_range = 0 0
2020-05-29 13:26:45 root@dietpi:~# ping -v4 dns9.quad9.net
PING dns9.quad9.net (149.112.112.9) 56(84) bytes of data.
64 bytes from dns9.quad9.net (149.112.112.9): icmp_seq=1 ttl=54 time=5.74 ms
64 bytes from dns9.quad9.net (149.112.112.9): icmp_seq=2 ttl=54 time=5.09 ms
64 bytes from dns9.quad9.net (149.112.112.9): icmp_seq=3 ttl=54 time=5.54 ms
```
🈯 ICMP(v4) sockets are allowed for root group, hence no error message anymore.
Now looking for a related option for ICMPv6, although not sure if this is related since disabled ICMP sockets are default everywhere.
____
No separate setting for IPv6 available or required, but has nothing to do with unreliable connection:
```
2020-05-29 13:40:19 root@dietpi:~# ping -v dns9.quad9.net
PING dns9.quad9.net(2620:fe::fe:9 (2620:fe::fe:9)) 56 data bytes
From dietpi.com (XXX) icmp_seq=20 Destination unreachable: Address unreachable
From dietpi.com (XXX) icmp_seq=21 Destination unreachable: Address unreachable
From dietpi.com (XXX) icmp_seq=22 Destination unreachable: Address unreachable
From dietpi.com (XXX) icmp_seq=23 Destination unreachable: Address unreachable
From dietpi.com (XXX) icmp_seq=24 Destination unreachable: Address unreachable
From dietpi.com (XXX) icmp_seq=25 Destination unreachable: Address unreachable
From dietpi.com (XXX) icmp_seq=26 Destination unreachable: Address unreachable
64 bytes from 2620:fe::fe:9 (2620:fe::fe:9): icmp_seq=27 ttl=61 time=1023 ms
64 bytes from 2620:fe::fe:9 (2620:fe::fe:9): icmp_seq=28 ttl=61 time=9.82 ms
64 bytes from 2620:fe::fe:9 (2620:fe::fe:9): icmp_seq=29 ttl=61 time=5.49 ms
64 bytes from 2620:fe::fe:9 (2620:fe::fe:9): icmp_seq=30 ttl=61 time=17.2 ms
64 bytes from 2620:fe::fe:9 (2620:fe::fe:9): icmp_seq=31 ttl=61 time=12.9 ms
^C
--- dns9.quad9.net ping statistics ---
31 packets transmitted, 5 received, +7 errors, 83% packet loss, time 30578ms
rtt min/avg/max/mdev = 5.492/213.746/1023.189/404.739 ms, pipe 2
```

_Originally posted by @MichaIng in https://github.com/MichaIng/DietPi/issues/3544#issuecomment-635860987_

Sudrien, 2020-05-30:
Made sure I did it form DietPi, although my other linux boxes get similar - 

```
root@ODROID-bet:~# ping dns9.quad9.net
PING dns9.quad9.net(2620:fe::fe:9 (2620:fe::fe:9)) 56 data bytes
From 2600:1702:49b0:a2f::1 (2600:1702:49b0:a2f::1): icmp_seq=1 Destination unreachable: No route
From 2600:1702:49b0:a2f::1 (2600:1702:49b0:a2f::1): icmp_seq=2 Destination unreachable: No route
From 2600:1702:49b0:a2f::1 (2600:1702:49b0:a2f::1): icmp_seq=3 Destination unreachable: No route
From 2600:1702:49b0:a2f::1 (2600:1702:49b0:a2f::1): icmp_seq=4 Destination unreachable: No route
^C
--- dns9.quad9.net ping statistics ---
4 packets transmitted, 0 received, +4 errors, 100% packet loss, time 8ms

root@ODROID-bet:~# ping -4 dns9.quad9.net
PING dns9.quad9.net (9.9.9.9) 56(84) bytes of data.
64 bytes from dns9.quad9.net (9.9.9.9): icmp_seq=1 ttl=57 time=48.8 ms
64 bytes from dns9.quad9.net (9.9.9.9): icmp_seq=2 ttl=57 time=48.4 ms
64 bytes from dns9.quad9.net (9.9.9.9): icmp_seq=3 ttl=57 time=47.9 ms
64 bytes from dns9.quad9.net (9.9.9.9): icmp_seq=4 ttl=57 time=48.2 ms
^C
--- dns9.quad9.net ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 7ms
rtt min/avg/max/mdev = 47.902/48.317/48.845/0.438 ms
```

Again, this might completely be an issue with the default router mapping (FriendlyWrt on an NanoPi_R1S-H5 at the moment)

Sudrien, 2020-05-30:
Installed `dnsutils` for `dig` - yeah it's getting the IPv4 address, not the IPv6 one.

(OpenWrt, this is in `bind-dig`)

`dig` identified my dns server setup wasn't what I thought it was. 

Toying around, My router was set to resolve from 8.8.8.8 (WAN and WAN6 interfaces) while my LAN was told to resolve from my VPN's DNS servers. Changing the Router's WAN and WAN6 configuration to match lead to LAN/Client pings getting the IPv4 address on pings.

So my answer seems to be that ping doesn't fall back to IPv4 like everything else does? Maybe a cURL call rather than a ping is a more reliable test.

MichaIng, 2020-05-30:
Yes indeed, this is quite a downside of `ping`. However probably it is actually a good thing to recognise issues with IPv6 resolving or connection at an early stage instead of running into related issues later in the middle of a software install. E.g. `wget` falls back to IPv4 after a quite long delay and we don't know how `apt` and `pip` and `npm` and all the other package/module managers and also e.g. 3rd party installer scripts handle it.

What would be helpful for now is to recognise if the connection fails when connecting to an IPv6 address and offer disabling IPv6 right from the connection test error handler.

The reason why I switched from `wget` to `ping` is that at this stage we don't want to test HTTP(S) connection (what curl and wget do) but the pure network connectivity at first. ICMP is quickest and other then HTTPS does not not rely valid certificates which require a successful network time sync, which again is after the initial connection test.
__________
We could make it again more complicated 😄:
1. Check `ping` against IPv4 address
2. Check `ping` against IPv6 address, if this fails, offer to disable IPv6
3. Check `ping` against domain

Joulinar, 2020-05-30:
what about a different view on this. 

1. deactivate IPv6 by default
2. If someone like, he can active IPv6 on Network Settings
3. during IPv6 activation, a ping6 as well as DNS resolution test to be done
4. if all working fine, aktive settings
5. if not, switch back to IPv4 only

MichaIng, 2020-05-30:
I guess it is rare, but it might be possible that someone has an IPv6-only network. Also I don't like it much to break effort to establish IPv6. Although it has downsides, there is no way around it for www. For local networks IPv4 is totally great and sufficient, for www earlier or later not all hosts will be reachable via IPv4 anymore.

Joulinar, 2020-05-30:
question is how much issues we had due to incorrect IPv6 configuration where IPv6 was not even used. Just due to the fact that it was activated by default?

Sudrien, 2020-05-31:
"The reason why I switched from wget to ping is that at this stage we don't want to test HTTP(S) connection (what curl and wget do) but the pure network connectivity at first."

To be sure, wget's `--no-check-certificate` option isn't the desired behavior either? 

There's also `mtr-tiny`, though in my scenario that resolved no better than ping did.

MichaIng, 2020-05-31:
> --no-check-certificate

This skips certs check but it will still be a HTTP or HTTPS check on port 80 or 443 with the related overhead and chance that the webserver is simply down. Ping (ICMP) is a kernel feature, so every host connected to www can answer to this, and it's tiny and fast.

> question is how much issues we had due to incorrect IPv6 configuration where IPv6 was not even used. Just due to the fact that it was activated by default?

I understand your argument. If it's only about bringing things to work on as must systems as possible without incidents, disabling IPv6 by default might be the way to go. Although we cannot know exactly since we do not know how many users actively use IPv6 within their home network which we would break then. And as said, for me compatibility is not everything. IPv6 is the future for www and if there are any issues, I prefer to face them and have a deeper look into it, check if there is really nothing missing our side and else collect and spread knowledge about how to fix it router/network-wise or have ISP contacted, whoever might need to do a fix/enhancements 😃.

The reason why we didn't have those issues a few versions ago btw was the the connection check was done via wget and wget as well as APT are on DietPi by default configured to prefer IPv4. However that only masked the issue partly, since `curl` (used widely as well throughout DietPi scripts and by many other software packages) and of course `ping` do not have a config option to prefer IPv4 (only a command option). So even the initial setup worked, systems with IPv6 connection issues did run into similar issues at a later time or totally different context.

So I really prefer to have have IPv6 enabled by default and as well aim to remove the "prefer IPv4" option. If IPv6 is an issue on that particular system/network, it is good for the user to know and, if no way to fix it network-wise, should be disabled completely. We could even add this to `DietPi-Survey` so we can an idea about share of systems where IPv6 needed or was wanted to be disabled.

Joulinar, 2020-05-31:
hmm what about following idea.

During initial setup, perform some IPv6 verification checks (tbd). If all checks passed, activate IPv6 as is. If a check fails, open user menu to inform about the failed IPv6 check and offer to deactivate IPv6 to be able to continue using IPv4 only.

MichaIng, 2020-05-31:
Yes that was also my idea: https://github.com/MichaIng/DietPi/issues/3573#issuecomment-636387403
Requires an additional dietpi.txt option to choose an IPv6 IP. I think I'll merge all check into a single function, to check everything in reasonable order, so basically a full network troubleshooter. Sounds like an overhead, however the more specific checks we do, the better we can identify the issue and offer specific solutions right within the error handler.
Note to myself: Add info to dietpi.txt that the resolver test URL must be available via IPv6 as well, so we can check if the DNS nameserver is able to resolve the IPv6 + IPv4 IPs both, separately when IPv6 is enabled. Makes sense since bad local/ISP nameservers, which do not resolve IPv6, are still an issue as well, although has become very rare.

MichaIng, 2020-12-17:
Btw, on dietpi.com the issue was an often missing RA (router advertisement) so that the default IPv6 route timed out (30 minutes default): https://superuser.com/q/1557088/1183051

However, long ping/connection times to IPv6 hosts remained, without finding a solution (together with VPS provider). Finally IPv6 remains prone to issues in 2020, which is a sad fact 😢.

I'm gonna mark this as closed, solution as always is to disable IPv6 if there are any issues or bad performance with it.

closed 2020-12-17 by MichaIng

unbaiat, 2021-09-19:
running dietpi latest (7.6.2) on a rpi3. I have pihole installed and a static ipv4 assigned. Please let me know if I need to add ipv6 in /etc/modules and install dhcpcd5.  If so, what is the correct way to configure a static ipv6 address on the dietpi.
Please note that the piholeis also acting as dhcp server.

MichaIng, 2021-09-19:
Usually you won't need at actively apply a static IPv6 address. LAN access to Pi-hole is done via IPv4 (I guess? Else why not?) and for remote connections SLAAC is doing everything pretty fine without any configurations required.

You don't need to manually enable the IPv6 kernel module (it is enabled by default when available) and you must not install dhcpcd, when using the default DietPi/Debian network stack which is based on ifupdown and /etc/network/interfaces.

Check out `ip a` to see your IPv6 addresses which should be an LLA address for connecting to the router and a GUA for connecting to remote IPv6 hosts and optionally an ULA for LAN IPv6 connections (usually not required/recommended). This is independant of the IPv4 DHCP server only for IPv6 setup of network hosts, based on the public IPv6 prefix given to you router/network.