MichaIng/DietPi #3575: SSL certificate for archive.raspberrypi.org invalid

PeterLacknase, 2020-05-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=30
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`

buster

- Kernel version | `uname -a`

Linux dietpi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

RPi 3 Model B+ (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)

5V / 2.0A

- SDcard used | (EG: SanDisk ultra)

SanDisk Extreme (64GB)

#### Steps to reproduce

1. 'sudo apt update'

#### Expected behaviour

It should sync as it does when changing

'deb https://archive.raspberrypi.org/debian/ buster main'

in /etc/apt/sources.list.d/raspi.list to

'deb http://archive.raspberrypi.org/debian/ buster main'

#### Actual behaviour

"https://archive.raspberrypi.org/debian buster Release Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 93.93.135.117 443]"

#### Extra details

The aforementioned error shows up on all my local devices (1x RPi 3 / 2x Zero W) - all running dietpi. Which is the expected result, because it's an external error. But one that affects dietpi, since it relies on that specific repo via ssl.

Joulinar, 2020-05-30:
Hi,

many thanks for your message. The domain `raspbian.raspberrypi.org` is outside DietPi support and we would need to wait until Raspbian has fixed the SSL certificate.

As a temp workaround you could change from secure SSL/HTTPS to a non-secure HTTP connection.

serpaco, 2020-05-30:
> 
> ### Creating a bug report/issue
> #### Required Information
> 
>     * DietPi version | `cat /boot/dietpi/.version`
> 
> 
> G_DIETPI_VERSION_CORE=6
> G_DIETPI_VERSION_SUB=30
> G_DIETPI_VERSION_RC=0
> G_GITBRANCH='master'
> G_GITOWNER='MichaIng'
> 
>     * Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
> 
> 
> buster
> 
>     * Kernel version | `uname -a`
> 
> 
> Linux dietpi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux
> 
>     * SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
> 
> 
> RPi 3 Model B+ (armv7l)
> 
>     * Power supply used | (EG: 5V 1A RAVpower)
> 
> 
> 5V / 2.0A
> 
>     * SDcard used | (EG: SanDisk ultra)
> 
> 
> SanDisk Extreme (64GB)
> #### Steps to reproduce
> 
>     1. `sudo apt update`
> 
> 
> #### Expected behaviour
> 
> It should sync as it does when changing
> 
> `deb https://archive.raspberrypi.org/debian/ buster main`
> 
> in /etc/apt/sources.list.d/raspi.list to
> 
> `deb http://archive.raspberrypi.org/debian/ buster main`
> #### Actual behaviour
> 
> "https://archive.raspberrypi.org/debian buster Release Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate. Could not handshake: Error in the certificate verification. [IP: 93.93.135.117 443]"
> #### Extra details
> 
> The aforementioned error shows up on all my local devices (1x RPi 3 / 2x Zero W) - all running dietpi. Which is the expected result, because it's an external error. But one that affects dietpi in general, since it relies on that specific repo (via ssl).

serpaco, 2020-05-30:
I'm experiencing the same thing.

serpaco, 2020-05-30:
How do you change it to a unsecure SSL

Joulinar, 2020-05-30:
you can do `nano /etc/apt/sources.list.d/raspi.list` and change from `https` to `http`

PeterLacknase, 2020-05-30:
As written above:

Change

`deb https://archive.raspberrypi.org/debian/ buster main`

in /etc/apt/sources.list.d/raspi.list to

`deb http://archive.raspberrypi.org/debian/ buster main`

I knew it's external error from the beginning, but one that affects a lot of people.

serpaco, 2020-05-30:
so in one of the files on the SD card before installing it? I'm trying to install it on my pi4

serpaco, 2020-05-30:
or in the command

serpaco, 2020-05-30:
sorry I'm so new to this. I was hoping to run my raspberry pi4 with an OS that uses FireFox as the browser vs chromium trying to not support google that much these days

Joulinar, 2020-05-30:
this is an external error that occurs today but we can't fix it from DietPi side. Once you hit by this, exit the setup, change the file and reboot.

serpaco, 2020-05-30:
great thanks so much

MichaIng, 2020-05-30:
Looks like it's fixed, since at least my browser does not complain about the https://archive.raspberrypi.org/ certificate. Otherwise probably rerun time sync, probably your system time is in the future, leading to cert being recognised as outdated falsely:
```
/boot/dietpi/func/run_ntpd 1
```

aspee, 2020-05-30:
Changed https to http received below error

`Err:8 http://archive.raspberrypi.org/debian buster/main armhf Packages
  Hash Sum mismatch
  Hashes of expected file:
   - Filesize:329832 [weak]
   - SHA256:4e63b0c51bb911af1fb9bbcd8de2c71b99e976d849dfcd3ef080eec1f70f092d
   - SHA1:5de47912237c8cb299bb05e75108890dbb21ad5c [weak]
   - MD5Sum:c9b40bc67f6b3661ea7dab4194d21b32 [weak]
  Hashes of received file:
   - SHA256:5e558dcd6595a4fece63a912540ec52b3239189b500da29cdb6dd01487e2c5e4
   - SHA1:102526fb2609f784c002e85fa664f322e1475bd1 [weak]
   - MD5Sum:eb6b2f3687bb9f69e2aa90119675b014 [weak]
   - Filesize:329832 [weak]
  Last modification reported: Fri, 29 May 2020 13:15:49 +0000
  Release file created at: Fri, 29 May 2020 14:01:26 +0000
Fetched 13.8 MB in 41s (339 kB/s)
Reading package lists... Done
E: Failed to fetch http://archive.raspberrypi.org/debian/dists/buster/main/binary-armhf/Packages.gz  Hash Sum mismatch
   Hashes of expected file:
    - Filesize:329832 [weak]
    - SHA256:4e63b0c51bb911af1fb9bbcd8de2c71b99e976d849dfcd3ef080eec1f70f092d
    - SHA1:5de47912237c8cb299bb05e75108890dbb21ad5c [weak]
    - MD5Sum:c9b40bc67f6b3661ea7dab4194d21b32 [weak]
   Hashes of received file:
    - SHA256:5e558dcd6595a4fece63a912540ec52b3239189b500da29cdb6dd01487e2c5e4
    - SHA1:102526fb2609f784c002e85fa664f322e1475bd1 [weak]
    - MD5Sum:eb6b2f3687bb9f69e2aa90119675b014 [weak]
    - Filesize:329832 [weak]
   Last modification reported: Fri, 29 May 2020 13:15:49 +0000
   Release file created at: Fri, 29 May 2020 14:01:26 +0000
E: Some index files failed to download. They have been ignored, or old ones used instead.
`

Try performing 
`apt-get clean
rm -rf /var/lib/apt/lists/*
apt-get clean
apt-get update
apt-get upgrade`

Same issue. Can lockdown get worse than this? :-(

Joulinar, 2020-05-30:
I guess there are changes or issues at `raspberry.org` side. So we would need to wait for them to fix

angrykid8, 2020-05-30:
I am unable to install Dietpi due to this, is the a file I can edit on the SD card in windows?

Joulinar, 2020-05-30:
Once you hit by this, exit the setup, change the file and reboot.

MichaIng, 2020-05-30:
Reboot is not even required, the following will continue first boot setup: `/DietPi/dietpi/login`

PeterLacknase, 2020-05-30:
Filed the issue:

https://github.com/RPi-Distro/repo/issues/175

Joulinar, 2020-05-30:
Let's hope they will fix it soon

PeterLacknase, 2020-05-30:
Should be sorted now.

Joulinar, 2020-05-30:
ok I will mark this topic as closed as the issue seems to be fixed at `archive.raspberrypi.org` side

closed 2020-05-30 by Joulinar